qradiolink/qradiolink #1: building SW error

9a4db, 2017-07-20:
Hello, in accordance to instructions reporting build error here
ubuntu 17.04

q@q-X756UX:~/qradiolink/build$ qmake-qt4 ..
q@q-X756UX:~/qradiolink/build$ make
/usr/lib/x86_64-linux-gnu/qt4/bin/uic ../mainwindow.ui -o ui_mainwindow.h
make: /usr/lib/x86_64-linux-gnu/qt4/bin/uic: Command not found
Makefile:511: recipe for target 'ui_mainwindow.h' failed
make: *** [ui_mainwindow.h] Error 127
q@q-X756UX:~/qradiolink/build$ 

73
Djani

kantooon, 2017-08-08:
Hi Djani,
It seems like your qt4 user interface compiler is missing. I suggest checking that you have the QT4 devel packages installed.
Cheers,
Adrian

kantooon, 2017-11-28:
Any updates? If there are none, I will close this issue as "can't reproduce"

closed 2017-11-29 by kantooon