qradiolink/qradiolink #2: Segmentation fault

vestale, 2017-09-27:
Hello,
After a successful build of simplex brench on virtual machine with Debian Jessie, after I press RX button I get Segmentation Fault and the application is clossing .
I tried several configurations in Settings tab of the application, but none with different result. 

> 
> linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown
> 
> 
> (process:4682): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
> Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
> Gtk-Message: Failed to load module "canberra-gtk-module"
> 
> (qradiolink:4682): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
> Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
> Qt: Session management error: None of the authentication protocols specified are supported
> "25/Sep/2017 01:57:17" 
> QColor::setRgb: RGB parameters out of range
> WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
> WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
> WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
> WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
> WARNING: Phonon::createPath: Cannot connect  Phonon::MediaObject ( no objectName ) to  Phonon::AudioOutput ( no objectName ). 
> WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
> Using Volk machine: sse4_2_32_orc
> Error creating proxy: The connection is closed (g-io-error-quark, 18)
> Error creating proxy: The connection is closed (g-io-error-quark, 18)
> Error creating proxy: The connection is closed (g-io-error-quark, 18)
> Error creating proxy: The connection is closed (g-io-error-quark, 18)
> Error creating proxy: The connection is closed (g-io-error-quark, 18)
> gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
> built-in sink types: uhd hackrf bladerf 
> 
> FATAL: No supported devices found to pick from.
> 
> Trying to fill up 1 missing channel(s) with null sink(s).
> This is being done to prevent the application from crashing
> due to gnuradio bug #528.
> 
> gr::log :INFO: audio source - Audio source arch: alsa
> gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
> built-in sink types: uhd hackrf bladerf 
> 
> FATAL: No supported devices found to pick from.
> 
> Trying to fill up 1 missing channel(s) with null sink(s).
> This is being done to prevent the application from crashing
> due to gnuradio bug #528.
> 
> gr::log :INFO: audio source - Audio sink arch: alsa
> gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
> built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
> Using device #0 Realtek RTL2838UHIDIR SN: 00000013
> Found Rafael Micro R820T tuner
> Exact sample rate is: 1000000.026491 Hz
> Segmentation fault

kantooon, 2017-09-27:
Thanks for the report. Could you paste here the content of qradiolink.cfg (same directory as executable)? Also, if you could build the application in debug mode and provide a backtrace, that would be great.

vestale, 2017-09-28:
Here is the content of qradiolink.cfg :
> rx_device_args = "rtl=0";
> tx_device_args = "0";
> rx_antenna = "RX";
> tx_antenna = "";
> rx_freq_corr = 0;
> tx_freq_corr = 0;
> callsign = "YO8SDE";

I did another build of  the application in debug mode ( CONFIG += debug ), but I didn't manage to use the backtrace.

vestale, 2017-10-11:
I did another tried with the last version on master branch.
But I have the same Segmentation fault error, when I press the RX button.
Everything is made using root user.

```
./qradiolink 
linux; GNU C++ version 4.9.1; Boost_105500; UHD_003.007.003-0-unknown

(process:2460): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Gtk-Message: Failed to load module "canberra-gtk-module"

(qradiolink:2460): GConf-WARNING **: Client failed to connect to the D-BUS daemon:
Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Qt: Session management error: None of the authentication protocols specified are supported
"10/Oct/2017 23:22:36" 
QColor::setRgb: RGB parameters out of range
WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
WARNING: Phonon::createPath: Cannot connect  Phonon::MediaObject ( no objectName ) to  Phonon::AudioOutput ( no objectName ). 
WARNING: bool Phonon::FactoryPrivate::createBackend() phonon backend plugin could not be loaded 
Using Volk machine: sse4_2_32_orc
gr::log :INFO: audio source - Audio sink arch: alsa
gr-osmosdr 0.1.3 (0.1.3) gnuradio 3.7.5
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy 
Using device #0 Realtek RTL2838UHIDIR SN: 00000013
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Segmentation fault
```

kantooon, 2017-10-11:
Hi, without a backtrace I'm afraid there's not much I can do here. However I noticed you're using gnuradio version 3.7.5, but the application requires at least 3.7.10. You might also try to compile libgnuradio-osmosdr from the latest git snapshot, that will give you the benefit of some new features and bugfixes.
Also, I advise you to not run the application as the root user, because it's not necessary.

vestale, 2017-11-01:
With the version 3.7.10.1-2 for gnuradio on Debian Stretch , everything is running ok.
Thanks for the help!

closed 2017-11-01 by vestale

kantooon, 2017-11-01:
Glad it works :)