qradiolink/qradiolink #3: Incompatible Protocol Buffer headers

ladecadence, 2017-10-02:
When building, I get this error:

> In file included from ../mumbleclient.h:29:0, 
>                  from ../mainwindow.h:27, 
>                  from ../main.cpp:17: 
> ../ext/Mumble.pb.h:17:2: error: #error This file was generated by an older version of protoc which is 
>  #error This file was generated by an older version of protoc which is 
>   ^~~~~ 
> ../ext/Mumble.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please 
>  #error incompatible with your Protocol Buffer headers.  Please 
>   ^~~~~ 
> ../ext/Mumble.pb.h:19:2: error: #error regenerate this file with a newer version of protoc. 
>  #error regenerate this file with a newer version of protoc. 
>   ^~~~~ 
> In file included from ../telnetserver.h:34:0, 
>                  from ../serverwrapper.h:25, 
>                  from ../main.cpp:32: 
> ../ext/QRadioLink.pb.h:17:2: error: #error This file was generated by an older version of protoc which is 
>  #error This file was generated by an older version of protoc which is 
>   ^~~~~ 
> ../ext/QRadioLink.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please 
>  #error incompatible with your Protocol Buffer headers.  Please 
>   ^~~~~ 
> ../ext/QRadioLink.pb.h:19:2: error: #error regenerate this file with a newer version of protoc. 
>  #error regenerate this file with a newer version of protoc. 
>   ^~~~~ 
> make: *** [Makefile:661: main.o] Error 1

Is there any way to regenerate this files?
This is in Antergos (ArchLinux based) up to date.
$protoc --version
libprotoc 3.3.2

Thanks.

kantooon, 2017-10-03:
Yes. I see that you already have protoc installed. Just run this from the main directory:
cd ext/
protoc --cpp_out=. Mumble.proto
protoc --cpp_out=. QRadioLink.proto
This should regenerate and overwrite the protocol definitions.

kantooon, 2017-10-18:
Instructions added to README

closed 2017-10-18 by kantooon