qradiolink/qradiolink #10: Support installing in a custom prefix

dkozel, 2017-12-07:
I have GNU Radio installed in a custom location, `~/local`. In order to build and install qradiolink I made two changes:

In qradiolink.pro I added `INCLUDEPATH += /home/dkozel/local/include`
In the generated Makefile added `-L/home/dkozel/local/lib` to the `LIBS` variable.

It looks like qmake has the usual ability to pass commandline arguments into the project file when generating the Makefile and maybe the support is already there and I'm just not aware of how to correctly pass the information into qmake.

marcusmueller, 2017-12-07:
Hey, half of the issue, namely hardcoded include and library paths, are solved by externalizing the library discovery to a build script (#12).

The installpath issue: this whole think feels like we're trying to teach qmake tricks that it's not built for. Maybe (this is Maybe with a capital M) it's easiest to just port to CMake; then, we'd get all the complexity and rage that CMake brings, but we'd also get things like protobuf automatism, discovery of dependencies (and complaining _exactly_ about those which are missing), and the installation flexibility.

kantooon, 2017-12-07:
Hi Marcus,
My intention was to switch to CMake as soon as possible, just didn't have enough time. Am I right that if I merge #12 it will also include #11?
Thanks,
Adrian

kantooon, 2017-12-10:
Won't fix until 0.4, sorry.

dkozel, 2017-12-11:
No worries. Thanks for creating the project and sharing it with everyone.

kantooon, 2019-10-21:
I hope the proposed solution is acceptable.

closed 2019-10-21 by kantooon

dkozel, 2019-10-27:
Looks reasonable! Thanks, I'll give it a shot quite soon.