qradiolink/qradiolink #8: No syntax version defined by proto files

dkozel, 2017-12-07:
The Mumble and QRadioLink proto files do not declare a version of protobuf to use. It defaults to version 2.

```
$ protoc --cpp_out=. Mumble.proto 
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: Mumble.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
$ protoc --cpp_out=. QRadioLink.proto 
[libprotobuf WARNING google/protobuf/compiler/parser.cc:547] No syntax specified for the proto file: QRadioLink.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)

```

kantooon, 2017-12-07:
Yes, using default behaviour is naughty, will fix that.

kantooon, 2017-12-10:
Fixed in 0.3.8, thanks.

closed 2017-12-10 by kantooon