qradiolink/qradiolink #9: Add build instructions for Ubuntu

dkozel, 2017-12-07:
In order to build on Ubuntu 17.10 I had to install the following packages. I had a full GNU Radio development environment already installed.

`$ sudo apt install libconfig++-dev libgsm1-dev libprotobuf-dev libopus-dev libpulse-dev libasound2-dev libcodec2-dev libsqlite3-dev libjpeg-dev libprotoc-dev protobuf-compiler`

I also had to install `libqwt-dev` but that actually uninstalled `libqwt5-qt4-dev` on my system. I haven't looked into the difference between those packages, but it seems likely to be qwt/qt versions.

kantooon, 2017-12-07:
I think the libqwt-dev dependency might be a mistake on my part. I don't think GNuradio works with libqwt6, so the correct dep should be libqwt5-qt4-dev

kantooon, 2017-12-10:
Depend on libqwt5-qt4-dev: fixed in 0.3.8, thanks.

closed 2017-12-10 by kantooon