qradiolink/qradiolink #15: installing qradiolink on Ubuntu 16.04 lts

andrew771211, 2018-01-06:
hi,
trying to install qradiolink on Ubuntu 16.04 lts  it gives the libboost issue because the default is the 1.58 version instead of the 1.62,
any idea to solve the problem?
I've checked that is not so simple install libboost 1.62 on this version of ubuntu
thanks

kantooon, 2018-01-06:
Can you please explain what is the libboost issue?

andrew771211, 2018-01-06:
It gives me error, I cannot install because I have libboost-program-options1.58.0, 
The installer doesn't find libboost-program-options1.62.0.
 libjpeg62-turbo and libstdc++6 as weel: Ubuntu 16.04 has libjpeg-turbo8 
My operating system is: 
Linux and-HP-250-G4-Notebook-PC 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
Thanks

kantooon, 2018-01-06:
You should be able to build against those versions as well. The only hard dependencies are Qt 4 and gnuradio >= 3.7.10

andrew771211, 2018-01-06:
I will try again,
I tried installing with *.deb file...

kantooon, 2018-01-06:
Those packages are only meant for Debian Stretch. Build it from source on Ubuntu (see the README).

closed 2018-01-06 by kantooon