qradiolink/qradiolink #16: installation on Kali linux

andrew771211, 2018-01-06:
Hi,
I have Linux kali 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-2kali1 (2017-06-22) x86_64 GNU/Linux
Compiling is ok without errors so:
protoc --cpp_out=. Mumble.proto   OK
protoc --cpp_out=. QRadioLink.proto    OK
cd ..
cd build/
qmake-qt4 ..    OK 
make   OK

but when I type: 
root@kali:~/qradiolink/build# ./qradiolink
I get:
qradiolink: malloc.c:2847: mremap_chunk: Assertion `((size + offset) & (GLRO (dl_pagesize) - 1)) == 0' failed.
Aborted
Thanks for help
Andrew

kantooon, 2018-01-06:
Can I get a backtrace please? Did you build it from the master branch or from next?

andrew771211, 2018-01-06:
git clone https://github.com/kantooon/qradiolink
cd qradiolink
mkdir build
cd ext/
protoc --cpp_out=. Mumble.proto
protoc --cpp_out=. QRadioLink.proto
cd ..
cd build/
qmake-qt4 ..
make  (-j flag is the same...)
./qradiolink

Thanks

kantooon, 2018-01-06:
By backtrace I mean:
run the program with gdb: gdb ./qradiolink
r
after crash:
bt

andrew771211, 2018-01-06:
root@kali:~# cd qradiolink
root@kali:~/qradiolink# cd build
root@kali:~/qradiolink/build# gdb ./qradiolink
bash: gdb: command not found
root@kali:~/qradiolink/build# gdb ./qradiolink
GNU gdb (Debian 7.12-6+b1) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qradiolink...(no debugging symbols found)...done.
(gdb) r
Starting program: /root/qradiolink/build/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
qradiolink: malloc.c:2847: mremap_chunk: Assertion `((size + offset) & (GLRO (dl_pagesize) - 1)) == 0' failed.

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff1a34a70 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff1a3619a in __GI_abort () at abort.c:89
#2  0x00007ffff1a79368 in __malloc_assert (assertion=assertion@entry=0x7ffff1b6dcd0 "((size + offset) & (GLRO (dl_pagesize) - 1)) == 0", file=file@entry=0x7ffff1b6a785 "malloc.c", line=line@entry=2847, function=function@entry=0x7ffff1b6e7d0 <__func__.11285> "mremap_chunk")
    at malloc.c:296
#3  0x00007ffff1a79613 in mremap_chunk (p=p@entry=0x555555ab8650 <DW.ref._ZTISt16invalid_argument>, new_size=new_size@entry=48) at malloc.c:2847
#4  0x00007ffff1a7dd79 in __GI___libc_realloc (oldmem=0x555555ab8660 <QListData::shared_null>, bytes=32)
    at malloc.c:3027
#5  0x00007fffeb6bd301 in QListData::realloc_grow(int) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fffeb6bd39f in QListData::append(int) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007fffeb78b368 in  ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007fffeb786b1d in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007fffeb63c773 in  ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff7de95da in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe2a8, env=env@entry=0x7fffffffe2b8) at dl-init.c:72
---Type <return> to continue, or q <return> to quit---q
Quit

kantooon, 2018-01-06:
Qt4 is a hard dependency, at least as long as GNU radio was built with it. You seem to have built qradiolink with Qt5 which is not supported. Make sure you have Qt 4 >=4.8 (both binaries and development headers) installed before building qradiolink. I hope to switch over to Qt5 when gnuradio is updated.

andrew771211, 2018-01-07:
I have gnuradio3.7.11
I should uninstall libQt5Core and gnuradio, reinstalling gnuradio 3.7.10
thanks

kantooon, 2018-01-07:
No, you should make sure you have Qt4 on your system, both development packages and all libraries, then try to build again. qmake-qt4 should automatically select Qt4 instead of Qt5, not sure what went wrong there. Gnuradio is fine.

andrew771211, 2018-01-07:
I have Qt4 v 4.8.7
this is the make file generated by qt4
[Makefile.txt](https://github.com/kantooon/qradiolink/files/1609613/Makefile.txt)
I'm not an expert in such things but I've looked that Gnuradio 3.7.11 depends on libQt5core
thanks

kantooon, 2018-01-07:
Oh, sorry then! If your gnuradio depends on Qt5 then you have to build qradiolink with qt5 as well, so execute qmake without the -qt4 suffix. Sorry for the confusion, in Debian gnuradio was typically built with qt4.

andrew771211, 2018-01-07:
OK Ok -))
I will try

andrew771211, 2018-01-07:
I have generated the makefile with:
qmake ..
and then make
But the result is the same
[Makefile.txt](https://github.com/kantooon/qradiolink/files/1609656/Makefile.txt)

kantooon, 2018-01-07:
What does qmake --version report? Judging from the makefile, you are still building with qt4.

andrew771211, 2018-01-07:
in fact 
qmake --version
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu

andrew771211, 2018-01-07:
qmake -qt=qt5 ..
Info: creating stash file /root/qradiolink/build/.qmake.stash
sh: 1: /usr/lib/qt5/bin/rcc: not found
sh: 1: /usr/lib/qt5/bin/uic: not found
root@kali:~/qradiolink/build# make
make: *** No rule to make target '/usr/lib/qt5/bin/uic', needed by 'ui_mainwindow.h'.  Stop.
Dear,
 at this point I stop and I wait further updates next weeks/months 
Thanks

kantooon, 2018-01-07:
Your latest error is due to missing qt5 development binaries. It's easily fixed by installing libqt5-dev-bin (or whatever your distro named the development binary utils).

kantooon, 2018-01-11:
I've set this to a documentation bug for now (update build instructions). Ideally, the build process should automatically detect which Qt version is required and fail early instead of building a broken binary. I hope to solve this with the introduction of CMake.

kantooon, 2019-09-14:
Fixed in 0.8.0-rc1

closed 2019-09-14 by kantooon