qradiolink/qradiolink #20: QColor::setRgb: RGB parameters out of range

andrew771211, 2018-01-21:
Finally I Upgraded to ubuntu 17.10 and qradiolink works!

but how I change the colors? the up push buttons ("TX, RX, Config...." )are yellow in a white background 

thanks!
good project!
And

kantooon, 2018-01-21:
Can you please attach a screenshot? The colours are hardcoded, can't be changed by the user. If the application does not look like in the screenshot, then something is overriding the Qt theme, which may be a bug.

andrew771211, 2018-01-21:
[image]

On top left: "TX TUNE SPECTRUM..." white background..

kantooon, 2018-01-21:
Ok, that looks seriously strange. Which window manager are you using? It may be overwriting the Qt theme. LXDE has also given me some headaches because some Qt widgets are displayed differently, like the Repeater, RX, TX buttons not having a background unless hovering.
KDE also displays dials differently, see my screenshot.

andrew771211, 2018-01-21:
MATE Desktop Environment 1.18.0

closed 2018-01-21 by andrew771211