qradiolink/qradiolink #23: Qradiolink installed no audio

satti001, 2018-03-14:
Qradiolink is installed on parrot dev equailent to kali linux. Installed using next version and hardware rtl sdr dongle. I am trying to receive wbfm but I cannot hear any audio but I can hear using command line rtl tool command. I wonder what I am doing wrong

kantooon, 2018-03-14:
First, try to run the application from the terminal and post any output.
I suggest you check on the spectrum tab (FFT) that you can see a signal present. If the spectrum appears as just noise, adjust the RX gain until you can see the WBFM clearly on the FFT.
If there is a WBFM signal centered on the FFT but you can hear nothing, adjust the squech dial to a lower value.

satti001, 2018-03-15:
Thanks for your reply I have tested that way and it works but after changing values of squelch and rx gain application crashed with segmentation fault

kantooon, 2018-03-15:
Unfortunately you're providing too little information. A segfault could indicate a hardware configuration problem or a software issue. You should try to run the application from the terminal and attach any output here. Also, to troubleshoot the segfault, you need to obtain a backtrace by running through gdb. With that information, I may be able to help you.

satti001, 2018-03-16:
I am trying to test it with usrp. Usrp ip address is 192.168.10.2. In gnu we provide the ip address in args same I am doing in qradiolink but terminal says no device found. I am probe usrp but unable to configure it on qradiolink.  Please can you help in this regard

kantooon, 2018-03-16:
Have you tried to use an osmosdr source/sink with gnuradio-companion?
If it works, the arguments to the device should be the same. I don't own an UHD networked device so haven't tested.

kantooon, 2018-03-25:
For support, use the projects mailing list, this tracker is for issues only.

closed 2018-03-25 by kantooon