qradiolink/qradiolink #25: install on Ubuntu 18.04.1 LTS

modezz, 2018-10-26:
Hi,
was no problem to compile but wont start:

after starting:
./qradiolink
realloc(): invalid pointer
Aborted (core dumped)

any clue?
Thanks

kantooon, 2018-12-03:
Sorry for late reply due to limited time. Most likely due to mismatching versions of Qt (built with Qt4 or Qt5 when GNU radio requires different version). Check which version of Qt is a dependency of GNU radio libraries and then use the same version to build the application. If this does not solve it, please provide a gdb trace.

modezz, 2018-12-22:
(gdb) run ./qradiolink
Starting program: /home/qradiolink/build/qradiolink ./qradiolink
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
realloc(): invalid pointer

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb)

modezz, 2018-12-23:
thanks - got it to start at least . did install qt5-default.

but now after astart it quits... see enclosed:
[New Thread 0x7fffcffff700 (LWP 13415)]
Starting qradiolink instance: 23/Dez./2018 07:42:31
Settings not found in configuration file.
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
QColor::setRgb: RGB parameters out of range

Thread 1 "qradiolink" received signal SIGSEGV, Segmentation fault.
0x000055555a78e3e0 in ?? ()
(gdb) bt
#0  0x000055555a78e3e0 in  ()
#1  0x00007fffedc35f14 in QwtPlotMarker::setSymbol(QwtSymbol const*) ()
    at /usr/lib/libqwt-qt5.so.6
#2  0x00007ffff62be2a7 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#3  0x00007ffff62b0631 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#4  0x00007ffff62b07b3 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#5  0x00007ffff2cb5c15 in QMetaProperty::write(QObject*, QVariant const&) const () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff2cde688 in QObject::setProperty(char const*, QVariant const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff4095f94 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff409d8ca in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff409ca0f in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff409cb38 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x000055555557ff3a in main ()
(gdb) quit
A debugging session is active.

modezz, 2018-12-23:
hanks - got it to start at least . did install qt5-default.

but now after astart it quits... see enclosed:
[New Thread 0x7fffcffff700 (LWP 13415)]
Starting qradiolink instance: 23/Dez./2018 07:42:31
Settings not found in configuration file.
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
QColor::setRgb: RGB parameters out of range

Thread 1 "qradiolink" received signal SIGSEGV, Segmentation fault.
0x000055555a78e3e0 in ?? ()
(gdb) bt
#0 0x000055555a78e3e0 in ()
#1 0x00007fffedc35f14 in QwtPlotMarker::setSymbol(QwtSymbol const*) ()
at /usr/lib/libqwt-qt5.so.6
#2 0x00007ffff62be2a7 in ()
at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#3 0x00007ffff62b0631 in ()
at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#4 0x00007ffff62b07b3 in ()
at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#5 0x00007ffff2cb5c15 in QMetaProperty::write(QObject*, QVariant const&) const () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6 0x00007ffff2cde688 in QObject::setProperty(char const*, QVariant const&) ()
at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7 0x00007ffff4095f94 in () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8 0x00007ffff409d8ca in () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9 0x00007ffff409ca0f in () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff409cb38 in () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x000055555557ff3a in main ()
(gdb) quit
A debugging session is active.

kantooon, 2018-12-23:
Seems like a Qt version mismatch between GNU radio and qradiolink. Did you build qradiolink with Qt4 or Qt5?
Also, please post output of
ldd /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so

modezz, 2018-12-24:
first time i compiled qradiolink with qt4 it stopped with the invalid pointer message, now compiled with qt5.

ldd /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so
	linux-vdso.so.1 (0x00007ffecdb21000)
	libgnuradio-filter.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so.3.7.11 (0x00007f0c3b30e000)
	libvolk.so.1.3 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.3 (0x00007f0c3af46000)
	libqwt-qt5.so.6 => /usr/lib/libqwt-qt5.so.6 (0x00007f0c3abf2000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f0c3a3ab000)
	liblog4cpp.so.5 => /usr/lib/x86_64-linux-gnu/liblog4cpp.so.5 (0x00007f0c3a16e000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f0c39bf1000)
	libgnuradio-fft.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-fft.so.3.7.11 (0x00007f0c399bd000)
	libgnuradio-runtime.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11 (0x00007f0c396cd000)
	libgnuradio-pmt.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.11 (0x00007f0c39478000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f0c39270000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f0c3906b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f0c38e4c000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f0c386e3000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f0c37f98000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f0c37c0f000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f0c37871000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f0c37659000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0c37268000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f0c36fec000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007f0c36d7d000)
	libQt5Svg.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007f0c36b28000)
	libQt5OpenGL.so.5 => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f0c368d1000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f0c366b4000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0c364b0000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f0c362ad000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f0c36093000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007f0c35e6e000)
	libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f0c35a61000)
	libfftw3f_threads.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3 (0x00007f0c3585a000)
	libboost_program_options.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.65.1 (0x00007f0c355d9000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f0c3b94f000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f0c3534d000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f0c3511b000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f0c34e7d000)
	libicui18n.so.60 => /usr/lib/x86_64-linux-gnu/libicui18n.so.60 (0x00007f0c349dc000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007f0c34625000)
	libdouble-conversion.so.1 => /usr/lib/x86_64-linux-gnu/libdouble-conversion.so.1 (0x00007f0c34414000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f0c340fe000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f0c33ecd000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f0c33c17000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f0c33963000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f0c33736000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007f0c31b8d000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f0c3191b000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f0c315e3000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f0c313bb000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f0c311b7000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f0c30fb1000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f0c30d9c000)

kantooon, 2018-12-24:
Qt5 should be fine for qradiolink for you. I have no way to test this, but the latest crash seems to happen in QWT, probably due to some code I added after release 0.5.1
Are you building from next branch? Because if you are, I pushed a new commit to remove the offending code which seems to bother qwt-qt5. Either pull the latest changes, or just try to build the 0.5.1 tag.

andrew771211, 2019-02-28:
I installed with:

qmake ..
make
after that:

and@and-HP-250-G4-Notebook-PC:~/qradiolink/build$ ./qradiolink 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Starting qradiolink instance: 28/feb/2019 19:10:16
Settings not found in configuration file.
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
QColor::setRgb: RGB parameters out of range
Segmentation fault (core dumped)

the windows starts and suddenly switch-off

kantooon, 2019-03-01:
Would it be possible to get a gdb traceback? This is could be the same issue as #28 but since I don't have the same setup of libraries, it's not possible for me to reproduce.

andrew771211, 2019-03-01:
and@and-HP-250-G4-Notebook-PC:~/qradiolink/build$ gdb ./qradiolink 2>&1 | tee ~/gdb-./qradiolink.txt
tee: /home/and/gdb-./qradiolink.txt: No such file or directory
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qradiolink...(no debugging symbols found)...done.
(gdb) handle SIG33 pass nostop noprint
Signal        Stop	Print	Pass to program	Description
SIG33         No	No	Yes		Real-time event 33
(gdb) set pagination 0
(gdb) run
Starting program: /home/and/qradiolink/build/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[New Thread 0x7fffdf0ef700 (LWP 8681)]
[New Thread 0x7fffd479f700 (LWP 8682)]
[New Thread 0x7fffd3f9e700 (LWP 8683)]
[New Thread 0x7fffd379d700 (LWP 8684)]
[New Thread 0x7fffd2d60700 (LWP 8685)]
Starting qradiolink instance: 1/mar/2019 19:13:19
Settings not found in configuration file.
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
[New Thread 0x7fffd1527700 (LWP 8686)]
QColor::setRgb: RGB parameters out of range
[New Thread 0x7fffb95cc700 (LWP 8687)]

Thread 1 "qradiolink" received signal SIGSEGV, Segmentation fault.
0x000055555a7d6e00 in ?? ()
(gdb)

kantooon, 2019-03-01:
You can obtain a backtrace by typing bt in gdb shell after the program segfaults. Please post only that information here.

andrew771211, 2019-03-02:
Oh yes,

and@and-HP-250-G4-Notebook-PC:~/qradiolink/build$ gdb ./qradiolink
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qradiolink...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/and/qradiolink/build/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[New Thread 0x7fffdf0ef700 (LWP 5759)]
[New Thread 0x7fffd479f700 (LWP 5760)]
[New Thread 0x7fffd3f9e700 (LWP 5761)]
[New Thread 0x7fffd379d700 (LWP 5762)]
[New Thread 0x7fffd2d60700 (LWP 5763)]
Starting qradiolink instance: 2/mar/2019 10:11:37
Settings not found in configuration file.
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
max_amp: 80 m_pitch: 320
p_min: 20 p_max: 160
Wo_min: 0,039270 Wo_max: 0,314159
nw: 279 tw: 40
[New Thread 0x7fffd1527700 (LWP 5764)]
QColor::setRgb: RGB parameters out of range
[New Thread 0x7fffb95cc700 (LWP 5765)]

Thread 1 "qradiolink" received signal SIGSEGV, Segmentation fault.
0x000055555a7d81a0 in ?? ()
(gdb) bt
#0  0x000055555a7d81a0 in  ()
#1  0x00007fffed967f14 in QwtPlotMarker::setSymbol(QwtSymbol const*) ()
    at /usr/lib/libqwt-qt5.so.6
#2  0x00007ffff62b32a7 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#3  0x00007ffff62a5631 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#4  0x00007ffff62a57b3 in  ()
    at /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so.3.7.11
#5  0x00007ffff2ca9c15 in QMetaProperty::write(QObject*, QVariant const&) const () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007ffff2cd2688 in QObject::setProperty(char const*, QVariant const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff4089f94 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff40918ca in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff4090a0f in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff4090b38 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x000055555557ff4d in main ()
(gdb) 

 thanks

kantooon, 2019-03-02:
It's the same issue as #28 
It looks like some settings of the Qwt plots are passed incorrectly through Gnuradio libraries and into the Qwt library. Unfortunately I can't reproduce this locally because I am still using the Qt4 build, but I can try to create a test brach removing all Qwt settings that you could build from and verify this theory.

andrew771211, 2019-03-02:
Ook 
thanks,
73!

andrew771211, 2019-03-03:
Hi,
the version 0.5.1 works!
I did:

qmake ..
make

and I'm testing with bladerf, seems ok!

kantooon, 2019-03-03:
Do you mean the tag 0.5.1?
It is strange, next should be the same as 0.5.1
In any case, I pushed a new branch issue28 which doesn't have a QWT stylesheet.

closed 2019-03-03 by kantooon

andrew771211, 2019-03-03:
this one:
https://github.com/kantooon/qradiolink/archive/0.5.1.zip

modezz, 2019-03-03:
confirmed - now running on Ubuntu 18.04.1 LTS, bit only this version linked from andrew771211 !!!!

a git pull and qmake/make did fail again

thanks!