qradiolink/qradiolink #26: check qwt include directories

ZeroChaos-, 2019-02-04:
qradiolink.pro has hardcoded paths for qwt include locations.  the correct way to get this value is to use pkg-config.  here is an example of the correct value from Gentoo

```
pkg-config --cflags-only-I Qt5Qwt
-I/usr/include/qwt6 -I/usr/include/qt5/QtConcurrent -I/usr/include/qt5 -I/usr/include/qt5/QtPrintSupport -I/usr/include/qt5 -I/usr/include/qt5/QtSvg -I/usr/include/qt5 -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5 -I/usr/include/qt5/QtGui -I/usr/include/qt5 -I/usr/include/qt5/QtCore -I/usr/include/qt5
```

kantooon, 2019-09-04:
QWT will no longer be used after #30 is merged, so this issue will go away.
Maintainers will need to update their script.

kantooon, 2019-09-05:
Fixed in next branch.

closed 2019-09-05 by kantooon