qradiolink/qradiolink #28: crashes immediately when using qt5

ZeroChaos-, 2019-02-04:
Not sure how useful this bt is for you, but I got this.  Any chance it helps?

```
zero@theprophet qradiolink % gdb qradiolink                                                                                                                                   (git)-[master] 
GNU gdb (Gentoo 8.1 p1) 8.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://bugs.gentoo.org/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from qradiolink...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7fffdfc7d700 (LWP 21400)]
[New Thread 0x7fffd5b8b700 (LWP 21401)]
[New Thread 0x7fffd538a700 (LWP 21402)]
Starting qradiolink instance: 4/Feb/2019 12:45:40
Settings not found in configuration file.
[New Thread 0x7fffcffff700 (LWP 21403)]
QColor::setRgb: RGB parameters out of range
[New Thread 0x7fffcdf56700 (LWP 21404)]

Thread 1 "qradiolink" received signal SIGSEGV, Segmentation fault.
0x00007fffef26c3e9 in QwtPlotMarker::setSymbol(QwtSymbol const*) () from /usr/lib64/libqwt6-qt5.so.6
(gdb) bt
#0  0x00007fffef26c3e9 in QwtPlotMarker::setSymbol(QwtSymbol const*) () from /usr/lib64/libqwt6-qt5.so.6
#1  0x00007ffff644c435 in ?? () from /usr/lib64/libgnuradio-qtgui-3.7.13.4.so.0.0.0
#2  0x00007ffff64436e9 in ?? () from /usr/lib64/libgnuradio-qtgui-3.7.13.4.so.0.0.0
#3  0x00007ffff64438a2 in ?? () from /usr/lib64/libgnuradio-qtgui-3.7.13.4.so.0.0.0
#4  0x00007ffff30b50d3 in QMetaProperty::write(QObject*, QVariant const&) const () from /usr/lib64/libQt5Core.so.5
#5  0x00007ffff30cf6af in QObject::setProperty(char const*, QVariant const&) () from /usr/lib64/libQt5Core.so.5
#6  0x00007ffff42748d6 in ?? () from /usr/lib64/libQt5Widgets.so.5
#7  0x00007ffff427a68b in ?? () from /usr/lib64/libQt5Widgets.so.5
#8  0x00007ffff4279e0d in ?? () from /usr/lib64/libQt5Widgets.so.5
#9  0x00007ffff4279edc in ?? () from /usr/lib64/libQt5Widgets.so.5
#10 0x00005555555838db in main ()
(gdb) 
```

kantooon, 2019-02-04:
Yes, indeed the whole Qt4-Qt5 affair is horrible but I have no way to check what every distro does with GNU radio dependencies, so relying on other people to fix this downstream. I see you already did that, so thanks!

ZeroChaos-, 2019-02-04:
Yes, gentoo ships gnuradio that works with qt5 (patches from debian maintainer)

Question is, is my bt useful to you to find the bug, or do you need me to rebuild with more debugging symbols available?

kantooon, 2019-02-05:
It's a call from Qt into QWT, QWT is an external library so there's nothing I can do. However your backtrace looks suspiciously like you don't have this commit in the code: https://github.com/kantooon/qradiolink/commit/280aecea122b8fc6cb8922e3a76293cbd42625ff
Can you make clean and rebuild, then confirm it still happens? If it does, then QWT simply doesn't accept the GNU radio constellation sinks configuration, and I can't do anything about it.

kantooon, 2019-03-03:
As reported in #25 version 0.5.1 seems to work, and in any case I pushed a new brach called issue28 which removes all QWT stylesheets, in case these are to blame here.

luzpaz, 2019-08-31:
can this ticket be closed then ?

kantooon, 2019-09-04:
A complete UI redesign is underway, removing dependencies on libgnuradio-qtgui and solving these linkage issues. See #30

kantooon, 2019-09-05:
The next branch is now based entirely on Qt5, should build and run fine on any system that can build and run Gqrx. Please test and report back so I can close this issue.

kantooon, 2019-09-14:
Fixed in 0.8.0-rc1

closed 2019-09-14 by kantooon