qradiolink/qradiolink #29: Broken pointer after startup

DL7AD, 2019-07-12:
Hi,
I build the sources on my Ubuntu 18.04.2 LTS with all dependencies. However I couldn't get libjpeg62-turbo but libjpeg62 while it wasn't available. It build perfectly. But on startup it crashed without doing anything.

```
sven@sven-desktop:/src/qradiolink/build$ ./qradiolink 
realloc(): invalid pointer
Aborted (core dumped)
sven@sven-desktop:/src/qradiolink/build$ echo $?
134
```

Here is a backtrace generated by gdb

```
sven@sven-desktop:/src/qradiolink/build$ gdb ./qradiolink 
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qradiolink...(no debugging symbols found)...done.
(gdb) r
Starting program: /src/qradiolink/build/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
realloc(): invalid pointer

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff205ee97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff2060801 in __GI_abort () at abort.c:79
#2  0x00007ffff20a9897 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff21d6b9a "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x00007ffff20b090a in malloc_printerr (str=str@entry=0x7ffff21d4efd "realloc(): invalid pointer") at malloc.c:5350
#4  0x00007ffff20b8eba in __GI___libc_realloc (oldmem=0x5555558a1060 <QListData::shared_null>, bytes=32) at malloc.c:3174
#5  0x00007fffec52f2e1 in QListData::realloc_grow(int) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00007fffec52f37f in QListData::append(int) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007fffec5fd358 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007fffec5f8b0d in qRegisterResourceData(int, unsigned char const*, unsigned char const*, unsigned char const*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007fffec4ae693 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff7de5733 in call_init (env=0x7fffffffdba8, argv=0x7fffffffdb98, argc=1, l=<optimized out>) at dl-init.c:72
#11 0x00007ffff7de5733 in _dl_init (main_map=0x7ffff7ffe170, argc=1, argv=0x7fffffffdb98, env=0x7fffffffdba8) at dl-init.c:119
#12 0x00007ffff7dd60ca in _dl_start_user () at /lib64/ld-linux-x86-64.so.2
#13 0x0000000000000001 in  ()
#14 0x00007fffffffdf3f in  ()
#15 0x0000000000000000 in  ()
(gdb) 
```

kantooon, 2019-07-13:
Acknowledged. 
Can you also post your build command and the output of `ldd /usr/lib/libgnuradio-qtgui.so`?

There is a transition now for packages of Debian Buster using Qt5. Up until now Ubuntu and Debian were using Qt4 for GNU radio graphical UI, and starting with the release of Buster I will transition to using Qt5 only, but it will take a while and may be a bumpy ride.

jaccovanschaik, 2019-07-15:
Hi,

I'm seeing the same problem. Here (for what it's worth) is my output of "ldd /usr/lib/x86_64-linux-gnu/libgnuradio-qtgui.so":

	linux-vdso.so.1 (0x00007ffd35d99000)
	libgnuradio-filter.so.3.7.13 => /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so.3.7.13 (0x00007f097bfce000)
	libvolk.so.1.4 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.4 (0x00007f097bde0000)
	libqwt-qt5.so.6 => /usr/lib/libqwt-qt5.so.6 (0x00007f097bc8a000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f097b627000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f097b2ba000)
	libgnuradio-fft.so.3.7.13 => /usr/lib/x86_64-linux-gnu/libgnuradio-fft.so.3.7.13 (0x00007f097b281000)
	libgnuradio-runtime.so.3.7.13 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.13 (0x00007f097b18f000)
	libgnuradio-pmt.so.3.7.13 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.13 (0x00007f097b136000)
	liblog4cpp.so.5 => /usr/lib/x86_64-linux-gnu/liblog4cpp.so.5 (0x00007f097aef6000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f097aeeb000)
	libboost_system.so.1.67.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.67.0 (0x00007f097aee4000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f097aec3000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f097a912000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f097a3ea000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f097a208000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f097a0ba000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f097a0a0000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f0979eb5000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f0979e34000)
	libQt5Svg.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Svg.so.5 (0x00007f0979ddc000)
	libQt5OpenGL.so.5 => /usr/lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f0979d82000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f0979d66000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f0979d60000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f0979d5b000)
	libboost_filesystem.so.1.67.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.67.0 (0x00007f0979d3c000)
	libboost_thread.so.1.67.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.67.0 (0x00007f0979d10000)
	libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f0979b02000)
	libfftw3f_threads.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3 (0x00007f0979af9000)
	libboost_program_options.so.1.67.0 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.67.0 (0x00007f0979a75000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f097c257000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f09799df000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f09799a8000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f09798af000)
	libicui18n.so.63 => /usr/lib/x86_64-linux-gnu/libicui18n.so.63 (0x00007f09795d4000)
	libicuuc.so.63 => /usr/lib/x86_64-linux-gnu/libicuuc.so.63 (0x00007f0979405000)
	libpcre2-16.so.0 => /usr/lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f097938a000)
	libdouble-conversion.so.1 => /usr/lib/x86_64-linux-gnu/libdouble-conversion.so.1 (0x00007f0979372000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f0979251000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f097921d000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f0979160000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f09790a5000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f0979076000)
	libicudata.so.63 => /usr/lib/x86_64-linux-gnu/libicudata.so.63 (0x00007f0977686000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f0977612000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f09774d8000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f09774af000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f09772a9000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f09770a3000)
	libbsd.so.0 => /usr/lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f0977089000)

My backtrace is almost identical to the one posted above. I am on Ubuntu 19.04. Let me know if you need any more information.

kantooon, 2019-08-07:
Apologies for not responding earlier, was away from internet. Please see this link for an answer: https://stackoverflow.com/questions/36725763/c-realloc-invalid-pointer
Short explanation: the Qt versions don't match, GNU radio was compiled with a different Qt version than qradiolink.

kantooon, 2019-09-04:
A complete UI redesign is underway, removing dependencies on libgnuradio-qtgui and solving these linkage issues. See #29

kantooon, 2019-09-05:
Fixed in next branch, the GNU radio Qt version does not matter anymore.

closed 2019-09-05 by kantooon

jaccovanschaik, 2019-09-05:
Thank you! Seems to work now.

kantooon, 2019-09-06:
Cool. Glad it works.