qradiolink/qradiolink #44: Error after building on Ubuntu 19.10

Yume303, 2019-09-20:
After succesfull building, when I try to run, I got the following error message:

Starting qradiolink instance: 20/szept./2019 17:22:55
Settings not found in configuration file.
Settings not found in configuration file.
QColor::setRgb: RGB parameters out of range
Segmentation fault

If anyone need more info about the building environment, please feel free to ask me...

kantooon, 2019-09-20:
Can you attach a gdb backtrace please?

Yume303, 2019-09-20:
If i try to run with gdb, it draws a window (full of garbage from the terminal.) 
Any suggestion?
[image]

kantooon, 2019-09-20:
Yes, sorry, been doing some refactoring and snuck in a bug there. That commit should fix it.

maxstaylor, 2019-09-27:
Hi. Got a compilation error
cyber@CyberWhell:~/qradiolink-0.8.0-2/build$ make
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I../../qradiolink-0.8.0-2 -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtSql -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -isystem /usr/include/libdrm -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o gr_mod_base.o ../gr/gr_mod_base.cpp
../gr/gr_mod_base.cpp: In constructor ‘gr_mod_base::gr_mod_base(QObject*, float, float, std::__cxx11::string, std::__cxx11::string, int)’:
../gr/gr_mod_base.cpp:71:100: error: ‘MODE_700C’ is not a member of ‘gr::vocoder::freedv_api’
     _freedv_tx700C_usb = make_gr_mod_freedv_sdr(125, 1000000, 1700, 2500, gr::vocoder::freedv_api::MODE_700C, 0);
                                                                                                    ^~~~~~~~~
../gr/gr_mod_base.cpp:79:100: error: ‘MODE_700C’ is not a member of ‘gr::vocoder::freedv_api’
     _freedv_tx700C_lsb = make_gr_mod_freedv_sdr(125, 1000000, 1700, 2500, gr::vocoder::freedv_api::MODE_700C, 1);
                                                                                                    ^~~~~~~~~
Makefile:1596: recipe for target 'gr_mod_base.o' failed
make: *** [gr_mod_base.o] Error 1

kantooon, 2019-09-27:
This is bothering other people as well. It's a mismatch between the GNU radio version and the libcodec2 version (your GNU radio version does not support the FreeDV 700C mode).
What is your GNU radio version? 3.7.13 should have support for this mode, but 3.7.10 does not.
I'm trying to figure out a way to deal with this issue better, other than using the Codec2 headers to figure out what to pass to GNU radio.

maxstaylor, 2019-09-28:
Thank you very much.
I just try to upgrade the version of GR and rebuild the qradiolink.

kantooon, 2019-09-28:
Please use this ticket: #46 to continue discussion for better tracking.

closed 2019-09-28 by kantooon