qradiolink/qradiolink #47: Audio Selection

Radiomix2000, 2019-09-30:
There is not possible to choose custom audio device. 
Would be good to add such option.

kantooon, 2019-09-30:
You can choose any device using the pavucontrol application (playback and recording). Or do you mean something else?

Radiomix2000, 2019-09-30:
No, I mean to choose in application the audiocard, which is not system default. I'm have 2 audiocards in my pc, and one is used only for radiocoms, digi modes, digivoice etc.. and other is used just as system default for listening to music and films.

kantooon, 2019-09-30:
So you're not using Pulseaudio? I have some Alsa code borrowed from another project, but it's not enabled by default. Are you looking at using Alsa directly?

kantooon, 2019-09-30:
Oh. I see what you mean, to choose within the application. Ok, I'll put it on the list.

kantooon, 2019-10-04:
[image]

This is what Pulseaudio devices look like under the new audio backend.

kantooon, 2019-10-04:
In release 0.8.1-4

closed 2019-10-04 by kantooon

kantooon, 2019-10-24:
I should mention that starting with 0.8.2 it's no longer necessary to restart the application to change audio device. The audio threads are restarted when changing the device in settings.