qradiolink/qradiolink #52: Need help with PlutoSDR make work with Qradiolink

Radiomix2000, 2019-10-04:
Hello, 
can anybody help me to make the Pluto SDR work with qradiolink?
I tried different device arguments, but the program still cannot find the device.
Soapy and gnuradio a installed. Soapy-pluto module is installed as well.
Can you share your device arguments for Pluto? Thanks for advance.

kantooon, 2019-10-04:
First find out if Soapy can see the Pluto:
SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = plutosdr
  driver = plutosdr
  uri = usb:2.78.5

Then make sure that gr-osmosdr has support for soapy devices (you should see a line like this when you start the program):
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf *soapy* redpitaya freesrp

Use these arguments for both RX device args and TX: soapy=0,driver=plutosdr

Radiomix2000, 2019-10-04:
Yes, I did it, and got:
`$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found Rafael Micro R820T tuner
Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 a063dc2b52970f
  part_id = a000cb3c0058475f
  serial = 000000000000000000a063dc2b52970f
  version = 2018.01.1

Found device 1
  device = plutosdr
  driver = plutosdr
  uri = usb:2.7.5

Found device 2
  available = Yes
  driver = rtlsdr
  label = Generic RTL2832U OEM :: 00000001
  manufacturer = Realtek
  product = RTL2838UHIDIR
  rtl = 0
  serial = 00000001
  tuner = Rafael Micro R820T
`

Then I started Qradiolink
`$ ./qradiolink 
Starting qradiolink instance: 4/okt/2019 18:54:44
Unable to open FTDI FT245R USB FIFO device 0x0403, 0x6001: -3 device not found
creating net device failed, run setcap "cap_net_raw,cap_net_admin+eip" qradiolink
QColor::setRgb: RGB parameters out of range
warning: The VAD has been replaced by a hack pending a complete rewrite
warning: The VAD has been replaced by a hack pending a complete rewrite
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in sink types: uhd hackrf bladerf soapy redpitaya freesrp file 

FATAL: SoapySDR::Device::make() no match

Trying to fill up 1 missing channel(s) with null sink(s).
This is being done to prevent the application from crashing
`

kantooon, 2019-10-04:
Well, in your case it would be soapy=1,driver=plutosdr since the pluto is device 1 not zero.

Radiomix2000, 2019-10-04:
Yes, I see that, and put soapy=1,driver=plutosdr, but still the same message 
`FATAL: SoapySDR::Device::make() no match`
It's double strange, because if I plug my Lime mini, and put 
`soapy=2,driver=lime` (the lime has device 2 in SoapySDRUtil --find)
it works perfectly.

kantooon, 2019-10-04:
Hmm. Have you tried other GNU radio applications with it? Gqrx for example? All I can think of is some Analog Devices library missing or not found by Soapy because it's not in its search path.

Radiomix2000, 2019-10-06:
still cannot fix it. Tried recompile and reinstall packages, nothing helps.
What's interesting, that the probe command 
`$ SoapySDRUtil --probe="driver=plutosdr"`
shows that Soapy see PlutoSDR
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9364
  ad9361-phy,xo_correction=39999722
  backend_version=0.18 (git tag: v0.18  )
  fw_version=v0.31
  hw_model=Analog Devices PlutoSDR Rev.B (Z7010-AD9364)
  hw_model_variant=1
  hw_serial=104400b8399100090a0009006d1165b516
  library_version=0.18 (git tag: c0012d0)
  local,kernel=4.14.0-42540-g387d584
  usb,idProduct=b673
  usb,idVendor=0456
  usb,libusb=1.0.22.11312
  usb,product=PlutoSDR (ADALM-PLUTO)
  usb,release=2.0
  usb,serial=104400b8399100090a0009006d1165b516
  usb,vendor=Analog Devices Inc.

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Sensors: xadc_temp0, xadc_voltage0, xadc_voltage1, xadc_voltage2, xadc_voltage3, xadc_voltage4, xadc_voltage5, xadc_voltage6, xadc_voltage7, xadc_voltage8, adm1177_current0, adm1177_voltage0, ad9361-phy_temp0, ad9361-phy_voltage2
     * xadc_temp0: 47.493738 C
     * xadc_voltage0 (vccint): 1.004883 V
     * xadc_voltage1 (vccaux): 1.790771 V
     * xadc_voltage2 (vccbram): 1.005615 V
     * xadc_voltage3 (vccpint): 1.004883 V
     * xadc_voltage4 (vccpaux): 1.789307 V
     * xadc_voltage5 (vccoddr): 1.339600 V
     * xadc_voltage6 (vrefp): 1.243652 V
     * xadc_voltage7 (vrefn): -0.003662 V
     * xadc_voltage8: 0.895752 V
     * adm1177_current0: 0.425163 A
     * adm1177_voltage0: 4.889160 V
     * ad9361-phy_temp0: 30.702000 C
     * ad9361-phy_voltage2: 0.529304 V

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED
  Corrections: DC removal
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: 0.065105, 1, 2, 3, 4, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A
  Corrections: DC removal
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: 0.065105, 1, 2, 3, 4, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 1, 2, 3, 4, 6, 7, 8, 9, 10 MHz

```

kantooon, 2019-10-06:
Did you check if you have the proper udev rules in place to be able to access it as non-root? Also, maybe try to see if the IP address gets configured and access it that way. I don't use it via IP, but it may work for you.

Radiomix2000, 2019-10-06:
good, I will try and report it

Radiomix2000, 2019-10-09:
Reinstalled all dependncies from source and problem gone.

closed 2019-10-09 by Radiomix2000