qradiolink/qradiolink #56: Question RX of BPSK 400 / SARSAT-CROSPAS format ?

9a4db, 2019-10-12:
Hi Adrian,
just wondering if someone tested reception of SARSAT BPSK 400 text
in direct (406 MHz) or relay mode (1540 MHz)?
Can existing presets in qradiolink SW allow BPSK 400 setting ?

73
Djani

kantooon, 2019-10-12:
No. The BPSK modem is configured for Codec2 rates (1 kbit/s and 2 kbit/s respectively). Can you give me a link describing the modulation parameters, FEC used, any scrambling, interleaving and framing for that mode? If it's interesting enough I might to implement it.

9a4db, 2019-10-12:
Thank you.
Yes that may be interesting case especially with new Galileo SAR (MEOSAR) RLS
Here you can find useful data about format:
https://www.cospas-sarsat.int/images/stories/SystemDocs/Current/cs_t.001_oct_2013.pdf
And here one presentation covering MEOSAR program development, SARSAT is extended
to Galileo Return Link Service:
https://www.sarsat.noaa.gov/BMW%202016_files/BMW%20Presentation%20Cospas-Sarsat%20Update%20006.pdf

kantooon, 2019-10-13:
Cool, thanks. I'll take a look. Have you checked if this is not already implemented in gr-satellites? In the future, I hope to have some integration with gr-satellites.

9a4db, 2019-10-13:
Not sure about gr-satellites, but know that focus there is on cubesat BPSK on 1200

9a4db, 2019-10-13:
My bad, forgot the QO-100 decoder on gr-satellites ->
"
qo100 QO-100 (Es'hail 2 AMSAT Phase 4-A), which transmits a 400baud BPSK beacon in the 3cm band. The beacon is relayed from the control centre in Qatar via a 13cm/3cm linear transponder on the satellite. The format of the telemetry is the same as that of the AO-40 beacon. Uncoded and FEC frames are sent alternatively. You must use SSB mode to receive this satellite (10489.800MHz)."
https://github.com/daniestevez/gr-satellites

kantooon, 2019-10-17:
I suggest talking to Dani about adding this sat mode in gr-satellites. This first in order to avoid duplication of work in case of future integration, and second because I tend to avoid implementing modes that cannot be also transmitted (for example DMR, YSF or D-Star are easy to implement receive-only with gr-dsd but harder to write the appropriate modems for transmitting).
I'll leave this open in case I change my mind later on.

9a4db, 2019-10-17:
OK, thank you :)