qradiolink/qradiolink #60: [Feature] DSSS using OOT module gr-dsss

Radiomix2000, 2019-11-14:
Hello, Kantoon,
I found that you feel months ago updated your GR packet gr-dsss. 
The parametrized GR graph works like a charm! Just setting necessary parameters, and you get powerful DSSS transceiver of whatever.
What about adding DSSS functionality to Qradiolink for voice and data transmission?

kantooon, 2019-11-14:
In my opinion it does not make much sense to do it for digital voice since there are no technical advantages that I can see for that (if you have any arguments for it, I'd be glad to hear them). However, it might technically make sense for very low rate digital telemetry links, with actual data rate around 50 - 100 bits/second. So yes, that was planned but didn't have time to actually do some research on possible uses and implement the higher level API required for telemetry.

Radiomix2000, 2019-11-19:
Spreading spectrum 10x times gives us +10 dB SNR, so we can work with really VERY weak signals.
Or reduce power to -10 dB, and still have communication.
So using 13 bits Barker code from example graph (gr-dsss) I got close to +13 dB SNR when used it with my codec2 3200 bps TX-RX chain.. or go down to noise level, when reduced power. This could be very interesting to experiment with.

kantooon, 2019-11-19:
Yes, you're right, the spreading gain is real, however it is expressed against an unspread signal operating at 10 times as fast bit rate. So the actual Eb/N0 number and BER is unchanged by the spreading, and you need the same power to transmit a spread signal with the same payload bitrate as an unspread signal (well actually it can be argued that you have to consume more energy since you are running the flowgraph at a higher sample rate). This is the same widespread misunderstanding as in LoRa's case. It is not the spreading of the signal that gives LoRa it's performance, it is the lower bitrate used with high spread modes. Here is a a paper where you can see that a higher spreading factor actually affects the actual BER and Eb/N0 adversely! (https://cora.ucc.ie/bitstream/handle/10468/5181/3697.pdf?sequence=1  --- Table 1)
So since a spread spectrum signal has the disadvantage of occupying more spectrum, what is the advantage? Well, there is a little one: in the case of a very narrow signal with a low bitrate (like WSPR for example) you need a very precise local oscillator and clock with precisions up to a couple of parts per billion to be exactly on frequency and be able to do clock recovery. This can be partly mitigated by spreading the signal and applying some software processing techniques like using a Costas loop to recover the carrier frequency prior to despreading a BPSK signal, leading to a slightly easier clock recovery that can work even with lower precision LO and even some thermal drift.
But the main reason DSSS is used is to be able to run multiple transmitters on the same frequency with orthogonal spreading codes and let the receiver discriminate which is which by means of testing each spreading code in parallel flow. This is the case of GPS signals, which use Gold codes to spread the signal. WCDMA is another use case, but it has other disadvantages like requiring precise MS power control to prevent a close by emitter from covering a more distant emitter so it's being phased out in favour of OFDMA (LTE, 5G etc.)

Radiomix2000, 2020-09-20:
Omg, Thank you! Wil try it. Very interesting!
https://github.com/qradiolink/qradiolink/commit/e4cc20023c7c9e3544cb9613feb04efffb536713

kantooon, 2020-09-20:
No, it won't be a dependency, encoder and decoder blocks are included. But it's still WIP ;-)

Radiomix2000, 2020-09-20:
Good, I will test it and report if any issues detected

kantooon, 2020-10-09:
Feature now present in the next branch. Feel free to experiment as the parameters are not very well defined yet, so there is no specific audio modem for this mode except except a text TX mode which allows me to study and understand the specific property of autocorrelating codes of carrying the signal clock as well as the data.

closed 2020-10-09 by kantooon

Radiomix2000, 2020-10-26:
You mean a text mode which is in "Link services" tab? Or I messed something up? Because this one in link services generates usual, not spread spectrum. Should I activate something in UI?

kantooon, 2020-10-26:
Yes, DSSS is just a text mode for now, use PSK8 (8 symbols per second,  208 chips per second, 240 Hz bw) in the mode selector and text file beacon for minimum hassle playing with it. Feel free to adjust the flowgraph parameters in gr_mod_dsss and gr_demod_dsss however you like to experiment. I'll expand on this mode later when I'm done getting a stable 3.8 release out the door.

Re. BPSK 2K, yes, that's the convolutional code working there. There's more to play with in gr 3.8, polar codes, LDPC etc, just a matter of time. I feel the Codec2 audio level is a bit too low now and needs the audio compressor on all the time, I think I reduced it too much in my quest to avoid audio clipping artefacts. TBC in the next release.

Radiomix2000, 2020-10-26:
Good news, the day promises to be interesting.