qradiolink/qradiolink #61: segmentation fault after install on Ubuntu 19.10

andrew19771211, 2019-11-16:
Hi
After successfully installed on Ubuntu 19.10, If I try to Receive or TXing the program exit with Segmentation faul
 t
hese are the configs:

rx_device_args = "bladerf=0";
tx_device_args = "bladerf=0";
rx_antenna = "RX0";
tx_antenna = "TX0";
rx_freq_corr = 39;
tx_freq_corr = 0;
callsign = "IW0HDI";
video_device = "/dev/video0";
audio_input_device = "default";
audio_output_device = "default";
tx_power = 7;
bb_gain = 1;
rx_sensitivity = 23;
squelch = -89;
rx_volume = 6;
tx_volume = 50;
voip_volume = 50;
rx_frequency = 431100000L;
tx_shift = 0L;
voip_server = "127.0.0.1";
voip_port = 64738;
voip_password = "";
rx_mode = 0;
tx_mode = 0;
rx_ctcss = 0.0;
tx_ctcss = 0.0;
ip_address = "10.0.0.1";
demod_offset = 0L;
rx_sample_rate = 1000000L;
scan_step = 0;
show_controls = 1;
show_constellation = 0;
show_fft = 1;
enable_duplex = 0;
fft_size = 32768;
fft_averaging = 1.0;
waterfall_fps = 15;
audio_compressor = 0;
enable_relays = 0;
rssi_calibration_value = -80;
control_port = 4939;
agc_attack = 2;
agc_decay = 4;
remote_control = 0;
burst_ip_modem = 0;
night_mode = 0;
scan_resume_time = 5;

and@and-HP-250-G4-Notebook-PC:~/qradiolink/build$ ./qradiolink
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[16/nov/2019 11:08:54] [Info] Starting qradiolink
[16/nov/2019 11:08:54] [Warning] creating net device failed, run setcap "cap_net_raw,cap_net_admin+eip" qradiolink
[16/nov/2019 11:08:55] [Info] Using audio input device default
[16/nov/2019 11:08:57] [Info] Using audio output device default
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF Serial # c367...d3a7 FW v1.9.1 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1962] Clamping bandwidth to 1500000 Hz
[bladeRF source] set_freq_corr: Frequency correction is not implemented.
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
Segmentation fault

Thanks

kantooon, 2019-11-16:
Hmm. Interesting. I was never able to test the BladeRF, in fact gr-osmosdr refuses to even build for me on Debian if I don't disable the support for BladeRF. I see you are using the Git version of gr-osmosdr: gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.13.4
The segmentation fault could come from lots of places, including gr-osmosdr or qradiolink itself. Any chance you can provide a GDB backtrace? If it's something I can fix in qradiolink, I will.
Also, is the BladeRF working with Gqrx or GNU radio Companion flowgraphs using gr-osmosdr?

andrew19771211, 2019-11-16:
thanks for the quick reply :)

Here is:

and@and-HP-250-G4-Notebook-PC:~/qradiolink/build$  gdb ./qradiolink 2>&1 | tee ~/gdb-./qradiolink.txt
tee: /home/and/gdb-./qradiolink.txt: No such file or directory
GNU gdb (Ubuntu 8.3-0ubuntu1) 8.3
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./qradiolink...
(gdb) 
(gdb) run
Starting program: /home/and/qradiolink/build/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[New Thread 0x7fffec1ec700 (LWP 5067)]
[New Thread 0x7fffea9f9700 (LWP 5068)]
[New Thread 0x7fffea1f8700 (LWP 5069)]
[New Thread 0x7fffe99f7700 (LWP 5070)]
[New Thread 0x7fffe8fac700 (LWP 5071)]
[16/nov/2019 11:36:03] [Info] Starting qradiolink
[16/nov/2019 11:36:03] [Warning] creating net device failed, run setcap "cap_net_raw,cap_net_admin+eip" qradiolink
[New Thread 0x7fffd3929700 (LWP 5072)]
[New Thread 0x7fffcb128700 (LWP 5073)]
[New Thread 0x7fffd3128700 (LWP 5074)]
[New Thread 0x7fffd2927700 (LWP 5075)]
[New Thread 0x7fffd2126700 (LWP 5076)]
[New Thread 0x7fffd17bf700 (LWP 5077)]
[Thread 0x7fffd17bf700 (LWP 5077) exited]
[16/nov/2019 11:36:05] [Info] Using audio input device default
[New Thread 0x7fffd17bf700 (LWP 5078)]
[New Thread 0x7fffd0fbe700 (LWP 5079)]
[Thread 0x7fffd0fbe700 (LWP 5079) exited]
[16/nov/2019 11:36:07] [Info] Using audio output device default
[New Thread 0x7fffd0fbe700 (LWP 5080)]
[New Thread 0x7fffcbabd700 (LWP 5081)]
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd plutosdr miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[New Thread 0x7fffc991a700 (LWP 5096)]
[bladeRF source] Device: Nuand bladeRF Serial # c367...d3a7 FW v1.9.1 FPGA v0.10.2
[bladeRF source] init: Buffers: 512, samples per buffer: 4096, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1962] Clamping bandwidth to 1500000 Hz
[bladeRF source] set_freq_corr: Frequency correction is not implemented.
[WARNING @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1705] RX DC calibration table not found. Manual gain control will be used instead.
[INFO @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:1706] To enable AGC, see "Generating a DC offset table" at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction
[Thread 0x7fffc991a700 (LWP 5096) exited]

Thread 7 "radioop" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd3929700 (LWP 5072)]
0x00007ffff7f77d50 in pmt::intrusive_ptr_release(pmt::pmt_base*) ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.13
(gdb)

andrew19771211, 2019-11-16:
I checked BladeRF  doesn't work with GNU radio Companion flowgraphs using gr-osmosdr

I try to solve that

kantooon, 2019-11-16:
Thanks. I suspect it's somewhere along the route gr-osmosdr -> libbladerf but I have never used this device so I can't tell what's wrong. The only thing I can try to do is make qradiolink more resilient when the source / sink is not created correctly but I think in this case the root cause is elsewhere.

andrew19771211, 2019-11-17:
I solved.., now qradiolink works fine.
Well Done !! 
(I didn't find the DC Remove, maybe next release will have)
Thanks

closed 2019-11-17 by andrew19771211

kantooon, 2019-11-17:
Glad to hear it's solved.