qradiolink/qradiolink #64: LimeNET micro RPI/CM3 8 GB build from source

9a4db, 2019-12-22:
Hello, 
can you advise please LimeNET micro RPI/CM3 8 GB
build from source.

I was following;
git clone https://github.com/qradiolink/qradiolink
cd qradiolink/
git checkout master
mkdir -p build
cd ext/
protoc --cpp_out=. Mumble.proto
protoc --cpp_out=. QRadioLink.proto
cd ../build/
qmake ..
make
./qradiolink
----------------------------------------------
protobuf and Qt5 installed

pi@raspberrypi:~/qradiolink/build $ qmake ..
Project MESSAGE: /usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++
Project MESSAGE: Building for GNU/Linux
Project MESSAGE: Building without OpenGL support
Project MESSAGE: Building with Qt audio support.
pi@raspberrypi:~/qradiolink/build $ make
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtMultimedia -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o imagecapture.o ../video/imagecapture.cpp
In file included from ../video/imagecapture.h:24,
                 from ../video/imagecapture.cpp:17:
../../qradiolink/src/settings.h:26:10: fatal error: libconfig.h++: No such file or directory
 #include <libconfig.h++>
          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:1053: imagecapture.o] Error 1
pi@raspberrypi:~/qradiolink/build $

kantooon, 2019-12-22:
You need libconfig++ and libconfig++-dev packages to be installed prior to building. See the list of all dependencies in the README, and try to apt install them all before building otherwise you'll have to repeat this step many times.

9a4db, 2019-12-23:
Thank you, 
afer running:
sudo apt-get install gnuradio-dev protobuf-compiler gr-osmosdr gnuradio libvolk1-dev libvolk1-bin libprotobuf17 libprotobuf-dev libopus0 libopus-dev libspeexdsp1 libspeexdsp-dev libpulse0 libpulse-dev libcodec2-0.8.1 libcodec2-dev libasound2 libasound2-dev libjpeg62-turbo libjpeg62-turbo-dev libconfig++9v5 libconfig++-dev qt5-qmake qt5-default qtbase5-dev libqt5core5a libqt5gui5 libqt5network5 libqt5sql5 qtmultimedia5-dev libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5 libftdi1-dev libftdi1 libsndfile1-dev libsndfile1

only missing was: libftdi-dev

After tha,t make started well, with just few warnings
After approx 10 min, ended with:

/usr/bin/ld: QRadioLink.pb.o:(.data.rel.ro+0x26c): more undefined references to `google::protobuf::Message::SpaceUsed() const' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:439: qradiolink] Error 1

Here already have protobuf:
pi@raspberrypi:~ $ sudo apt install libprotobuf-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libprotobuf-dev is already the newest version (3.6.1.3-2+rpi1).
The following package was automatically installed and is no longer required:
  point-rpi
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pi@raspberrypi:~ $ 

Any more suggestion?

kantooon, 2019-12-23:
Hmm, that's a weird one, never seen it before. It seems the linker can't link into libprotobuf-c++. Which branch are you building off? I've made changes lately to that part that may be incompatible with your version of libprotobuf (can you please check and post its version?), so I suggest the following:
 * remove the generated *.pb.h and *.pb.cc files from **ext/** dir
 * clean the build directory
 * check out a stable tag, latest one is 0.8.2-3: `git checkout 0.8.2-3`
 * run the build again

You'll be missing some latest features, but the build may work with that version. Alternatively, you can download a source tarball from the releases page, 0.8.2-1 and above until  you hit 0.8.3-alpha

9a4db, 2019-12-23:
Thank You,
I am looking for basic functionality only... it is LimeNET micro after all :)
Will try advised later today...

Have stable running Soapy, now just need TX/RX application
=>
pi@raspberrypi:~ $ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-ga489f3dc
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libLMS7Support.so (19.04.1-9d19559c)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
pi@raspberrypi:~ $ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
[INFO] Make connection: 'LimeNET-Micro [USB 2.0] 58399D024CE9'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeNET-Micro
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=FT601
  hardware=LimeNET-Micro
  boardSerialNumber=0x9d024ce9
  firmwareVersion=6
  gatewareVersion=1.0
  hardwareVersion=19
  protocolVersion=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: YES
  Sensors: clock_locked, lms7_temp
     * clock_locked (Clock Locked): true
        CGEN clock is locked, good VCO selection.
     * lms7_temp (LMS7 Temperature): 30.980951 C
        The temperature of the LMS7002M in degrees C.
  Registers: BBIC
  Other Settings:
     * SAVE_CONFIG - Save LMS settings to file
       [key=SAVE_CONFIG, type=string]
     * LOAD_CONFIG - Load LMS settings from file
       [key=LOAD_CONFIG, type=string]
[... 68 lines cut ...]
    PAD gain range: [0, 52] dB
    IAMP gain range: [-12, 12] dB
  Full freq range: [0, 3800] MHz
    RF freq range: [30, 3800] MHz
    BB freq range: [-8, 8] MHz
  Tune args:
     * LO Offset - Tune the LO with an offset and compensate with the baseband CORDIC.
       [key=OFFSET, units=Hz, default=0.0, type=float, range=[-8e+06, 8e+06]]
     * BB - Specify a specific value for this component or IGNORE to skip tuning it.
       [key=BB, units=Hz, default=DEFAULT, type=float, range=[-8e+06, 8e+06], options=(DEFAULT, IGNORE)]
  Sample rates: [0.1, 65] MSps
  Filter bandwidths: [5, 40], [50, 130] MHz
  Sensors: lo_locked
     * lo_locked (LO Locked): false
        LO synthesizer is locked, good VCO selection.
  Other Settings:
     * TSP_CONST - Digital DC test signal level in LMS7002M TSP chain.
       [key=TSP_CONST, default=16383, type=int, range=[0, 32767]]
     * CALIBRATE -  DC/IQ calibration bandwidth
       [key=CALIBRATE, type=float, range=[2.5e+06, 1.2e+08]]
     * ENABLE_GFIR_LPF - LPF bandwidth (must be set after sample rate)
       [key=ENABLE_GFIR_LPF, type=float]
     * TSG_NCO - Enable NCO test signal
       [key=TSG_NCO, default=4, type=int, options=(-1, 4, 8)]
pi@raspberrypi:~ $

9a4db, 2019-12-23:
Helo Adrian,
Problem persist:
/usr/bin/ld: radioprotocol.o: in function `RadioProtocol::buildRepeaterInfo()':
radioprotocol.cpp:(.text+0x8a0): undefined reference to `google::protobuf::internal::kEmptyString[abi:cxx11]'
collect2: error: ld returned 1 exit status
make: *** [Makefile:430: qradiolink] Error 1
pi@raspberrypi:~/qradiolink/qradiolink/build $ 

I used qradiolink-0.8.2-3.zip deleted complete directory on pi and repeated:
mkdir -p build
cd ext/
protoc --cpp_out=. Mumble.proto
protoc --cpp_out=. QRadioLink.proto
cd ../build/
qmake ..
make

kantooon, 2019-12-23:
Can you post your version of libprotobuf? I am using libprotobuf17 here, make sure you are not using libprotobuf-lite17.
To completely avoid this, fall back in release versions until you stop getting those errors. There should not be too many tries.

9a4db, 2019-12-23:
pi@raspberrypi:~ $ sudo apt-get install libprotobuf-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libprotobuf-dev is already the newest version (3.6.1.3-2+rpi1).
The following package was automatically installed and is no longer required:
  point-rpi
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pi@raspberrypi:~ $ 

Have this one, have no clue is it good enough :)

9a4db, 2019-12-23:
Tested with:
qradiolink-0.5.2.tar.gz 
no luck on start gsm.h trouble
manage to find only some arduino libs 
and
qradiolink-0.8.1-9
qradiolink-0.8.2-rc1
having the same problem like
qradiolink-0.8.2-3

Guess, now is turn for someone more proficient on RPI 👍

kantooon, 2019-12-23:
I forgot to ask the most basic question: which version of Debian is your OS based on? Debian 10 (current stable) is supported, so if your version of Raspbian is based on Debian 10 then they are doing something strange to libprotobuf and/or protoc.

9a4db, 2019-12-24:
I am using 2019-09-26-raspbian-buster 3,7 GB image,
which have enough space up to 8 GB on CM3 from LimeNET micro package.
Have no clue on which Debian version is based.
Lime and Soapy utility are up and running.

kantooon, 2019-12-30:
> /usr/bin/ld: QRadioLink.pb.o:(.data.rel.ro+0x26c): more undefined references to `google::protobuf::Message::SpaceUsed() const' follow
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:439: qradiolink] Error 1
> libprotobuf-dev is already the newest version (3.6.1.3-2+rpi1).

Sorry for the delay. It seems to me that the code generated by protoc is incompatible with the version of libprotobuf being linked. Can you please post the output of ** protoc --version ** ?
It looks like there is a need for reproducible builds, including on Raspbian, but I don't have enough resources right now to do that. This issue pottentially affects all arm64 platforms.

9a4db, 2019-12-30:
Thank you Adrian, 
For temporary solution I used Soapy Remote and now able to run at last Cubic SDR.
For example SDRangel is crashing on Soapy Remote device selection.
My RPI used for testing is at moment busy feeding FR24 instead of broken F- hardware.
Will let you know when able to compile on RPI again.
Wish you all the best for new 2020. and thank you for the good SW work and efforts.
73 Djani

kantooon, 2020-01-05:
Djani, I don't know about Soapy remote. Theoretically it might work, but in practice it's always harder. Don't expect it to work perfectly though, or even well enough. Had some weeks to test my setup, when confronted with the real deal all this theoretical stuff goes out the window, the reality is my setup does not work well enough at the moment. Things may improve in the future or they may not. Don't rely on it for anything serious, at the moment it's good when you are bored and want to try various things.

9a4db, 2020-01-05:
Dear Adrian, that is good enough....
Really appreciate your efforts.