qradiolink/qradiolink #66: can't run QRadioLink-0.8.3-alpha2-x86_64.AppImage

cn0xroot, 2020-01-21:
OS： ubuntu 16.04
wget appimage from github.

```
./QRadioLink-0.8.3-alpha2-x86_64.AppImage             
/tmp/.mount_QRadioCXODkh/AppRun.wrapped: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.11' not found (required by /tmp/.mount_QRadioCXODkh/usr/bin/../lib/libuhd.so.3.13.1)
/tmp/.mount_QRadioCXODkh/AppRun.wrapped: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /tmp/.mount_QRadioCXODkh/usr/bin/../lib/libuhd.so.3.13.1)
/tmp/.mount_QRadioCXODkh/AppRun.wrapped: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by /tmp/.mount_QRadioCXODkh/usr/bin/../lib/libfreesrp.so.0)

```

also try build it from source:

```
init3:build master 47d ⍉ ➜ pwd              
/home/init3/SDR/qradiolink/build
init3:build master 47d ➜ qmake ..
Project MESSAGE: 
Project MESSAGE: Building without OpenGL support
Project MESSAGE: Building with Qt audio support.
init3:build master 47d ➜ make
g++ -c -m64 -pipe -O2 -D_REENTRANT -Wall -W -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../qradiolink -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtMultimedia -I/usr/include/qt4 -I. -I. -I../../qradiolink -I. -o imagecapture.o ../video/imagecapture.cpp
In file included from ../video/imagecapture.cpp:17:0:
../video/imagecapture.h:21:19: fatal error: QCamera: not such file or directory
compilation terminated.
Makefile:657: recipe for target 'imagecapture.o' failed
make: *** [imagecapture.o] Error 1
```

kantooon, 2020-01-22:
The AppImage failure means that your version of libstdc++ is binary incompatible with the AppImage. Please add some information about your OS version so we can determine if the AppImage can work on it.

The build failure message means that build time dependencies (Qt in this case) are missing. If you cannot build it from source yourself, you could try asking your distribution maintainers to package the application since they might be in a better position to determine which dependencies are required to build the software.

cn0xroot, 2020-01-25:
Thanks for quilk reply.

More info:
`OS:Ubuntu 16.04.6 LTS 64 Bit`

```
strings /usr/lib32/libstdc++.so.6 | grep libstdc
libstdc++.so.6
 ../../../../../../src/libstdc++-v3/src/c++11/debug.cc
libstdc++.so.6.0.21
init3:~ ➜ strings /usr/lib32/libstdc++.so.6 | grep libstdc++
libstdc++.so.6
 ../../../../../../src/libstdc++-v3/src/c++11/debug.cc
libstdc++.so.6.0.21

```

```
qmake -v
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
```

cn0xroot, 2020-01-25:
Ready to update qt to 5.14.
Will try to rebuild qradiolink againn after that. Thx!

closed 2020-01-25 by cn0xroot