qradiolink/qradiolink #67: fatal error: gnuradio/vocoder/freedv_api.h: No such file or directory

doronbehar, 2020-04-05:
**Describe the bug**

I get this error when trying to build 0.8.2-3:

```
./gr/gr_mod_base.h:27:10: fatal error: gnuradio/vocoder/freedv_api.h: No such file or directory
   27 | #include <gnuradio/vocoder/freedv_api.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

**To Reproduce**

I compile my GNUradio with the following components:

```
-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * testing-support
--   * volk
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-atsc
--   * gr-audio
--   * * alsa
--   * * oss
--   * * jack
--   * gr-channels
--   * gr-noaa
--   * gr-pager
--   * gr-trellis
--   * gr-uhd
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-fcd
--   * gr-wavelet
--   * gr-zeromq
--
-- ######################################################
-- # Gnuradio disabled components
-- ######################################################
--   * python-support
--   * doxygen
--   * sphinx
--   * gnuradio-companion
--   * gr-comedi
--   * gr-qtgui
--   * gr-utils
--   * gr-wxgui
--
-- Using install prefix: /nix/store/syg7iqv95ygbwj0x2ibdl750174sqfp1-gnuradio-3.7.14.0
-- Building for version: 3.7.14.0 / 3.7.14.0
```

**Expected behavior**

I'm able to build qradiolink just fine.

**Additional context**

I'm using NixOS and I'm trying to update our qradiolink package. I'm getting this error with  0.8.3-alpha2 as well. Haven't tried to build straight from master yet.

kantooon, 2020-04-08:
Helo @doronbehar 
It seems the gnuradio headers for the FreeDV vocoder component are missing from your system. As far as I know, these headers should be present in all Debian-derived packages, however I can't verify other distributions. An alternative I can suggest is build for Gnuradio 3.8 using the gr3.8_port branch, as Gnuradio 3.8 should contain the FreeDV api everywhere. Alternatively, you could investigate if your distribution needs to alter the packaging for Gnuradio 3.7.14 to include these gr-vocoder headers.

doronbehar, 2020-04-08:
@kantooon I'm the packager of GNURadio 3.7.14 which I tried to build qradiolink against :). Please note that this version is not the old 3.7.13.4 from 2018 which [Debian still maintains](https://packages.debian.org/search?keywords=gnuradio) - see https://www.gnuradio.org/news/2020-02-16-gnu-radio-v3-7-14-0-release/ .

Here's [the full log of the cmake configure phase I got when building 3.7.14](https://gist.github.com/92569ba3a0a391385dc90c94072c58b2).

<details><summary>Here are the files that our build creates in <code>include/gnuradio</code> for 3.7.14</summary>

```
./result/include/gnuradio
├── analog
│   ├── agc2_cc.h
│   ├── agc2_ff.h
│   ├── agc2.h
│   ├── agc3_cc.h
│   ├── agc_cc.h
│   ├── agc_ff.h
│   ├── agc.h
│   ├── api.h
│   ├── cpfsk_bc.h
│   ├── cpm.h
│   ├── ctcss_squelch_ff.h
│   ├── dpll_bb.h
│   ├── fastnoise_source_c.h
│   ├── fastnoise_source_f.h
│   ├── fastnoise_source_i.h
│   ├── fastnoise_source_s.h
│   ├── feedforward_agc_cc.h
│   ├── fmdet_cf.h
│   ├── frequency_modulator_fc.h
│   ├── noise_source_c.h
│   ├── noise_source_f.h
│   ├── noise_source_i.h
│   ├── noise_source_s.h
│   ├── noise_type.h
│   ├── phase_modulator_fc.h
│   ├── pll_carriertracking_cc.h
│   ├── pll_freqdet_cf.h
│   ├── pll_refout_cc.h
│   ├── probe_avg_mag_sqrd_cf.h
│   ├── probe_avg_mag_sqrd_c.h
│   ├── probe_avg_mag_sqrd_f.h
│   ├── pwr_squelch_cc.h
│   ├── pwr_squelch_ff.h
│   ├── quadrature_demod_cf.h
│   ├── rail_ff.h
│   ├── random_uniform_source_b.h
│   ├── random_uniform_source_i.h
│   ├── random_uniform_source_s.h
│   ├── sig_source_c.h
│   ├── sig_source_f.h
│   ├── sig_source_i.h
│   ├── sig_source_s.h
│   ├── sig_source_waveform.h
│   ├── simple_squelch_cc.h
│   ├── squelch_base_cc.h
│   └── squelch_base_ff.h
├── api.h
├── atsc
│   ├── api.h
│   ├── basic_trellis_encoder_impl.h
│   ├── bit_timing_loop.h
│   ├── consts.h
│   ├── convolutional_interleaver.h
│   ├── create_atsci_equalizer.h
│   ├── create_atsci_fs_checker.h
│   ├── create_atsci_fs_correlator.h
│   ├── data_interleaver_impl.h
│   ├── deinterleaver.h
│   ├── depad.h
│   ├── derandomizer.h
│   ├── diag_output_impl.h
[... 1903 lines cut ...]
│   ├── ulaw_decode_bs.h
│   └── ulaw_encode_sb.h
├── wavelet
│   ├── api.h
│   ├── squash_ff.h
│   ├── wavelet_ff.h
│   └── wvps_ff.h
└── wxgui
    ├── api.h
    ├── histo_sink_f.h
    ├── oscope_guts.h
    ├── oscope_sink_f.h
    ├── oscope_sink_x.h
    └── trigger_mode.h

23 directories, 952 files
````

</details>

[Here's a link to the build log of 3.7.13.4](https://hydra.nixos.org/build/116170235/nixlog/1)

As you can see, there is no mention of a `freedv_api.h` file being installed while almost all components gnuradio has to suggest, are compiled. If you could find a build log of a distribution that ships this file, I'd be most obliged to further investigate that.

Regards.

kantooon, 2020-04-08:
ok, I'll try to take a look tonight at the logs.

kantooon, 2020-04-08:
Alright, I checked the Gnuradio repo and 3.7.14 source does not have these headers in CMakeLists:  https://github.com/gnuradio/gnuradio/blob/v3.7.14.0/gr-vocoder/include/gnuradio/vocoder/CMakeLists.txt
While 3.8.0 has them: https://github.com/gnuradio/gnuradio/blob/v3.8.0.0/gr-vocoder/include/gnuradio/vocoder/CMakeLists.txt
Since A. Maitland Bottoms wrote this code and maintains it in Debian, I imagine this was not backported in some other distributions not derived from Debian.
I think your options here are:
1. upgrade to Gnuradio 3.8 and use the gr3.8_port branch of this repo which is level with master, or 
2. use the Debian Buster 3.7.14 source patches which contain the FreeDV API to build your distro's Gnuradio.

Radiomix2000, 2020-10-04:
Don't know is this info usefull. But in past when I compiled (on x86 platform) qradiolink with GR 3.7.13.4 compiled from sources or installed using apt get, there was no problem with freedv headers. Later, , chronologically, some time after gr 3.8.1 release, I was not able to compile qradiolink with gr 3.7.13.4. I had the same error like issue authtor had. What is interesting, when I make the same steps on my Pi4 (gr 3.7.13.4 installed, and last version of qradiolink), everything works fine. It's not clear,it seems like problem in some gr repository, where is something changed. Generally very strange story for me..

doronbehar, 2020-10-04:
It's because different distributions build GR differently, unfortunately.

kantooon, 2020-10-04:
In the following weeks I hope to finally make 3.8 the default GNU radio version used, and then perhaps some distro specific patches might be enough to solve this issue with FreeDV.

kantooon, 2020-10-06:
The gr_3.8_port branch is now fully up to date with the latest changes in master and next, and some dependencies were added to enable building it on Debian 11 Bullseye. Some time in the next weeks, the gr 3.7 code will be frozen for maintenance and release branches and packages will be switched to GNU radio 3.8.

Radiomix2000, 2020-10-08:
Good news! So it means you do not plan to continue develop gr 3.7 branch and will archive it?

> and then perhaps some distro specific patches might be enough to solve this issue with FreeDV.
Can you point, what this patches are, and where to get them?  I want to build Qradiolink on ubuntu 18.04 machine with gnuradio 3.7 installed, but have this freedv_700C error. 
Again, this is very strange, because some time ago I was able to compile qradiolink 0.8.0.3 with gnuradio 3.7 installed from ubuntu packages repository (apt get install gnuradio(-dev)). And now it's not possible, like somebody changed some dependencies, or some package version in repository. I'm totally frustrated. The only thing is to use the patch you spoke about in prev. message.

kantooon, 2020-10-08:
Most of the work will now focus on making it compatible with GR 3.8 (lots of issues there) and hopefully Ubuntu based distros. The gr 3.7 master branch is pretty stable so I don't envision changes for it. I  would recommend the 3.7 version to anybody who wants something stable and working. 
Regarding Ubuntu, I'd have to set up VMs to try and troubleshoot the build, but it can't happen until I get 3.8 working at least on Debian.

kantooon, 2020-10-08:
Regarding your frustration, I can understand that. After upgrading to Debian 11, nothing works for me as it used to do anymore. The PlutoSDR is not always detected correctly, Soapy is unable to create the sink and source buffers, crashes all around. My first reaction would be to just drop this completely, stick with Debian Buster and forget about the GR 3.8 upgrade. However, I know better that eventually I'll return to digging in and eventually fix the massive breakage in a year or so.

doronbehar, 2020-10-08:
@kantooon would you recommend packagers of gnuradio3.7 to ship it with the FreeDV patches? Also, I'm a little bit confused as for the upcoming 3.8 changes here in qradiolink - does mainline gnuradio3.8 need FreeDV patches as well?

kantooon, 2020-10-09:
@doronbehar I guess I'm trying to say that this situation (missing freedv_api, missing various gnuradio modules) is best addressed by distro maintainers locally by applying patches to disable certain functionality like FreeDV if needed.
For GR 3.7 I knoiw for certain that patches are not needed for Debian and that they might be needed on Ubuntu and derived distros because of FreeDV. For GR 3.8, it is stll unclear, because until this week the gr_3.8_port branch was not actively developed, so it is possible certain changes made this week rendered patches unnecessary on Ubuntu. I think I'd need more reports regarding what works and what doesn't. The Debian 11 Bullseye build is working as expected, except for a couple of issues which I try to document here and which so far are not blocking a release.
One final thing: mainline gnuradio 3.8 does not require any patches. Whether built from source or provided by Debian, qradiolink builds cleanly with it.

doronbehar, 2020-10-09:
In https://github.com/NixOS/nixpkgs/pull/99685 we are rewriting the GNUradio packages on NixOS. It seems that 3.8.2.0 doesn't include any freedv headers as well:

[image]

However, it seems that there _is_ something about freedv there according to this:

https://www.gnuradio.org/doc/doxygen/classgr_1_1vocoder_1_1freedv__tx__ss.html

But [this list of files](https://packages.debian.org/sid/amd64/gnuradio/filelist) Still shows no freedv headers.

And on a look at the repo, the files are supposed to be installed:

https://github.com/gnuradio/gnuradio/blob/v3.8.2.0/gr-vocoder/include/gnuradio/vocoder/freedv_api.h

I think the files are not installed, because of this:

https://github.com/gnuradio/gnuradio/blob/67d8415f4d635c186fdf03a7a1d18b84d666a0c5/gr-vocoder/CMakeLists.txt#L55-L58

So it's a matter of how [codec2](github.com/drowe67/codec2) is built, and it'd have been clearer if gnuradio would have uncommented those cmake status messages.

This needs further investigation and then it should all be written to the README because it's super untrivial, even for the experienced Distro packager. I'm afraid that no distro (at least debian doesn't) prove to provide freedv api for codec2 and then gnuradio 3.8.

doronbehar, 2020-10-09:
The problem is that codec2 freedv support was made after the latest release of it:

https://github.com/drowe67/codec2#freedv-2020-support-building-with-lpcnet

& https://github.com/drowe67/codec2/blob/master/README_freedv.md vs https://github.com/drowe67/codec2/blob/v0.9.2/README_freedv.md

kantooon, 2020-10-09:
Check if there's any cmake switch for GR to enable codec2 API (Freedv). But my guess is that cmake does not detect a compatible libcodec2 so maybe try to build gnuradio after installing libcodec2 0.9+ and libcodec2-dev.
I'd look myself but busy now.

doronbehar, 2020-10-09:
I'm rather positive that codec2 support is enabled on gnuradio if codec2 is present during build but the Freedv api is just unreleased yet on codec2's side, causing all the confusion.

doronbehar, 2020-10-09:
And if gnuradio detects if codec2 is built with freedv or not, and installs it's api accordiingly if needed.

kantooon, 2020-10-09:
Just to clarify: libcodec2 contains both codec2 and freedv API, and the freedv API is released since at least 6 years. The freedv LPCNet support might not be released yet, but qradiolink does not use LPCNet yet anyway so libcodec2 0.9+  should be fine.
Just run a quick cmake and check the output for codec2/freedv entries. I haven't built gnuradio from source in the last two years so I'm not sure what cmake does.

doronbehar, 2020-10-09:
Damn you are probably right. The build of codec2 on NixOS produces:

```
result/include
└── codec2
    ├── codec2_cohpsk.h
    ├── codec2_fdmdv.h
    ├── codec2_fifo.h
    ├── codec2_fm.h
    ├── codec2.h
    ├── codec2_ofdm.h
    ├── comp.h
    ├── comp_prim.h
    ├── filter.h
    ├── freedv_api.h
    ├── freedv_api_internal.h
    ├── fsk.h
    ├── golay23.h
    ├── horus_api.h
    ├── kiss_fft.h
    ├── modem_stats.h
    ├── varicode.h
    └── version.h

1 directory, 18 files
``

So `freedv_api.h` is there! But the Gnuradio 3.8 build fails to detect it - somewhere around here:

https://github.com/gnuradio/gnuradio/blob/67d8415f4d635c186fdf03a7a1d18b84d666a0c5/gr-vocoder/CMakeLists.txt#L55-L58

kantooon, 2020-10-09:
Check if subcomponent codec2 is enabled but subcomponent freedv is not. If former, find_packages might fail there... Pkgconfig maybe? Could be worth asking on GR mailing list or IRC if it turns out to be something else.

doronbehar, 2020-10-09:
I've finally nailed it: Some cmake flags helped:

```nix
      "-DLIBCODEC2_LIBRARIES=${codec2}/lib/libcodec2.so"
      "-DLIBCODEC2_INCLUDE_DIRS=${codec2}/include"
      "-DLIBCODEC2_HAS_FREEDV_API=ON"
      "-DLIBGSM_LIBRARIES=${gsm}/lib/libgsm.so"
      "-DLIBGSM_INCLUDE_DIRS=${gsm}/include/gsm"
```

Shown in Nix syntax - `${gsm}` is where `gsm` is "installed", and `${codec2}` is where codec2 is.

With these flags, GR3.8 finally installs `freedv_api.h` for me, I wonder what's wrong with the way they try to detect it, but I guess it should be fine with the upcoming qradiolink release, that will supports 3.8.

Reminder to other packagers: GR3.7 source code doesn't include these header files at all, so don't expect them to be installed. It's possible that a certain debian build included some backported patches that added freedv support for GR3.7, but I haven't found evidence for this, or for a GR3.7 in any distro, that installs GR's `freedv_api.h`.

kantooon, 2020-10-09:
Good, glad you managed to gently coax it into doing the right thing. Hopefully Codec2/FreeDV was the only thing missing for you. When you have time, give the gr_3.8_port branch a shot and let me know if it builds. I'm not saying it's ready to release yet, there are heaps of Qt deprecation warnings that I have to fix and some bugs, but at least I'll be able to close this issue as solved.

Radiomix2000, 2020-10-10:
> It's possible that a certain debian build included some backported patches that added freedv support for GR3.7

For sure. I installed Debian 10, then apt install gnuradio gnuradio-dev. This installs 
```
gnuradio-companion --version
GNU Radio Companion 3.7.13.4
```
Then it really has necessary files.
[image]

What I found:
[gnuradio_3.7.13.4-4.debian.tar.xz] from https://packages.debian.org/buster/gnuradio is patch collection.
Inside archive in patches directory is "add-freedv-blocks" and some other interesting patches for Debian. This is ok and clear. I just compiled Qradiolink in Debian 10 without any problems (GR 3.7.13.4 installed via apt-get). But how to explain, that 9 months ago it was possible to compile Qradiolink under Ubuntu 18.04 with GR 3.7.13.4(apt-get, not sources), and it's not possible, cause freedv API error. I'm still using GR 3.7 and don't want to upgrade to 3.8, because huge amount of OOT blocks just doesn't work under GR3.8 Everything is broken with this GR3.8. Damn! Nobody care about backward compatibility.

kantooon, 2020-10-10:
The developer who created the FreeDV blocks in gnuradio is also a Debian developer, so it's natural for Debian to have first class support.
The strategy now is to keep branch `gr3.7_maint` for any GNU radio 3.7 fixes and development if someone feels so inclined. The `master` branch will remain at the last release which is based on GR 3.7. The `next` branch will be all new development based on GR 3.8. It is pointless right now to do a release based on GR 3.8 since there are too many issues to solve. 
If time allows I'll configure a new Travis jobs to do bulds on GR 3.7 as well as 3.8, right now it's been switched to 3.8 and 3.7 based builds will fail.
Also, Ubuntu maintainers of GR should probably be contacted and let know that GR is missing some modules if you encounter this issue. This is probably not the only application facing the problem.

doronbehar, 2020-10-13:
`gr_3.8-port` works fine for me, and the GUI seems functional :tada:. Here's a proper fix for this issue, IMO: https://github.com/qradiolink/qradiolink/pull/78 .

PR #78 (2020-10-13): README: Explain better the need for support for gr-vocoder
Fixes #67.
commit 8f918ed1 (2020-10-13): README: Explain better the need for support for gr-vocoder (#78)

Fixes #67.
closed 2020-10-13 by kantooon