qradiolink/qradiolink #68: USRP1 on Qradio link

nyjilgeorge1, 2020-04-22:
Dear Qradiolink Team, 

Thank you very much for this wonderful creation.
I am a USRP1 with basic TX/RX board user. as we know USRP1 connected through USB2.0.
I would like to request that, Guide me to configure USRP1 to Qradiolink.

Thanks in Advance.
Nyjil George
VU3ZNG

kantooon, 2020-04-25:
George, if you can get GNU radio working with the USRP1, (and there is plenty of information on http://www.gnuradio.org ) then you can get qradiolink to work as well since qradiolink uses GNU radio for all DSP and connection to hardware. There is nothing specific about any hardware in qradiolink. Check the user guides for GNU radio and especially gr-osmosdr: https://osmocom.org/projects/gr-osmosdr/wiki

kantooon, 2020-08-01:
No feedback, closing.

closed 2020-08-01 by kantooon