qradiolink/qradiolink #70: 20.04 + gnuradio 3.8 support

alphafox02, 2020-07-27:
Now that 20.04 has gr-osmosdr included and working, would you consider a 3.8 port?

kantooon, 2020-07-28:
Check out the gr_3.8_port branch. It should be working, thanks to @devnulling

alphafox02, 2020-07-29:
Thank you. I gave it a quick try on 20.04 but the build part resulted in some failures. I’ll document it all and get you some useful feedback ASAP.

kantooon, 2020-07-30:
Many thanks. I can't do much right now about GR 3.8 since Debian 10 is still on 3.7. But I can try to extend the Travis job to use Debian testing specifically for 3.8. It would be good to know of any issue in advance.
Out of curiosity, what is this 20.04 you speak of?

alphafox02, 2020-07-30:
I should’ve been more specific, I was referring to Lubuntu/Ubuntu 20.04 LTS. I work on a project called DragonOS with the goal of making it super easy for others to get up and running with SDRs as easy as possible. I basically preinstall and test everything, then turn it into an installation ISO.

kantooon, 2020-07-30:
Ok, understood, very cool and good luck with the project.

alphafox02, 2020-07-30:
Here's the results I'm running in to at first go with Lubuntu 20.04 and the 3.8 branch. I tried using similar packages as what you call for when using Debian. Also had to install a package to get further into the make process, but no success yet. 

`make -j4
/usr/lib/qt5/bin/uic ../mainwindow.ui -o ui_mainwindow.h
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink-gr_3.8_port -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o imagecapture.o ../video/imagecapture.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink-gr_3.8_port -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o ../main.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink-gr_3.8_port -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o audioencoder.o ../audio/audioencoder.cpp
g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink-gr_3.8_port -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o audioprocessor.o ../audio/audioprocessor.cpp
In file included from ../src/radiocontroller.h:40,
                 from ../main.cpp:33:
../src/relaycontroller.h:21:10: fatal error: ftdi.h: No such file or directory
   21 | #include <ftdi.h>
      |          ^~~~~~~~
compilation terminated.
make: *** [Makefile:1132: main.o] Error 1
make: *** Waiting for unfinished jobs....
../video/imagecapture.cpp: In member function ‘void ImageCapture::process_image(int, QImage)’:
../video/imagecapture.cpp:78:38: warning: unused parameter ‘id’ [-Wunused-parameter]
   78 | void ImageCapture::process_image(int id, QImage img)
      |                                  ~~~~^~
dragon@dragon:~/Downloads/qradiolink-gr_3.8_port/build$ sudo apt-get install libftdi1-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libftdi1-dev is already the newest version (1.4-2build2).
0 upgraded, 0 newly installed, 0 to remove and 102 not upgraded.
dragon@dragon:~/Downloads/qradiolink-gr_3.8_port/build$ sudo apt-get install libftdi1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libftdi1 is already the newest version (0.20-4build8).
0 upgraded, 0 newly installed, 0 to remove and 102 not upgraded.
dragon@dragon:~/Downloads/qradiolink-gr_3.8_port/build$ sudo apt-get install libftdi libftdi-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libftdi
dragon@dragon:~/Downloads/qradiolink-gr_3.8_port/build$ sudo apt-get install libftdi-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libftdi-dev
0 upgraded, 1 newly installed, 0 to remove and 102 not upgraded.
Need to get 178 kB of archives.
After this operation, 2,143 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu focal/universe amd64 libftdi-dev amd64 0.20-4build8 [178 kB]
Fetched 178 kB in 1s (152 kB/s)      
Selecting previously unselected package libftdi-dev.
(Reading database ... 333936 files and directories currently installed.)
Preparing to unpack .../libftdi-dev_0.20-4build8_amd64.deb ...
Unpacking libftdi-dev (0.20-4build8) ...
Setting up libftdi-dev (0.20-4build8) ...
Processing triggers for man-db (2.9.1-1) ...
dragon@dragon:~/Downloads/qradiolink-gr_3.8_port/build$ make -j4g++ -c -pipe -O2 -D_REENTRANT -Wall -W -fPIC -DUSE_QT_AUDIO -DQT_NO_DEBUG -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink-gr_3.8_port -I. -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/ [... cut ...]
  152 | <typename I,   // I models InputIterator
      |           ^
/usr/include/boost/move/algorithm.hpp:152:11: error: expected ‘)’ before ‘__extension__’
  152 | <typename I,   // I models InputIterator
      |           ^
/usr/include/boost/move/algorithm.hpp:152:11: error: expected ‘>’ before ‘__extension__’
/usr/include/boost/move/algorithm.hpp:152:11: error: expected unqualified-id before ‘)’ token
  152 | <typename I,   // I models InputIterator
      |           ^
make: *** [Makefile:1296: radiocontroller.o] Error 1
`

kantooon, 2020-07-31:
Alright, so let's see what we have here. So Ubuntu has both libftdi-dev and libftdi1-dev. But the second one has a header with a different name. Kind of confusing, but you fixed it.
Then, the old, very old trouble with /usr/include/gnuradio/vocoder/freedv_api.h and /usr/include/codec2/freedv_api.h which bombs out here in quite the ugly way.
See, Debian's FreeDV in GNU radio module is developed and maintained by Maitland Bottoms. So the specific Debian version of libcodec2 works ok with the specific Debian GNU-radio package because someone keeps them synced and compatible. Looks like Ubuntu does not take the Debian packages verbatim, or otherwise something is wrong in Debian testing with these that I'm not aware of (since I run stable). This is a difficult one, and unless the same packages are maintained in Ubuntu by Maitland, it's going to require work in Ubuntu's GNU-radio code. To be continued.

alphafox02, 2020-08-04:
It’s always something with gnuradio haha. I’m hoping I don’t have to change the packaged version. I’ll give this a try on Debian 11 and see if I get the same result.

kantooon, 2020-08-05:
Try the app image first, it should work on any recent enough distribution. But I agree it's desirable to be able to build from source.

alphafox02, 2020-08-05:
Trying the appimage on Lubuntu 20.04, although I must admit it's not a bone stock installation. Also trying to learn how to debug as I got a segmentation fault as soon as I tried to run it. 

(gdb) run
Starting program: /home/dragon/Downloads/QRadioLink-0.8.3-1-x86_64.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 6793]
process 6789 is executing new program: /usr/bin/env
process 6789 is executing new program: /usr/bin/bash
[Detaching after fork from child process 6797]
process 6789 is executing new program: /tmp/.mount_QRadiocuCsFz/usr/bin/qradiolink
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x555500000001) at malloc.c:3102
3102    malloc.c: No such file or directory.
(gdb)

alphafox02, 2020-08-05:
AppImage works okay as I'm sure you're already aware on Buster and Bullseye.

kantooon, 2020-08-05:
Do a bt (backtrace) on that, see if it returns anything interesting. You can ignore the message about malloc.c missing, it's trying to find the source code.

alphafox02, 2020-08-05:
(gdb) bt
#0  __GI___libc_free (mem=0x555500000001) at malloc.c:3102
#1  0x00007ffff50cdb0f in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /lib/x86_64-linux-gnu/libstdc++.so.6         
#2  0x00007ffff4b611c7 in boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*) ()                                                                          
   from /tmp/.mount_QRadioKAZgbt/usr/bin/../lib/libboost_filesystem.so.1.67.0
#3  0x00007fffee2a20ec in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#4  0x00007fffee2a21c0 in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#5  0x00007fffee2afe4e in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0                                  
#6  0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2
#7  0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2
#8  0x00007ffff4d90785 in __GI__dl_catch_exception (exception=<optimized out>, 
    operate=<optimized out>, args=<optimized out>) at dl-error-skeleton.c:182
#9  0x00007ffff7fe50bf in ?? () from /lib64/ld-linux-x86-64.so.2
#10 0x00007ffff4d90728 in __GI__dl_catch_exception (exception=<optimized out>, 
    operate=<optimized out>, args=<optimized out>) at dl-error-skeleton.c:208
#11 0x00007ffff7fe45fa in ?? () from /lib64/ld-linux-x86-64.so.2
#12 0x00007ffff2cbe34c in dlopen_doit (a=a@entry=0x7fffffffdb00) at dlopen.c:66
#13 0x00007ffff4d90728 in __GI__dl_catch_exception (
    exception=exception@entry=0x7fffffffdaa0, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:208
#14 0x00007ffff4d907f3 in __GI__dl_catch_error (objname=0x55555575c360, 
    errstring=0x55555575c368, mallocedp=0x55555575c358, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:227
#15 0x00007ffff2cbeb59 in _dlerror_run (
    operate=operate@entry=0x7ffff2cbe2f0 <dlopen_doit>, 
    args=args@entry=0x7fffffffdb00) at dlerror.c:170
#16 0x00007ffff2cbe3da in __dlopen (file=<optimized out>, mode=<optimized out>)
    at dlopen.c:87
#17 0x00007ffff3f19110 in ?? ()
--Type <RET> for more, q to quit, c to continue without paging--
  .3.13.1                                                                                                                                                                                    
#18 0x00007ffff3f191f9 in ?? () from /tmp/.mount_QRadioKAZgbt/usr/bin/../lib/libuhd.so.3.13.1                                                                                                
#19 0x00007ffff3f193c8 in ?? () from /tmp/.mount_QRadioKAZgbt/usr/bin/../lib/libuhd.so.3.13.1                                                                                                
#20 0x00007ffff3f26d09 in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) () from /tmp/.mount_QRadioKAZgbt/usr/bin/../lib/libuhd.so.3.13.1                                 
#21 0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2                                                                                                                             
#22 0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2                                                                                                                             
#23 0x00007ffff7fd013a in ?? () from /lib64/ld-linux-x86-64.so.2                                                                                                                             
#24 0x0000000000000001 in ?? ()                                                                                                                                                              
#25 0x00007fffffffe2a3 in ?? ()                                                                                                                                                              
#26 0x0000000000000000 in ?? ()

alphafox02, 2020-08-05:
Could it be something with libuhd versions? I see where it seems to find what I have installed, but then later references an older version in the appimage.

alphafox02, 2020-08-05:
Just did apt-get remove libuhd3.15.0 which in turn removed about 400Mb of stuff, Gnuradio etc... but hey now the appimage runs!

alphafox02, 2020-08-05:
Don't suppose a special requested appimage without UHD included is possible :) if in fact that's even the problem. Working great now though and that's interesting to see how Gnuradio 3.7 is all contained within the appimage or so it seems.

alphafox02, 2020-08-05:
Well, I actually did apt-get install libuhd3.15.0, which only installed that specific thing. Appimage still works, so it must be something else that was installed that it didn't like. Now I'll go see if i can install things back until it no longer works.

alphafox02, 2020-08-05:
Started back at step one using the ISO I’ve worked on. Pulled down the app image, tried to run, experienced same segmentation fault. I then have to go remove what I mentioned and put it back. No idea what’s causing this. Sorry for the bother - might just be something on my end.

kantooon, 2020-08-06:
Hmm, ok. Good investigation.
UHD is a problem package, because it comes with firmware, and I can't include the firmware in the image. Dropping just UHD from it won't work that well either, because then gr-osmosdr might be linked against the wrong version and try to find some symbols in libuhd that changed.
One possibility would be to drop gr-osmosdr completely from the app image and rely in the system one. Perhaps this app-image idea is not so good after all.

alphafox02, 2020-08-06:
What is strange though is that I tried the appimage on Debian Bullseye, even added the uhd-soapysdr package and it worked. Now I need to go investigate more, because I seem to recall Debian Bullseye being the newer uhd+gnuradio and that I already had the firmware downloaded. What you're saying makes sense though.

kantooon, 2020-08-15:
@alphafox02 I've looked again at your build error, and I think I can make an educated guess: can you verify that your libcodec2 header file /usr/include/codec2/codec2_ofdm.h contains this code at the beginning?

    #ifdef __cplusplus
    extern "C" {
    #endif

This is not the first time I see this sort of stuff, never in Debian, always in other distros (Fedora for instance). I don't know why this slips through so often. When shipping a C library, the headers should be made compatible with g++ by the maintainer if upstream doesn't do it.
You probably won't like what I'm about to say, but I advise you base your distro on Debian instead of Ubuntu :)

alphafox02, 2020-08-24:
Sorry it took so long - you were right.

/* Defines */

#ifdef __cplusplus
extern "C" {
#endif

I'd love to switch back to Debian, but I have too much time into Ubuntu and the installer I'm working with can do UEFI/Encrypted. I can't do that when trying to remaster Debian (yet).

alphafox02, 2020-08-24:
Can I somehow use Debian based packages to solve the issue? Or if you happen to build an app image that has UHD 3.15 or maybe no gr-osmosdr. I'll test either way.

kantooon, 2020-10-09:
@alphafox02 just a heads-up: the issue with freedv_api.h that was originally blocking your build has been solved (workaround) in the gr_3.8_port branch. This branch will very soon merged into next and will become the base for new releases. You can now try the source build again.

alphafox02, 2020-10-09:
This is excellent news. I’m going to try it this weekend. I actually just had my usrp radio fixed and I have others to test to.

alphafox02, 2020-10-10:
I was able to build and run it last night. I was following the directions and was so tired that at first I was going in and pulling master with git and wondering why it wasn’t building.. then I realized duh it’s now the 3.8 one. 

It runs with the rtlsdr and since I’m new to this I wasn’t sure if I was doing something wrong, but I tried turning on rx for fm and didn’t seem to find any broadcasts. I’ll try it again soon. Thanks for fixing this.

closed 2020-10-10 by alphafox02

kantooon, 2020-10-10:
GR 3.8 is now in the `next` branch which will now be the active dev branch.

alphafox02, 2020-10-18:
QRadioLink is now Included in DragonOS as shown here https://youtu.be/UKQu6B4Ofqc

Still have a lot of learning to do about the program. Not really sure when the webcam video is supposed to work or be used, any tips on video? I’ve looked over the support page and besides a short mention of how to get the webcam detected I don’t really see where it’s used. 

Good to see it all working though on 20.04.

kantooon, 2020-10-18:
Nice. Congrats on getting it working. The webcam is used for transmitting a digital video signal, which you can also receive at the same time if you use the same video mode on the receiver and use duplex mode with a zero transmit shift. You need to enable both the transmitter and the receiver to get this working. Make sure to use a bit higher transmit gain since the video signal is quite wide (around 350 kHz).
The camera device is usually `/dev/video0` but can also be some other device, or you can have have multiple webcams so you can configure `/dev/video1` instead in the settings, depending on what works for the system.
There is no audio on the video signal right now, and there's a small bug where the Costas loop won't lock on the right sample offset every time (you can monitor the signal quality in the constellation display, normally all symbols should be in the corners of the square, the more apart they are, the more noisy is the signal). I guess I should expand a bit in the docs, but you know, time.

kantooon, 2020-10-18:
Oh, by the way. I have now a release candidate in the releases, 0.8.5-rc1. It fixes some issues like some memory leaks which appeared after the Qt 5.14 upgrade.

alphafox02, 2020-10-18:
Now it makes sense. I thought I was pushing video over IP like Skype or something haha. I’ll go back and try again and also try the updates you mention. You know one thing that was weird, whenever I hit save on settings or maybe I didn’t, can’t recall, but anyways if I went to the spectrum and hit Rx the rtlsdr wouldn’t really see anything. 

That’s when I went back to settings and saw both ppm settings were like 10k+ some random number. Not sure why that happens. I set both setting back to zero, hit save, went back to spectrum and hit Rx. All was fine after that. I’ve replicated it a couple times but not sure exactly what I was doing caused it.

kantooon, 2020-10-19:
I've never seen that happening, but it's possible there is a bug somewhere and I'll see if I can manage to reproduce it here. Do you see that only with the RTL device or other devices as well?

alphafox02, 2020-10-19:
I’ll try and document it better. I had it happen while trying to activate a b205mini mini uhd as well. I also figured out where on the tx side you pick video 256k but every time I pick it the program exits out. I’ll document that to, something about failed to access (assuming the video). I’ve tried the built in video webcam and an external which all show up under /dev and with v4l2-ctl.

kantooon, 2020-10-19:
Cool, thanks. I assume your user needs permissions to access the device under `/dev/video`. There's usually a video group which your user needs to be a part of e.g. `video:x:44:adrian`

kantooon, 2020-10-21:
You were right, it's a bug in my code dealing with frequency correction. Serves me right for removing the gr-osmosdr code which handled that. I've opened #79

Radiomix2000, 2020-10-22:
Just for information. Yesterday installed fresh Ubuntu 20.04 and Qradiolink from source, gr_3.8 branch. So can confirm, it starts on Ubuntu 20.04 without issues. Now will proceed to further testing. 👍

alphafox02, 2020-10-24:
I thought I saw a way to directly contact you. I’m putting finishing touches on a classroom and want to have QRadioLink on all the desktops along with a central mumble  server. Just looking for advice on setup and practical exercises for QRadioLink. 

If the person instructing is licensed to transmit, what would you recommend he or she demonstrate? 

For my own understanding, if I receive with the rtlsdr and pass it over the VoIP server and then on the other end have QRadioLink link and ettus I should then be able to rebroadcast what’s being received via the voip server? 

Sorry for asking here. If there’s a better place I’ll move the discussion there.

kantooon, 2020-10-24:
There's also a mailing list at Sourceforge, although it does not see much traffic: https://sourceforge.net/projects/qradiolink/lists/qradiolink-devel
If it's something really private, you can also find my email in the git log, I don't put it up to avoid spam.

Regarding your use case, it will probably work like you describe. Depending on your country's legislation, you might not need a license for ISM band usage as long as you keep the power very low, below 10 mW (which is huge for a single room demo btw... I regularly made 3km+ QSOs on 70 cm with 10 mW). But check your country's legislation first. 

More commonly, this is how I envisaged its operation: have a bunch of SDRs as local nodes/repeaters, best if full duplex like the USRP or Pluto or LimeSDR, all connected to a central umurmur server. If used in simplex mode (duplex disabled), if one user in the coverage area of one SDR keys up, audio goes to all other nodes and is rebroadcasted, during which time other users can only listen, like in a simplex QSO. If used in the full duplex mode, it will operate like a networked repeater, in that it will also transmit simultaneously in the coverage area of the repeater. This is very similar in concept with svxlink, we have a nation-wide FM repeater network based on it, except it works with normal analog FM radios and doesn't use the Mumble protocol and doesn't have additional non-radio clients like Mumble has. In addition, it supports digital transmissions as well as analog modes, maybe mixed modes, many different server rooms which carry different traffic, like a sort of D-Star on drugs. If you're wondering if this system can scale to national coverage levels, it is yet unknown to me as we haven't yet begun large scale testing. I'd stick with small regional networks at first. This kind of operation would be reserved for licensed amateur radio operators of course.
To make full use of such facilities I recommend getting an amateur radio license if you don't have one, and it will allow you to experiment at will in our bands.

73,

alphafox02, 2020-10-24:
Ill reach out, completely understand about spam. The gears are really turning after reading your response. I want to replicate what you’re saying. I think I understand how it would work. If all remote nodes/SDRs have a network connection to the server that is where the keyed up broadcast info is coming from and then in turn being broadcast locally via the local sdr/qradiolink? Maybe best I save further questions for email. I’ll also attempt this setup on my end ASAP.