qradiolink/qradiolink #71: Does not run on Ubuntu xenial

probonopd, 2020-08-01:
**Describe the bug**
Does not run on Ubuntu xenial because it was compiled on a too new system

**To Reproduce**
Steps to reproduce the behavior:
1. Run Ubuntu xenial
2. Try to run the AppImage from https://github.com/qradiolink/qradiolink

**Expected behavior**
Works

**Screenshots**

```
/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /run/firejail/appimage/.appimage-9205/AppRun.wrapped)

/lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.27' not found (required by /run/firejail/appimage/.appimage-9205/usr/bin/../lib/libvolk.so.1.4)
```

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: xenial

**Additional context**

Reference:
https://github.com/AppImage/appimage.github.io/pull/2172

kantooon, 2020-08-01:
There may be workarounds for this, such as using an older glibc.
Ubuntu Xenial does not provide the libraries this project requires. The build system will never support it.

probonopd, 2020-08-02:
So you are saying the libraries this project requires cannot be built on xenial?

kantooon, 2020-08-02:
I wouldn't know for sure since I never used Ubuntu, but it might be a huge task. Among others, FreeDV support requires at least gnuradio 3.7.13 (the list of deps for gnuradio is huge and a recent libvolk is required as well), some features used from Qt 5.5+, libopus might be problematic, libprotobuf and protoc would definitely be a problem.
To a Gentoo user I'd say yeah go ahead and build everything from source, to a Ubuntu user, I don't know if it's feasible. Even Centos7 is a problem because the gnuradio packages they provide are broken (missing libgnuradio-fec)

probonopd, 2020-08-02:
I see. In this case it might be easier to just bundle _everything_ privately (like the `-s deploy` option in my experimental https://github.com/probonopd/go-appimage/tree/master/src/appimagetool can do) if one wants the AppImage to be able to run on older systems as well.

kantooon, 2020-08-02:
That would be a great idea, thanks a lot for letting me know of this option. I'll see about adjusting the Docker build at the next release.