qradiolink/qradiolink #72: QT5 is requested but QT4 is used?

waterwin, 2020-09-05:
**Describe the bug**
I am following install and build instructions on a Ubuntu 1804 virtual box which has GNU Radio installed.

I get this error message:
qmake: could not exec '/usr/lib/x86_64-linux-gnu/qt4/bin/qmake': No such file or directory

Why is it trying to use Qt4 when Qt5.2 + is a pre requisite?

kantooon, 2020-09-06:
The packages which you called dev versions are actually packages containing the headers for the libraries used. They are needed to be included so that the code knows which prototype functions to use and link to in the library.
You did manage to get further than other people using Ubuntu. If you do manage to build it in the end, we can make this an install guide for Ubuntu because I don't use it and I'm not familiar with its innards.

Regarding the last error you got, this seems to be because you are using a Qt5 version below Qt 5.11 which is listed in the dependencies. The Qt 5.9 documentation doesn't list the back() function: https://doc.qt.io/qt-5.9/qstring.html while lates Qt does have it: https://doc.qt.io/qt-5/qstring.html#back
Now, if this is the only Qt version problem, it would be easy to make this compatible with early Qt5 by replacing that function. However, I'm afraid there might be some other places where functionality belonging to Qt >= 5.11 is used. Since this version is default in Debian, I suggest perhaps you could investigate installing Qt >= 5.11.
Regardless, I'll make the change for the particular error you got right now in the next branch, it's not a large one.

waterwin, 2020-09-09:
This was a bit unexpected, I am normally not at the front of solutions, not being a programmer but a radio-operator ;-)
I have installed a VirtualBox with Debian and installed qradiolink which works somewhat, all to have a reference to what to expect in Ubuntu in another virtual box.
I have just installed QT 5.12.9 on Ubuntu and still have the same error. But I have to say that I am not totally sure that I installed Qt in the right way and/or am using the needed version. Qt is a bit above my understanding so far.
It is a real pity Debian is using a much newer Qt version than Ubuntu is.
In the Debian machine I still have to get my Adalm Pluto working with qradiolink. And a RSP1A SDRplay.
Cheers.ErwinPE3ES/F4VTQ

kantooon, 2020-09-09:
Ok, so at least you can get it to work. Next step I suggest to try the AppImage binaries on Ubuntu. I re-enabled their build today after something got broken last month upstream so the last release has it again. They should be standalone, however you should be aware they come with an older version of gr-osmosdr, so I'm not sure how that works out for LimeSDR devices. The PlutoSDR should be fine once SoapySDR and SoapyPlutoSDR are installed (last one preferably from source since no distro packages it correctly yet).

waterwin, 2020-09-09:
Not much is happening with the appimage.My CubicSDR is also an appimage and that works but I see no reaction when double clicking or choosing run for the QRadioLink sw.Will try in another VirtualBox as well.
Thanks for the support!E.

kantooon, 2020-09-10:
The reason is that Debian's version of gr-osmosdr is linked to libuhd, so you can use an USRP if you want, but on the other hand I can't include the library in the AppImage because it has its own firmware for the USRP and that firmware can be upgraded separately. Not ideal, I know, but I want to at least keep compatibility with USRP hardware.

waterwin, 2020-09-10:
Yes and it looks as if you have just lost that. Uhd from git installs libuhd 4.0.0 and the app image fails with checking for a specific 3.13.1 or so.  Why not check for having at least the needed version?
E.

kantooon, 2020-09-10:
You can checkout a certain tag from git, so you're not forced to use the latest version. But yeah.

waterwin, 2020-09-11:
Yes and then it is checking for Airspy, another type of hardware I do not have ...Is the checking not a bit too general or restrictive in the AppImage?Or should I load a lot of in fact unneeded dependencies first? Even when using the AppImage.

./QRa*/tmp/.mount_QRadiolhJF4N/AppRun.wrapped: error while loading shared libraries: libairspyhf.so.1: cannot open shared object file: No such file or directory

kantooon, 2020-09-12:
I can see you don't have much experience with Linux so I'll try to explain. The libraries which are used in an application have to be linked at compile time. The linking can be done in two ways: statically and dynamically, where dynamic linking means that the library is shared with other programs (distributed separately) and only loaded when the program is run. Dynamic linking is standard in GNU/Linux distributions for various reasons. In this case the libraries that the program needs here (libuhd, libairspyhf etc.) are not dependencies of QRadioLink. The program can work just fine without them if you don't need to use such a device. However, they are dependencies of libgnuradio-osmosdr, and in this case the maintainer of gr-osmosdr decided that it should be built and packaged with all possible dependecies in order to support as many devices as possible. Imagine if you were actually trying to use an USRP instead of a Pluto and gr-osmosdr would not have been linked with libuhd. Then you could not use such a device with the program.

Now if you want to only link to shared libraries that are actually usefull to you, that is possible as well, but you have to build the gr-osmosdr library from source and CMake will detect which libraries you actually have and only build those.
The AppImage format is really not a very common executable format and it does not have very well documented build tools to suport it. As such, I am providing them only in the hope they are usefull, but the recommended way to use the program is either to install the Debian packages or build it from source. I am also looking for volunteers to package and maintain the software on their favourite distribution, but since this a niche of a niche, there are not many people interested to do that.
Hope this helps.

waterwin, 2020-09-12:
With regard to your comment on my Linux knowledge I absolutely agree. ;-) So all valid information is helping me a lot.

But I disagree with your comment on osmosdr as they state themselves:
Before building the block you have to make sure that all the dependencies (see list of supported devices above) **you are intend to work with** are properly installed.
So you could decide to build gr-osmosdr without any dependencies on hardware blocks and let the user decide what to install for the needed local hardware.
Whether that approach would work with building an AppImage, falls again in the category above (not having enough Linux knowledge). My approach would be to have the AppImage only check for existence of blocks, not for forcing the availability of specific blocks. And make sure the gr-osmosdr part of the AppImage is not doing its own checking.

I will re-install gr-osmosdr under Ubuntu and try to wiggle my way past all AppImage checking.

Doing the same under Debian is giving me headaches as Debian appears to force the security rules for unofficial ppa's a lot stricter than Ubuntu is doing. Blocks me on "no release file found" ever so often.

kantooon, 2020-09-13:
Sorry, I meant no offense with that comment. You obviously have a lot of experience with other operating systems and you made some pertinent suggestions, but they just don't apply here due to the way things are with gr-osmosdr. You have described a plugin-based architecture, which is not how most components that are a dependency of qradiolink are built, but there are a couple (Qt and SoapySDR) which use this architecture. For example, SoapySDR can work with any number of plugins without needing them to be built in at compile time. You can add a PlutoSDR plugin anytime, or a LimeSDR plugin, and it will automatically detect and enable them. But gr-osmosdr doesn't work that way. Either the person who built the library (which is usually the package maintainer in a Linux distribution) compiled some hardware support in, or he didn't. If he didn't, you're out of luck. You have to rebuild it and install it from source yourself. Or you reach a case such as yours here, where some libraries are needed even though you don't use them. As far as I am concerned, this is outside of what I can do, since gr-osmosdr is upstream and I use the package verbatim in the AppImage.

There is an issue here though, but it's not with gr-osmosdr though, it is related to how the AppImage building application (linuxdeploy) selects which libraries to include in the final binary. In this case, this application used inside the Travis build/package job didn't include some libraries that you need to run the AppImage and which are dependencies of gr-osmosdr. So there is a case for altering this build-time component to get it to behave correctly, Unfortunately, this application is not very well documented so I might just remove the AppImage entirely from the deliverables until a proper solution is found.

Now, regarding Debian and PPAs, you should know that Debian regards PPAs as even slightly harmful for the ecosystem and thus does not encourage them or support them very well. The reason is that with PPAs each user is encouraged to build their own package repository instead of working towards submitting them upstream so all users can use them. This tends to lead long term to fractured communities where everybody has to maintain their own packages instead of having only one place where this is done (at the Debian distribution assembly lines). We tend to agree with this idea, so we either use the distribution provided packages or we build some piece of software from source in the case where we want to use some of the latest features. But in this case, software like GNU radio and Qt are just too big and too hard to build from source so we settle for the Qt and GNU radio packages provided by Debian, while we recommend building SoapySDR plugins like SoapyPlutoSDR, LimeSuite and SoapyLMS7 from source because they are moving with a larger average speed than the rest of the dependencies.
Hope this doesn't confuse you .

waterwin, 2020-09-13:
And none taken because 1. I indeed lack knowledge in Linux, and
2. You take a lot of time to write and put a lot of effort in detailed explanations and guidance.

I learn that I can better build from source on Debian in contrast to using ppa's
And will build from source instead of using the AppImage of QRadioLink on Ubuntu. With the right versions of dependencies installed that should be doable.

kantooon, 2020-09-13:
On Debian 10 (current stable), you don't even have to build qradiolink from source, unless you want the latest code from next (or if you want to change the source code yo suit your needs). You can just install the deb archive provided with `apt install ./qradiolink_0.8.3-5_amd64.deb` or using `dpkg -i qradiolink_0.8.3-5_amd64.deb`. In the latter case, you will have to install dependencies manually first by using the command provided in the README.

My advice is, if using either the PlutoSDR or the LimeSDR, to build SoapyPlutoSDR and LimeSuite along with the SoapyLMS7 module from source, because Debian ships older versions that lack some functionality. They are quick to compile and install, will take less than 10 minutes for all. I personally also build gr-osmosdr and SoapySDR from source to be able to use newer versions. The rest of the dependencies can be the Debian default.

Also note that you can use qradiolink from a VirtualBox virtual machine just fine, you just need to install virtualbox-guest-additions-iso ( `apt-get install virtualbox-guest-additions-iso` ) on the VM to enable USB3 pass-through in the VM configuration. By default VirtualBox only has USB1.0 which is not enough for these SDR devices.

waterwin, 2020-09-13:
Installed -5 deb

[ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
  sendto(udp://[ff02::c]:1900) [99: Cannot assign requested address]

I used device string usb=1.2.5,soapy=0 as this works on the same machine (VirtualBox) with my Adalm Pluto and the installed GNU Radio Companion
I will re-read the manual

kantooon, 2020-09-14:
For the Pluto, use the device string `soapy=0,driver=plutosdr` (notice no space in the string) or even just `soapy=0` may work.
Since the Pluto has only one antenna selectable, you might be able to put anything there, just make sure you don't leave the antenna fields empty. Otherwise see the Analog Devices documentation.

kantooon, 2020-11-01:
I believe this issue is resolved, if it is not, please reopen.

closed 2020-11-01 by kantooon