qradiolink/qradiolink #74: All PSK and FSK digital modes broken with GR 3.8

kantooon, 2020-10-07:
**Describe the bug**
None of the digital modes except FreeDV work after upgrading to GR 3.8

**To Reproduce**
Attempt to use digital modes

**Expected behavior**
Decoding should work

**Desktop (please complete the following information):**
Debian 11

**Additional context**
Change of behaviour in convolutional decoder rendered STREAMING mode unusable with current modem architecture.
No workaround appears possible. Any fix likely to require extensive rewrite of modems.

kantooon, 2020-10-08:
Should be fixed with workaround but more testing required.

kantooon, 2020-10-12:
Workaround blocks had their own issue. Re-opening to document proper workaround this time.

kantooon, 2020-10-12:
This is related to issue https://github.com/gnuradio/gnuradio/issues/3376

To work around this, run volk_profile and ensure that the line starting with ` volk_8u_x4_conv_k7_r2_8u` reads
`volk_8u_x4_conv_k7_r2_8u spiral spiral`, and if not, change it. After this, modems will work.

hiperiondev, 2025-01-23:
Its' solved?

kantooon, 2025-01-24:
Only some old versions of GNU Radio 3.8 were affected by this. 3.10 should not be affected.

closed 2025-03-15 by kantooon