qradiolink/qradiolink #81: Hack RF Setup

BrunoKlu, 2020-11-03:
Hello, I'm trying to use hackRF one with QradioLink.
I'v been to QradioLink Software, then setup > General settings 
And did paste the HackRF adress Hackrf=434e23 in RX and TX device args. Then when I run RX in the spectrum window, all is going very slow, then crashes. [image]

Any ideas

Many thanks

Bruno

kantooon, 2020-11-03:
Can you provide more details? Operating system, GNU radio version, qradiolink version, and a copy of the console log when this happens might help. Also, can you try to see if you see the same issue or not in Gqrx.
thanks.

BrunoKlu, 2020-11-03:
Hi Kantooon

Thanks for your reply, I did some more investigation, QRadio log does not input any issue, it appears that it is more a matter of processing power from the laptop. It is going wrong when I'm going full screen. If I stay windowed then it is a bit slow but fine.

I'm using it with DragonOS Focal.
And hack rf version 2017.02.1, GPRX is working perfectly. No idea how to get QRadioLink version.

Many thanks for your help

kantooon, 2020-11-03:
re. version, you probably gave installed it from somewhere, right? So the package will have a version, like 0.8.5-rc3 which is the latest version available here.
Now, for the actual issue. Processing power should really not be a problem unless you use huge sample rates or large FPS values. DO note that you can also configure the FFT bins size. If your bin size is too large, the waterfall will move really slow. This is documented in docs/README.md
I strongly suggest you run `volk_profile` before using anything GNU radio related including qradiolink. This will make sure the code takes advantage of any CPU optimization options present.

BrunoKlu, 2020-11-04:
Re, the version I have here was preinstalled on DragonOS so I didn't dl the file.. how can I get the version ? I'v played with FFT and sample rate, that helps, the only thing is that compared with GQRX , it is faster display in GQRX. I did the " volk_profile " command. Thanks for your help !

kantooon, 2020-11-04:
Thanks for the feedback, I personally can't reproduce this issue here. qradiolink uses 50% less CPU than Gqrx for the same settings: 1 Msps, FPS 15, FFT size 65536.
At 4 Msps, with FPS 30, FFT size 131072, qradiolink also uses almost half the CPU as Gqrx when receiving. It would be strange if it was otherwise, since Gqrx has more faithful audio and that brings a performance penalty. See screenshots attached below.

[image]
[image]

kantooon, 2020-11-04:
Also, please let me know how well is HackRF doing when transmitting with qradiolink. It's one of the devices which have unknown (possible) support in the README and I'd like a confirmation that it is well supported (while I know it can't do full duplex).

kantooon, 2020-11-15:
I've managed to reproduce the performance issue using the highest FPS setting and low fftsize, which generates very many rapid GUI update requests.

The reason the issue occurs is that Qt GUI rendering (painting) is by design single threaded, and when there is too much data to be rendered it will saturate one CPU core at 100%.

The way to avoid this is not to run too high FPS if your computer cannot handle it. I could lower the maximum available FPS value, but there may be people who do have the CPU power and fo whom it might be useful. As such, I won't attempt to do anything about this. Perhaps in the future there will be some rendering optimizations but don't count on it.

closed 2020-11-15 by kantooon