qradiolink/qradiolink #82: Install fails on Raspberry Pi 4

srs4511351, 2020-11-06:
I have an SDRplay RSP1A.
After trying to satisfy all of the prerequisites and following the installation in README.md, the installation fails with errors.
Here is the last bit of installation messages:

In file included from ../src/mumbleclient.h:34,
                 from ../src/mumbleclient.cpp:17:
../src/sslclient.h:48:5: error: 'QSslCipher' does not name a type; did you mean 'QSslKey'?
     QSslCipher getCipher();
     ^~~~~~~~~~
     QSslKey
../src/sslclient.h:65:5: error: 'QSslSocket' does not name a type; did you mean 'QUdpSocket'?
     QSslSocket *_socket;
     ^~~~~~~~~~
     QUdpSocket
make: *** [Makefile:7622: mumbleclient.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from ../src/mumbleclient.h:34,
                 from ../mainwindow.h:38,
                 from ../mainwindow.cpp:17:
../src/sslclient.h:48:5: error: 'QSslCipher' does not name a type; did you mean 'QSlider'?
     QSslCipher getCipher();
     ^~~~~~~~~~
     QSlider
../src/sslclient.h:65:5: error: 'QSslSocket' does not name a type; did you mean 'QUdpSocket'?
     QSslSocket *_socket;
     ^~~~~~~~~~
     QUdpSocket
In file included from ../src/mumbleclient.h:34,
                 from ../mainwindow.h:38,
                 from ../main.cpp:27:
../src/sslclient.h:48:5: error: 'QSslCipher' does not name a type; did you mean 'QSlider'?
     QSslCipher getCipher();
     ^~~~~~~~~~
     QSlider
../src/sslclient.h:65:5: error: 'QSslSocket' does not name a type; did you mean 'QUdpSocket'?
     QSslSocket *_socket;
     ^~~~~~~~~~
     QUdpSocket
make: *** [Makefile:7079: mainwindow.o] Error 1
make: *** [Makefile:6590: main.o] Error 1

Do I need to include any other information?

----Steve

kantooon, 2020-11-06:
Can you please check which Qt version you have installed? At least Qt 5.7 is necessary, preferably Qt >= 5.14
Also, I was not aware that SDRplay is supported by gr-osmosdr. Are you using a fork or the official version? If it's indeed supported, I'd like to add this information to the README.

srs4511351, 2020-11-06:
I have 2 versions of QT installed, the original Qt version 5.11.3 for
Desktop that is in the Raspberry Pi OS packager and Qt version 5.14.2 for
Desktop that I installed.
When installing qradiolink, for the qmake command, I used
"/opt/Qt5/bin/qmake .." to specify Qt version 5.14.2.
I used 5.14.2 because TLS was not supported until Qt 5.12. I had this
error: ../src/sslclient.cpp:32:32: error: 'TlsV1_3' is not a member of
'QSsl'

I specified the master fork. Below is the procedure I followed. The failure
was in make.

git clone https://github.com/qradiolink/qradiolink
cd qradiolink/
git checkout master
mkdir -p build
cd src/ext/
protoc --cpp_out=. Mumble.proto
protoc --cpp_out=. QRadioLink.proto
cd ../../build/
/opt/Qt5/bin/qmake ..
make -j3

As for gr-osmosdr, how does the installer choose to use it? I started with
the SDRplay RPI v.07 image, which has soapysdr and Cubic SDR installed. The
SDRplay API would have been preinstalled.

https://www.sdrplay.com/community/viewtopic.php?t=4211
*I did not use this: *This 'sdrplay3' branch is available here:
https://github.com/fventuri/gr-osmosdr/tree/sdrplay3

https://www.sdrplay.com/linux-gr-source/
...building osmosdr

https://osmocom.org/projects/gr-osmosdr/wiki
Makes reference to SDRplay RSP through SDRplay API library

Maybe I need to do a special osmosdr build? If you could look this over,
maybe you could make sense out of it.

----Steve

kantooon, 2020-11-06:
Ok, let me check, I think the TLS v1.3 version define in the code is not correct, I'll adjust that so the build can pass with 5.11 as well. But the fact that it says it's missing QSslSocket is not normal. That's been a part of Qt since at least version 4, specifically the libqt5network5 library and corresponding headers. I can suspect that the Qt5 dev package (the headers required to build it) are missing some components. The dev package should be the same version as the version it is trying to link into.

kantooon, 2020-11-06:
Regarding the SDRplay. If you are normally using it through SoapySDR, that's going to work, you just need the relevant Soapy plugin. If you don't, then you'll probably need a fork like the one you mentioned. As a general rule, if the device is usable in Gqrx, it will also be usable here since it uses the exact same gr-osmosdr API.

srs4511351, 2020-11-06:
Thanks. I just need to know how to install everything. Making it simpler
would have 12 hours of compiling QT and the potential errors.
Every step of the way was fraught with stumbling blocks. I am unable to
find instructions from anyone who has successfully installed it on RPI.

With the installation I have of Gqrx (on a separate installation), I have
to use Soapy Remote, which I don't like as it uses the network, which is
silly when used on the same machine.
Apparently, I don't know how to build it with SoapySDR included.
It appears that  gr-osmosdr is listed in the QRadioLink requisites, so I
installed it. Also, there was a QRadioLink installation error related to it.
Error Resolved:
#include <osmosdr/sink.h>
compilation terminated.

----Steve

kantooon, 2020-11-06:
I'm afraid the only easy way is to have a distribution provided package, because the Raspberry Pi OS is not exactly similar to Debian although it's derived from it. I can't provide even Debian packages currently, because Travis CI which was used for this purposes does not support open source projects anymore.
I know what sort of issues you are going through, I had the same problems with raspbian in the past and eventually just ended up using Debian on it.

Re. SoapySDR support, in the official gr-osmosdr version this is built in so you don't have to do anything about it. But there are numerous unofficial forks floating around and I don't know about any of those.

srs4511351, 2020-11-06:
Any guidance is appreciated. Let me know if you have any answers. Enabling
me to use Qt 5.11 would be nice. I can restore my system to one from before
Qt 5.14.
I may also try qradiolink-0.8.3-5, but I prefer to use the latest.

----Steve

srs4511351, 2020-11-07:
I decided to try an older version of qradiolink on another installation
where I did not upgrade gnuradio and QT5. I hope I did it right.
Is there documentation for configuration and for the user interface?

git clone https://github.com/qradiolink/qradiolink
cd qradiolink/
git checkout 0.8.3-5
sh ./build_debian.sh

After installation, it started the program and it runs! How can I check the
version?
Here are my notes with text from the terminal:
***********************
[with soapy remote running]

pi@sospisdr:~/qradiolink/build $ ./qradiolink
qt5ct: using qt5ct plugin
[7/Nov/2020 00:41:49] [Info] Starting qradiolink
qt5ct: D-Bus global menu: no
[7/Nov/2020 00:41:50] [Debug] Looking up available audio devices
[7/Nov/2020 00:41:50] [Info] Using audio input device default
[7/Nov/2020 00:41:50] [Info] Using audio output device default
[7/Nov/2020 00:41:50] [Info] Setting VOIP bitrate to 24600
[7/Nov/2020 00:41:51] [Info] Using audio input device default
[7/Nov/2020 00:41:51] [Info] Using audio output device default

[clicked RX on GUI]

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf
rfspace airspy airspyhf soapy redpitaya freesrp

FATAL: Wrong rtlsdr device index given.

Trying to fill up 1 missing channel(s) with null source(s).
This is being done to prevent the application from crashing
due to gnuradio bug #528.

[Does not use soapy remote and Gqrx runs at this point]
***********************
Original RX device args: rtl=0
Tried the string from Gqrx: driver=remote,remote=tcp://192.168.1.76:55132
,remote:driver=sdrplay,soapy=1
Is there a string to try that will not require remote?

Will now receive. I had to play with the gains to get anything.
In setup > RX gains, There are the IF and RF gains, identified only by tool
tips. The level values are not labeled, but print in the terminal window as
the slider is moved.
Broadcast FM sounded OK. No Stereo. AM is really tinny. Is there something
shaping the audio?
The low audio freq cut helps with SSB, but I would like to be able to
adjust this.
Other demululators have different audio frequency responses. Is this
configurable?
The FreeDV demodulators have dropouts and glitches. The basic ones are
clean.
When adjusting RX gain on FM, it basically froze. It would come to life
occasionally.
Restarted the program.
AGC in Radio Settings seems to work when the gains are cranked to the
extreme (as a test).
Constellation doesn't display anything.

Now I need to figure out how to get something to transmit.

----Steve

kantooon, 2020-11-07:
Ok, glad you could make it work. The only way to check version right now is to know either the version of the deb package installed or the git tag / commit hash from which it was built. When I have some time, I will try to add a command line parameter to retrieve the version at runtime.

Most documentation, as far as I have been able to document it, is located in the docs/README.md file inside the repository.
Your audio "freezes" are most likely the squelch coming in and out, you can adjust it manually and there is also an Autosquelch button which will adjust it to squelch above the current level, so don't press it while you want to listen to something.
SSB, AM, and FM filters are adjustable by dragging on the margins of the filter box on the screen. You can also zoom in on the signal by hovering on the bar where frequency numbers are displayed and using the mouse scroll wheel or dragging left and right. Same for the vertical bar with signal levels on the left.
Be aware that qradiolink is a voice communications transceiver, so it is not really optimized as a receiver for best sound. If you need best sound quality, Gqrx will be a lot better. The audio sample rate is only 8000 Hz so it will cut out frequencies outside the human voice spectrum.

The constellation display is onyl needed for digital signals, BPSK, DQPSK, 2FSK and 4FSK (except FreeDV which has different internals). It will also be useful for the video signal or the IP modem signal. The video transmission in your older version has no sound. That is implemented on the next branch (development branch) but not part of any release yet. To transmit a text message, just press "Send text" or "Text file beacon" which will broadcast a text file on repeat. To stop text transmission, press "PTT" twice.
The tooltips will hopefully guide you from here. This being a hobby to me, I don't always have time to properly document things.

srs4511351, 2020-11-08:
Thanks for the operation tips. I need to figure out how to install the
newer version.
qradiolink is interesting because I should be able to get a transmitter
working with it.

I need to figure out how to use the non-network sdrplay driver. The
Internet says to use device string soapy=0,driver=sdrplay, but it doesn't
work for me.

The "freezes" that I was referring to was the application freezing. The
squelch is set fully counterclockwise. Large changes in the RX gain control
can cause the application to freeze.

The AGC does not seem to be working. If gains are set to hear weak
stations, audio clipping can be heard on strong ones.
Sometimes, decreasing attenuation seems to invoke AGC, but it does not
recover.
Sometimes turning up RX gain(atten) resets the AGC by appearing to have the
opposite effect.
After restarting the application, the gains are changed or reset.

As for AM sounding tinny, it is due to attenuation in the audio low
frequencies. With AM, the only bandpass adjustment is the width. That will
cut or extend the high frequencies of both sidebands, but has no effect on
the low frequencies, which are centered between the filter edges. The low
frequency rolloff would have to be done somewhere else.

On USB, changing the low cut which displays on the screen as you adjust it,
has no effect on the sound unless you adjust it past/higher than the high
cut. It cannot be adjusted lower than 200 Hz. The bandpass adjustment has
no effect on the FreeDV filters. LSB does the same thing in the opposite
direction.

----Steve

kantooon, 2020-11-08:
This is great feedback, much appreciate your taking the time to describe the issues. One issue at a time:
1. For soapy devices, what seems to help sometime when a device is not found correctly, is to put in more parameters that `SoapySDRUtil --find` gives in the string, without spaces. For example:
Found device 1
  addr = 24607:1027
  driver = lime
  label = LimeNET-Micro [USB 2.0] 583A25BFD101
  media = USB 2.0
  module = FT601
  name = LimeNET-Micro
  serial = 00583A25BFD101
The device string could be `soapy=1,driver=lime,addr=24607:1027,serial=00583A25BFD101`
To make matters a little worse, this also depends on which gr-osmosdr is used. I have version 0.20 here and it behaves in a different way to version 0.19.

2. The freezes, I now remember a similar issue seen some time ago with an early pre-production Lime board. The freeze is definitely caused by one of my mutexes, but the root cause in that case was an issue in the Soapy Lime plugin, which got fixed later so I didn't bother looking at that code again. But now based on your description I think I may have to actually change something in the code as a workaround.

3. The AGC in GNU radio is a source of headaches, there are actually three different blocks which behave in their own way, I will try to reproduce the issues and change the parameters to the blocks.

4. I don't really ever use AM, not having a shortwave antenna, but the filter rolloff for low frequencies may indeed be too high. That can easily be corrected. Probably the same for the SSB low frequency cutoff, although SSB is different because it uses a complex taps filter for transmitting which is tightly coupled to the receiver filter adjustment. There is potential for improvement there, but needs changes to the UI code as well.

5. FreeDV has a fixed width filter for the digital signal implemented on FreeDV side, so changing the filter width on GNU radio's side will have no effect, unless you mean for close adjacent SSB signals (they will only pass through if FreeDV is not synced in digital mode RX, if FreeDV is synchronized there will be no analog pass-through.

kantooon, 2020-11-08:
One other comment. Looking at your logs above I noticed this:
[7/Nov/2020 00:41:50] [Info] Using audio input device default
[7/Nov/2020 00:41:50] [Info] Using audio output device default

If the system has Pulseaudio (default in most distros), don't use the `default` sink and source. Instead, for better sound quality use the Pulseaudio sinks and sources, which should be located right at the end of the audio devices listing. You can recognise them easily because the names are long and specific:
[image]
Of course, this advice does not apply for an Alsa only system.

srs4511351, 2020-11-08:
1. SoapySDRUtil --find gives the 3 lines below:
Found device 0
  driver = sdrplay
  label = SDRplay Dev0 RSP1A 19110C0797

That makes the device string "soapy=0,driver=sdrplay", which is like what I
find on the Internet. It doesn't work in qradiolink or gqrx.
When I put it in RF device args, the terminal shows:
--
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf
rfspace airspy airspyhf soapy redpitaya freesrp

FATAL: SoapySDR::Device::make() no match
--
SoapySDRUtil --find="soapy=0,driver=sdrplay" does find the device.

4. I think this is the audio device that you suggested. Strangely, there
are no stereo devices. I know that with other SDRs, I can have FM Stereo.
The sound quality seemed to be the same on this or any device. While
messing around, the audio level decreased a lot. It's like the thing that I
referred to in the AGC issue. Restarting Soapy Remote and qradiolink
brought the audio level up to normal.
[image: qradiolink_audio.png]
It seems to me that the low frequency rolloff for music stations, AM and FM
should be really low.

5. The FreeDV filters don't work very well. They are full of pops, squeals
and random sound from who knows where.
Still, the plain USB/LSB filters default to a low cut of 200 Hz and
dragging that end doesn't do anything.

The frequency display can be adjusted by hovering over a digit and rolling
the mouse wheel. It works very well, much better than CubicSDR.

----Steve

kantooon, 2020-11-08:
Just to clarify, the AM mode is meant for amateur radio not for receiving broadcast radio. So it is optimized for human voice spectrum. And the WideFM receiver is mono only, not stereo. 
Have you used FreeDV before with an SSB transceiver? I think what you are describing as noise is just random spurious decodes, they are quite normal.

srs4511351, 2020-11-08:
I agree that Qradiolink would be best used for communications applications.
However, user’s choice would be nice for the audio quality. Those who use
AM like the audio quality it gives.
I have been thinking about trying to get the latest version working.

I have never used FreeDV. I ran it recently to see what it’s like, but I
don’t see anything I would use. It doesn’t seem to be able to access an
SDR.
The pops, click and squeaks in their filters are clearly not working right,
not in the other filters, and is very annoying. I could try to get an audio
recording is you would like.

----Steve

kantooon, 2020-11-08:
A recording would be good. I'm intrigued about what you're describing. FreeDV is a digital mode, and you need to be listening to a ham transmitting it to be able to decode. Is that what you are trying to do? Listening to any analog signals with it would be useless, it's not what it's meant for. It's fairly sensitive to accurate tuning due to the OFDM waveform, but otherwise very usable.

srs4511351, 2020-11-08:
I have to admit that I was unaware of the purpose of DV. They don't talk
about it on the air enough that I had ever heard of it.
The FreeDV filters do decode SSB and almost sound good, so I thought is was
supposed to work. One works fairly well.
I'll try DV later.
If you still want a recording, I can do that.

----Steve

srs4511351, 2020-11-09:
After reading qradiolink/docs/README.md, I see that this program has a lot
of unique and interesting capabilities.
It seems to be written for those who already know what everything is and
how to do them.

Am I correct in assuming that with qradiolink, I will be able to run my
qradiolink radio station from a remote location over the Internet?
How can I use the FreeDV digital voice modulator and demodulator?
Apparently I don't need to virtually connect it to the FreeDV application
to get the demodulated audio.
I couldn't find any FreeDV signals on the documented frequencies late this
evening. I'll keep looking.
There are also a lot of digital filters and a video one. Are they to be
used on the air? I looked up the usual digital modes, but they don't match
the filter names. Are these the IP radio modems? I'll need more help to
figure that out.

I attached a recording of the FreeDV800XA LSB filter made using qradiolink.
I was tuned to an 80 meter signal.

----Steve

kantooon, 2020-11-09:
1. Yes it helps if you have amateur radio knowledge.
2. Yes you can do remote operation via Internet
3. The same as the other modes, just select the appropriate modulation and transmit or receive. FreeDV here is standalone, does not require anything other than a SDR device.
4. They are not called digital filters. They are called digital operating modes, the filters are just a DSP component in a larger chain. And yes, obviously over the air. To use high power you need to have an amateur radio license, these are generally issued by the state regulatory agency following an exam. Contact some local hams in your country and they will help walk you through the procedure to get licensed, it is not very complicated, you just need to fulfill some legal requirements. 
5. To learn more about digital signal processing, head over to the https://www.gnuradio.org website or start looking at some amateur radio resources online, there is a lot to learn but there are a lot more interesting things to do as an amateur radio opeator than just shortwave: microwave communications, sattelite operations, repeater operations etc.

srs4511351, 2020-11-10:
1. Yes, I have been an amateur for 40 years. Combining the digital stuff
with it is new.
2. Thanks
3. I tried it today. I finally caught a FreeDV signal on the air. When
chatting on the FreeDV HF QSO Finder, I find that they are using FreeDV 700D
and qradiolink supports 700C. I'll have to wait for an upgrade to try it.
With this implementation, it would save a lot of configuration.
4. See 1.
5. I do need to read up on digital for amateur radio. There's more going on
than digital signal processing. I have been receiving FT8 for a few months
now. With GridTracker, I can easily see where the band is open. The digital
operating modes are new to me, especially the voice modes.

I tried installing the latest version on my system with Qt 5.11 after
seeing "The version check for TLS v1.3 was looking at the wrong version
#82" on github. I hoped that you had fixed it.
 I still get:

./src/station.cpp
../src/sslclient.cpp: In constructor 'SSLClient::SSLClient(QObject*)':
../src/sslclient.cpp:32:32: error: 'TlsV1_3' is not a member of 'QSsl'
     _socket->setProtocol(*QSsl::TlsV1_3*);
                                ^~~~~~~
../src/sslclient.cpp:32:32: note: suggested alternative: 'TlsV1_2'
     _socket->setProtocol(QSsl::TlsV1_3);
                                ^~~~~~~
                                TlsV1_2
make: *** [Makefile:1396: sslclient.o] Error 1
make: *** Waiting for unfinished jobs....

----Steve

srs4511351, 2020-11-10:
I found a binary package of Qt5.14.1-RaspberryPi3, which should run on my
Pi4.
I deleted the build directory, ran /usr/local/Qt-5.14.1/bin/qmake ..
re-ran make, which ran quite a bit longer, and ended in a new error. Can
you help me to determine what went wrong? Could it have used gnuradio 3.7?
I also installed gnuradio 3.8.

/usr/bin/ld:
gr_mod_ssb.o:(.data.rel.ro._ZTV10gr_mod_ssb[_ZTV10gr_mod_ssb]+0x40):
undefined reference to
`gr::hier_block2::set_log_level(std::__cxx11::basic_string<char,
std::char_traits<char>, std::allocator<char> >)'
/usr/bin/ld:
gr_mod_ssb.o:(.data.rel.ro._ZTV10gr_mod_ssb[_ZTV10gr_mod_ssb]+0x44):
undefined reference to `gr::hier_block2::log_level[abi:cxx11]()'
collect2: error: ld returned 1 exit status
make: *** [Makefile:597: qradiolink] Error 1

Thanks, Steve

kantooon, 2020-11-10:
If you're an amateur radio operator, great, it means you can use it to the full extent.
Note about FreeDV 700D: the current implementation of this mode has an issue with having more than one instance of mod/demod loaded - when trying to stop the receiver or transceiver by using the relevant button, the program will segfault in libcodec2. Other FreeDV modes are not affected. I've been meaning to contact David Rowe, the author of FreeDV, about this issue, but I've been a little busy with other things lately. I actually added 700D at some point, but removed it later due to the instability it caused. I want to add it back too, but I'm not sure how hard will this issue be to resolve on their side.
The other digital modes are for voice and text mainly for experimentation with different DSP techniques. The prefix denotes the modulation type, and the number suffix (1K, 2K, 10K etc.) denotes the effective bit rate. There are 3 codecs used for digital voice (outside of FreeDV of course): Codec2 700C, Codec2 1400, and Opus 9400 bits per second.

Now, about your first error, the define in the code should check the version and not enable TLS v1.3 unless using Qt 5.14. I can only assume that the version check fails for some reasons on your system (multiple mixed Qt versions and the wrong one is found? I don't really know).
The second error seems easier. Yes, I think it is trying to link into GNU radio version 3.7 instead of 3.8. These things tend to happen when you mix library versions, the good news is that it is usually fixed quite easy, by having the appropriate PATH and LD_LIBRARY_PATH variables set.
You also need to ensure that the Qt / GNU radio headers (they are usually in a package suffixed with -dev) are consistent. It can happen that qmake uses one version of Qt for headers and another for ld to link into.
Of course, it also depends on you install locations, but usually (this is what I do) the source build is configured to be installed in `/usr/local/lib` and LD_LIBRARY_PATH has `/usr/local` before `/usr/lib`. I also assume that you tried to run manually `ldconfig` already. My advice would generally be aimed at Debian, since this is what I use everywhere, but I hope the Raspberry Pi OS does not stray to far from normal defaults.

And here is another curious thing: if you built 0.8.5-rc3 successfully, there is no dependency or version difference to the latest code from `next` apart from some gstreamer plugins used for video mode transmission. So you just need to ensure some consistency when switching builds and it should work just as well. 0.8.5-rc3 requires GR 3.8 as well. I have to prepare for work now, but if you do some research into having multiple library versions installed in different locations, I'm sure you can pull this off. I do this all the time, despite the general advice not to do it.

srs4511351, 2020-11-11:
It looked like there was no easy way to separate gnuradio 3.7 from 3.8, so
I uninstalled 3.7. This caused qradiolink make to fail, so I
reinstalled gnuradio 3.8 and gr-osmosdr.
I ended up with gnuradio 3.8.2.0

I installed Release 0.8.5-rc3. I used git checkout master, which I
assume means the latest.
This time, the compiler completed. qradiolink runs, but I can't get it to
receive anything.

The terminal window displays:
[11/Nov/2020 01:35:13] [Info] Starting qradiolink
[11/Nov/2020 01:35:14] [Debug] Looking up available audio devices
[11/Nov/2020 01:35:14] [Info] Using audio input device default
[11/Nov/2020 01:35:14] [Info] Using audio output device
alsa_output.platform-bcm2835_audio.analog-mono
[11/Nov/2020 01:35:15] [Info] Setting VOIP bitrate to 24600
[11/Nov/2020 01:35:15] [Info] Using audio output device
alsa_output.platform-bcm2835_audio.analog-mono

When I click the RX button, 2 controls appear to the right of RX gain: IFGR
and RFGR which is what the 2 gain controls are called in my SDR.
Then the terminal displays:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32,
scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.1.76:36786
[INFO] Client side status bound to 192.168.1.76:57770
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.1.76]:38315
[INFO] Server side stream connected to [::ffff:192.168.1.76]:36786
[INFO] Server side status connected to [::ffff:192.168.1.76]:57770
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4
bytes, window=1024 KiB
[WARNING] Set thread priority 0.5 failed: Operation not permitted
[INFO] Client side stream connected to 192.168.1.76:38315
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4
bytes, window=43008 KiB

Soapy Server says:
SoapyServerListener::accept([::ffff:192.168.1.76]:49224)
SoapyServerListener::accept([::ffff:192.168.1.76]:49226)
SoapyServerListener::close()
SoapyServerListener::accept([::ffff:192.168.1.76]:49228)

With squelch set to minimum and playing with the gain controls, I never get
anything on the display. Sometimes the S meter moves, then it freezes.
I can select the frequency by clicking on the display. It was initially in
the range of the frequency it had when I clicked RX.
If I set the frequency 1-2 MHz lower, for example, the display does not
move, but clicking on it changes the frequency to something in the range I
changed it to. The display does not zoom or scroll.
The display does work normally when RX is stopped.

Later I changed the frequency, clicked RX to stop it, and when I clicked RX
again to start, it displayed a signal panorama, but only one screenfull,
then it froze. Once, it worked long enough to get a tiny bit of the
waterfall to display.
------------------
The older version that I was initially successful with was 0.8.3-5.
When I got TLS error, I tried it with only Qt5 5.11 installed. There was no
other version at that time.
To identify the right Qt5 version, I used
/usr/local/Qt-5.14.1/bin/qmake ..

After installing gnuradio 3.8, I added
export PYTHONPATH=/usr/local/lib/python3/dist-packages:$PYTHONPATH
export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
Later, I added /usr/local/Qt-5.14.1/bin to the path.
I did a ldconfig after gnuradio installation as instructed and also after
qradiolink although the instructions do not call for it.

Uninstalling gnuradio 3.7 and reinstalling 3.8 made the difference.

----Steve

kantooon, 2020-11-13:
I've been pretty busy at work these days, but it looks like you made good progress with the install, except for the Soapy Remote thing.

I've made some changes to the code based on your feedback: enabled AGC for AM mode in addition to SSB, widened AM audio filter and changed it from bandpass to low_pass for better low frequency respose. These changes will be present in the next release along with other new features.

If you manage to get a fully working installation on Raspberry Pi OS, it would be great if you wrote a short installation guide, this has been a pain point for many users who choose to use this OS.

srs4511351, 2020-11-14:
I have been able to get it to install, but I have no idea why it is not
working. I was hoping that you may know what is wrong with my qradiolink
installation.
Yes, I would be willing to share my procedure with the world, but I still
can't get the latest version to work.
I decided to start over. I removed the existing gnuradio, Qt4 and Qt5
installations. I reinstalled gnuradio 3.8.2.0 and gr-osmosdr
I installed the binary package of Qt-5.14.1

I also installed the latest gqrx from source. It uses gnuradio 3.8.
Initially it didn't run because it couldn't find libQt5Network.so.5.
After an hour of searching, I found that the LD_LIBRARY_PATH that I set in
.profile is being deleted by the operating system, so I had to add the path
to a file in /etc/ld.so.conf.d/  What a PAIN!
Now it runs and works well if I select the filter and mode options before
starting the DSP. When the DSP is running and I change things like LSB or
change the filter width from the pulldown menu, the panoramic display
freezes.

With that somewhat working, and on the new installation, I get more errors
when I run qradiolink.
Here is the terminal window contents

[14/Nov/2020 03:31:45] [Info] Starting qradiolink
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_num_locks
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_id_callback
qt.network.ssl: QSslSocket: cannot resolve CRYPTO_set_locking_callback
qt.network.ssl: QSslSocket: cannot resolve ERR_free_strings
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_cleanup
qt.network.ssl: QSslSocket: cannot resolve EVP_CIPHER_CTX_init
qt.network.ssl: QSslSocket: cannot resolve sk_new_null
qt.network.ssl: QSslSocket: cannot resolve sk_push
qt.network.ssl: QSslSocket: cannot resolve sk_free
qt.network.ssl: QSslSocket: cannot resolve sk_num
qt.network.ssl: QSslSocket: cannot resolve sk_pop_free
qt.network.ssl: QSslSocket: cannot resolve sk_value
qt.network.ssl: QSslSocket: cannot resolve SSL_library_init
qt.network.ssl: QSslSocket: cannot resolve SSL_load_error_strings
qt.network.ssl: QSslSocket: cannot resolve SSL_get_ex_new_index
qt.network.ssl: QSslSocket: cannot resolve SSLv23_client_method
qt.network.ssl: QSslSocket: cannot resolve SSLv23_server_method
qt.network.ssl: QSslSocket: cannot resolve X509_STORE_CTX_get_chain
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_noconf
qt.network.ssl: QSslSocket: cannot resolve OPENSSL_add_all_algorithms_conf
qt.network.ssl: QSslSocket: cannot resolve SSLeay
qt.network.ssl: Incompatible version of OpenSSL
[14/Nov/2020 03:31:45] [Debug] Looking up available audio devices
[14/Nov/2020 03:31:45] [Info] Using audio input device default
[14/Nov/2020 03:31:46] [Info] Using audio output device default
[14/Nov/2020 03:31:46] [Info] Setting VOIP bitrate to 24600

I click the RX button. The two RX gain controls appear and ...

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32,
scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.1.76:45122
[INFO] Client side status bound to 192.168.1.76:59098
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.1.76]:40946
[INFO] Server side stream connected to [::ffff:192.168.1.76]:45122
[INFO] Server side status connected to [::ffff:192.168.1.76]:59098
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4
bytes, window=1024 KiB
[INFO] Client side stream connected to 192.168.1.76:40946
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4
bytes, window=43008 KiB
[WARNING] Set thread priority 0.5 failed: Operation not permitted

----Steve

kantooon, 2020-11-14:
For the Qt Network errors, the hint is in this message:
qt.network.ssl: Incompatible version of OpenSSL

This warning:
[WARNING] Set thread priority 0.5 failed: Operation not permitted
Might come from SoapyRemote. Since I've never used the Soapy remote feature, I don't know if it means anything.
So, the question is: does reception work now or do you still have issues with it?

srs4511351, 2020-11-14:
I just remembered another thing. When building gqrx, qmake failed on
osmosdr. I used cmake instead and it worked.  I found other people who had
the same problem.

I am fairly certain that [WARNING] Set thread priority 0.5 failed:
Operation not permitted appears when starting gqrx also.
I tried running qradiolink with sudo and that message did not appear, but
there were other issues.

----Steve

srs4511351, 2020-11-14:
I forgot to answer your question.
No, qradiolink is not working. I do not receive signals. When I press the
RX button, the 2 gain controls from my SDR appear, so I think it is
talking.

gqrx does work as described previously.

----Steve

kantooon, 2020-11-14:
I would suggest to solve first the reported OpenSSL compatibility issue. However, your main issue seems to come from SoapyRemote not receiving samples from the device. I have never used SoapyRemote, so I can't give any useful advice here. Perhaps you could ask on the SoapySDR discussion forum or bug tracker? I am not sure if the version of  gr-osmosdr is the same as the one you were using before, but you might want to check that as well.

srs4511351, 2020-11-14:
Thanks. I am not able to uninstall OpenSSL. I will have to figure out how
to upgrade it. I have version 1.1.1

it would be nice if raspberry pi would offer packages that are remotely
current!

----Steve

srs4511351, 2020-11-14:
My version libssl1.1-1.1.1d-0+deb10u3+rpt1 seems to be the latest stable
version.
The latest docs on https://github.com/openssl/openssl only have OpenSSL
1.1.1 man pages.

It seems to be Qt that is complaining. Qt for X11 Requirements page does
not list openssl as a requirement.
Do you think that whoever built the Qt5.14.1 binaries had another version
of openssl? Otherwise, there is nothing out there that can explain this.

----Steve

kantooon, 2020-11-14:
the QtNetwork module is linked / uses OpenSSL. Qt is a big thing, there are lots of additional modules besides the UI: network, audio, etc.
This pain is precisely why I always use Debian on pi boards.

srs4511351, 2020-11-14:
I found another binary, qt5-opengl-dev_5.12.5_armhf.deb
Maybe I should give that a try?

Are there Debian distros out there that come as a restorable disk image?

----Steve

srs4511351, 2020-11-15:
I installed  qt5-opengl-dev_5.12.5_armhf.deb and reinstalled qradiolink.
The openssl messages are gone, but qradiolink still does not work. When I
click the start RX button, it still queries the SDR and displays the 2 gain
controls using the names that the SDR uses.

As for the qradiolink warning messages,
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[WARNING] Set thread priority 0.5 failed: Operation not permitted
They appeared on the successful installation with gnuradio 3.7.13.4 and Qt
5.11
I noticed that there are a lot less messages that appear before I click the
RX button on the failed installation vs the good installation.

I have attached the terminal window messages from the working qradiolink
0.8.3-5 with gnuradio 3.7.13.4 and Qt 5.11
gqrx 2.13.2 with gnuradio 3.7.13.4 and Qt 5.11
qradiolink  0.8.5-rc3 (today's git clone / checkout master) With gnuradio
3.8.2 and qt5-opengl-dev_5.12.5
gqrx 2.13.2 With gnuradio 3.8.2 and qt5-opengl-dev_5.12.5

They are all divided into sections, so you should be able to tell which is
which.

I did poke around the SDRplay web site and found a procedure to install
gr-osmosdr for SDRplay. After uninstalling the generic version that I
installed, it seemed to mess up my gnuradio installation. gr-osmosdr
installation failed because it cannot find gnuradio.
I had tried to re-run  gnuradio make install, but it failed with errors.
The reinstall is compiling now.

----Steve

========================
With gnuradio 3.7.13.4 and Qt 5.11

Command prompt text from working qradiolink 0.8.3-5

qt5ct: using qt5ct plugin
[14/Nov/2020 23:58:45] [Info] Starting qradiolink
qt5ct: D-Bus global menu: no
[14/Nov/2020 23:58:45] [Debug] Looking up available audio devices
[14/Nov/2020 23:58:46] [Info] Using audio input device default
[14/Nov/2020 23:58:46] [Info] Using audio output device pulse
[14/Nov/2020 23:58:46] [Info] Setting VOIP bitrate to 24600
[14/Nov/2020 23:58:47] [Info] Using audio input device default
[14/Nov/2020 23:58:47] [Info] Using audio output device pulse

Clicked RX to start SDR

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.1.76:49199
[INFO] Client side status bound to 192.168.1.76:42760
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.1.76]:37740
[INFO] Server side stream connected to [::ffff:192.168.1.76]:49199
[INFO] Server side status connected to [::ffff:192.168.1.76]:42760
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=1024 KiB
[WARNING] Set thread priority 0.5 failed: Operation not permitted
[INFO] Client side stream connected to 192.168.1.76:37740
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
SSSSSS

Quit porgram
[... 74 lines cut ...]
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BookmarksFile is /home/pi/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya 
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] Using format CF32.
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya 
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] SoapyRemote::setupRxStream(remoteFormat=CS16, localFormat=CF32, scaleFactor=32767, mtu=1500, window=44040192)
[INFO] Client side stream bound to 192.168.1.76:43421
[INFO] Client side status bound to 192.168.1.76:38146
[INFO] Using format CS16.
[INFO] Server side stream bound to [::ffff:192.168.1.76]:35655
[INFO] Server side stream connected to [::ffff:192.168.1.76]:43421
[INFO] Server side status connected to [::ffff:192.168.1.76]:38146
[WARNING] StreamEndpoint resize socket buffer: set 43008 KiB, got 1024 KiB
[INFO] Configured sender endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=1024 KiB
[WARNING] Set thread priority 0.5 failed: Operation not permitted
[INFO] Client side stream connected to 192.168.1.76:35655
[INFO] Configured receiver endpoint: dgram=1452 bytes, 357 elements @ 4 bytes, window=43008 KiB

srs4511351, 2020-11-18:
I have made huge strides in getting a system that can run qradiolink.
It seems that the soapysdrplay/gr-osmosdr software that I had was not
compatible with GNU Radio 3.8. Some was not compatible with the
latest SDRplay API driver.
I spent several days finding the rare software that will work with my
SDRplay RSP1A and GNU Radio 3.8.
I started with a freshly loaded OS installation and built it up from there.
I removed GNU Radio 3.7 and Qt 5.11
I now have gqrx working with it and without having to use Soapy Remote.
This is in my opinion a great accomplishment. I emailed someone who
claimed on twitter to have it working, but he did not respond to me.
He is a Linux and SDR expert and said that it was not an easy task.

I installed the FreeDV application from the Add/Remove Software
application. Is that what is required as a prerequisite?

While installing some of the prerequisites and the other software,
something I installed must have installed GNU Radio 3.7, and Qt 5.11
because they came back!
I removed GNU Radio 3.7, and Qt 5.11, reinstalled Qt 5.12 and tried to
install qradiolink.

So, with the system working better than ever, the qradiolink install
crashed:

../src/video/videoencoder.cpp
In file included from ../../qradiolink/src/gr_modem.h:36,
                 from ../src/radiocontroller.h:46,
                 from ../main.cpp:33:
../../qradiolink/src/gr/gr_mod_base.h:26:10: fatal error:
gnuradio/vocoder/freedv_api.h: No such file or directory
 #include <gnuradio/vocoder/freedv_api.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:6446: main.o] Error 1
make: *** Waiting for unfinished jobs....

What's up? Did I install the correct FreeDV?

----Steve

kantooon, 2020-11-18:
You seem to have hit the same problem as described in issue #67 
Check the comments on that ticket, and ensure that cmake finds the freedv headers (from libcodec2-dev) during the gnuradio build process. Otherwise gnuradio won't have freedv support compiled in and the qradiolink build will fail like this.
Sorry for your troubles, I think it would have been easier to just install Debian Bullseye on the board and just use the default Debian packages for everything.

srs4511351, 2020-11-18:
Would Debian have up to date ARM-Pi packages? Raspberry does have them, but
GNU Radio 3.8 is too new for them!
The strange thing is that I didn't have this error in previous installation
attempts. Issue #67 is mostly talking about v 3.7.
libcodec2-dev is installed in the package manager
I don't know how to tell if cmake finds the freedv headers (from
libcodec2-dev) during the gnuradio build process. Is there an installation
procedure that you can share with me that will install everything?
I used this which was in the instructions that I followed. This gave me
version 3.8.2.0 which some found did not include the FreeDV headers.
git checkout maint-3.8
git submodule update --init --recursive

I hope I don't have to reinstall GR 3.8.

Here is some info that may help.

gnuradio-config-info --print-all
/usr/local
/usr/local/etc
/usr/local/etc/gnuradio/conf.d
/home/pi/.gnuradio

[enabled-components]
testing-support;python-support;volk;doxygen;sphinx;gnuradio-runtime;gr-ctrlport;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;*
alsa;*
oss;gr-channels;gr-trellis;gr-uhd;gr-utils;gr_modtool;gr-video-sdl;gr-vocoder;gr-wavelet;gr-zeromq
3.8.2.0
cc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
c++ (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
/usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall
-Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall
-Wno-uninitialized

----Steve

kantooon, 2020-11-18:
The README includes some comments regarding gnuradio components (derived by various users from issue #67 . The messages would be output when you run cmake as various API are found.
Again, my suggestion is to use Debian Bullseye. The number of packages available would dwarf anything that Raspberry Pi OS can provide (unless you have very specific requirements to use Raspberry Pi OS). Compilation on Bullseye with the `next` branch is basically straightforward. GNU radio 3.8 out of the box.

srs4511351, 2020-11-18:
I did another cmake. Does it include the right things? I see codec2 and
freedv
I have libcodec2-dev-0.8.1-2

-- Configuring gr-vocoder support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-vocoder support.
--   Override with -DENABLE_GR_VOCODER=ON/OFF
-- Found LIBCODEC2: /usr/lib/arm-linux-gnueabihf/libcodec2.so
-- Could NOT find LIBGSM (missing: LIBGSM_LIBRARIES LIBGSM_INCLUDE_DIRS)

-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * testing-support
--   * python-support
--   * volk
--   * doxygen
--   * sphinx
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gnuradio-companion
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * * jack
--   * * portaudio
--   * gr-channels
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-video-sdl
--   * gr-vocoder
--   * * codec2
--   * * freedv
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components
-- ######################################################
--   * gr-qtgui
-
----Steve

srs4511351, 2020-11-19:
I simply reinstalled GR 3.8 and now qradiolink installs.
I must have initially missed libcodec2

Qradiolink runs, but when I use the same device string that worked in gqrx,
it doesn’t work. It doesn’t even get the RX gain controls like it did
before. I don’t think it is communicating. When I click the X to close the
window, it doesn’t close.
I installed the next branch. Is that bad?

I’m too stubborn to quit on Raspberry Pi OS now. I may still end up with
Debian.

----Steve

On Nov 18, 2020, at 3:32 PM, Steven Sostrom <srs@umich.edu> wrote:

I did another cmake. Does it include the right things? I see codec2 and
freedv
I have libcodec2-dev-0.8.1-2

-- Configuring gr-vocoder support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-vocoder support.
--   Override with -DENABLE_GR_VOCODER=ON/OFF
-- Found LIBCODEC2: /usr/lib/arm-linux-gnueabihf/libcodec2.so
-- Could NOT find LIBGSM (missing: LIBGSM_LIBRARIES LIBGSM_INCLUDE_DIRS)

-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * testing-support
--   * python-support
--   * volk
--   * doxygen
--   * sphinx
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gnuradio-companion
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * * jack
--   * * portaudio
--   * gr-channels
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-video-sdl
--   * gr-vocoder
--   * * codec2
--   * * freedv
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components
-- ######################################################
--   * gr-qtgui
-
----Steve

srs4511351, 2020-11-19:
Hooray! I have qradiolink working!
I installed the "next" branch.
I was in a hurry to leave when I tried it fresh out of the compiler.
I think it took a while to get going, and I adjusted the gains.
The 2 RX Gains, IFGR and RFGR appear after a couple of seconds.
IFGR operates strangely. The graphical control only goes from 50%
rotation to full rotation. The other half doubles back. As you turn it
counter clockwise from 100%, it decreases until the 50% point, then it goes
back up, all the while the actual gain is still going toward 0.

I am hearing bleedthrough from the opposite sideband. This is not happening
with gqrx. I hear a squawking sound like a mistuned SSB signal when they
speak, but they are not in the bandpass, but they are in the sideband not
selected by the bandpass. (LSB selected hears something from the USB a KHz
or so away from the bandpass).

Messing with the gain to where it was way too high caused it to freeze or
just stopped receiving. Clicking the X to close the application did
nothing. I hit CTRL-C in the terminal window.
After restarting qradiolink, the log window says [Fatal] Could not init RX
device, check settings and restart
I guess I'll have to restart the computer.

The menus are white with yellow text...

----Steve

srs4511351, 2020-11-19:
Well, I think I broke it.

After doing a power-off reset, following the events in the
previous message, qradiolink consistently crashes shortly after I click RX.
Is there a setting or settings file that got corrupted when I cranked the
gain until it quit working?
In the info below from the terminal window, you can see a Segmentation
fault.
gqrx still works.

[19/Nov/2020 00:02:44] [Info] Starting qradiolink
[19/Nov/2020 00:02:45] [Debug] Looking up available audio devices
[19/Nov/2020 00:02:45] [Info] Using audio input device default
[19/Nov/2020 00:02:46] [Info] Using audio output device default
[19/Nov/2020 00:02:46] [Info] Setting VOIP bitrate to 24600
[19/Nov/2020 00:02:46] [Info] Using audio input device default
[19/Nov/2020 00:02:47] [Info] Using audio output device default
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl_tcp uhd rfspace soapy redpitaya
[INFO] devIdx: 0
[INFO] hwVer: 255
[INFO] Using format CF32.
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
Segmentation fault
______________________

After setting RX Gain to 99, I get this when I click RX, instead of the
segmentation fault.
Is it sending illegal values to the SDR?

[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail
[ERROR] error in activateStream() - Init() failed: sdrplay_api_Fail

Setting RX Gain to 50 again cases the Segmentation fault when I click RX.

----Steve

srs4511351, 2020-11-19:
Sorry for the roller coaster ride!
I finally found the settings file and renamed it. It works again. I have
the bad file if you want it.

----Steve

On Nov 18, 2020, at 3:32 PM, Steven Sostrom <srs@umich.edu> wrote:

I did another cmake. Does it include the right things? I see codec2 and
freedv
I have libcodec2-dev-0.8.1-2

-- Configuring gr-vocoder support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-vocoder support.
--   Override with -DENABLE_GR_VOCODER=ON/OFF
-- Found LIBCODEC2: /usr/lib/arm-linux-gnueabihf/libcodec2.so
-- Could NOT find LIBGSM (missing: LIBGSM_LIBRARIES LIBGSM_INCLUDE_DIRS)

-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * testing-support
--   * python-support
--   * volk
--   * doxygen
--   * sphinx
--   * gnuradio-runtime
--   * gr-ctrlport
--   * gnuradio-companion
--   * gr-blocks
--   * gr-fec
--   * gr-fft
--   * gr-filter
--   * gr-analog
--   * gr-digital
--   * gr-dtv
--   * gr-audio
--   * * alsa
--   * * oss
--   * * jack
--   * * portaudio
--   * gr-channels
--   * gr-trellis
--   * gr-uhd
--   * gr-utils
--   * gr_modtool
--   * gr-video-sdl
--   * gr-vocoder
--   * * codec2
--   * * freedv
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components
-- ######################################################
--   * gr-qtgui
-
----Steve

kantooon, 2020-11-19:
I can't follow this too closely due to real life commitments, but regarding the SSB filters, they don't have a very sharp transition and have a maximum attenuation of 60 dB. They are optimized for less CPU consumption rather than maximum rejection. Just doubling the amount of taps in a filter can significantly increase the number of instructions being executed.
You can experiment with the filter parameters is src/gr/gr_demod_ssb..cpp to find the optimum values for your usage scenario.

kantooon, 2020-11-19:
The SSB filters had too low attenuation at 200 Hz in stopband, so I've increased the number of taps to 163 and attenuation to 90 dB which I think is sufficient for this window type. Should be fixed in `next`.

srs4511351, 2020-11-19:
Thanks! Your help installing this was quite valuable. My next project is to
document this.
Is there a format or editor that I should use if I want to submit it to
your github site? I Have never done that before.
Cubic SDR has the filter problem too.
When I install it again, I'll check it out.

Are the AGC updates supposed to be in next?

----Steve

kantooon, 2020-11-19:
AGC, SSB filters and the fixes for the weird dial movements (older issue fixed a while ago) are all in the `next` branch. You'll need to pull the changes in first, of course. They will also be part of the next release.
Regarding the docs, just text format and attach here or do a pull request, either is fine. If you can handle the markdown language for headings and other paragraph decorations it's fine, if not I'll handle formatting myself no worries.
Any other issues please open separate tickets, this one is getting hard to follow already.

srs4511351, 2020-11-19:
Thanks. I'm new to github, so I don't know how to pull the changes in.
I must have needed to do that because the AGC isn't working in my
recent build..

----Steve

kantooon, 2020-11-20:
If you are already on branch next, just `git pull`. Otherwise `git checkout next` followed by `git pull`.
This has nothing to do with Github, which is just a web site hosting code. It is intrinsic to git workflows.

srs4511351, 2020-11-26:
I have finally documented a path to install a working system on Raspberry
Pi OS for GNU Radio 3.8 and Qt 5.12.
This is better than just getting QRadioLink working. It is an operating
platform and I have several other SDR packages working.
The attached document only gets you up to running QRadioLink.

On my latest installation:

Built using:

cd ~/
git clone https://github.com/qradiolink/qradiolink
cd qradiolink/
git checkout next
git pull
sh ./build_debian.sh

Is this the right way to get the latest changes? I'm not shure that I got
all of the improvements that you made.

The terminal said that "Auto" was an illegal value for RX antenna. I had to
enter 0 to get the SDR to start. I didn't have to do that with previous
versions.

AM low frequency response is improved
No AGC on SSB

Twice, I have had to rename the qradiolink.cfg file to reset the settings.
I think is was after I entered -1 in RX Freq Correction (ppm)

----Steve

kantooon, 2020-11-28:
You're right, the maximum gain of the AGC was limited to a value that was too low. This has now been corrected, and a new AGC range was introduced: from 0 to 6, where value 0 corresponds to a loop update value of 1 in GNU radio, the fastest attack and decay value.

The value you need to put in the antenna fields depends on the device and the defaults are there just as an example.

Let me know if the SSB filter adjacent channel response is improved.

srs4511351, 2020-11-29:
The installation procedure that I sent works. When I started from scratch
and followed it, I had to make changes that I forgot, but I think it is
good now. However, I may need to work on it more.
While installing SdrGlut, I saw that it displayed some older SoapySDR
modules: 0.6. Later, I upgraded SdrGlut and it started reporting version
0.8, so there must be 2 versions installed. The applications do work.

I installed the latest version of GRadioLink tonight.
AGC is improved. I suggest that you provide a threshold setting. It works
the best when I set RF gain higher. On the stronger signals, there is a
short period of clipping before it acts, even with attack set to 0. It may
start to act quickly, but it takes a while to fully adjust. The audio of
the strong signals clip for a short time. This is improved by turning the
audio down. It seems like it should act quicker.
It may have been working before but I have to adjust the RF gain relatively
high to hear any difference and may not have noticed.
Once, I turned the RF gain up too high and the AGC totally blanked the
radio for several minutes.

The opposite sideband rejection is vastly improved.

The thing about having Auto as a default in RX antenna is that the SDR will
fail to run and new users will think that something is broken. Other
software uses "RX" "0" did work. Maybe a word of warning is good.

The 2 RX Gains, IFGR and RFGR appear when the SDR starts.
IFGR operates strangely. The graphical control only goes from 50%
rotation to full rotation. The other half doubles back. As you turn it
counter clockwise from 100%, it decreases until the 50% point, then it goes
back up, all the while the actual gain is still going toward 0. The range
of numbers for IFGR seems to be from 20 to 59 in other SRD applications.
RFGR goes from 0 to 9.

----Steve

kantooon, 2020-11-29:
This is the behaviour of the AGC block in GNU radio (it's been a source of headaches before too for me). I understand the need for a threshold but the AGC does not give me too many settings to play with. I'll try to see if I can find more usable values for a faster attack rate.
Regarding the behaviour of the dials, it is possible there is still some bug in this 3rd party code that was introduced recently. I'll try to see if I can reproduce it.

If you have attached any file, it is not visible here. I assume you used the email in which case the attachment was lost. You need to attach it here in the Github web interface for it to be visible.

kantooon, 2020-12-01:
I have pushed a commit which should fix the dial rotation visual problem for dials with positive ranges. Thanks for your patience.

srs4511351, 2020-12-02:
Thanks.
The IFGR dial is working correctly, but I can't get any AGC action like I
did after you made the changes.
I did get some really slow AGC action on AM when I changed the RFGR RF gain
control for the SDR. Did those changes get lost?
I really knew that it isn't working when I turned the noisy garbage
disposal on and nearly jumped out of my skin!

As sometimes happens, messing with the gains totally froze my system. This
time, it crashed the SDR and I had to reboot.

----Steve

kantooon, 2020-12-02:
The AGC settings should be stored in the config file, so unless you overwrite it, they should not get lost. However, it is possible your device here might have other settings, possibly a hardware AGC control which is overriden during initialization or after using a stage gain. Hardware AGC control is not implemented in the code, since each device has a different behaviour. Unfortunately since I don't have this hardware I can't reproduce the problems and try to fix, it is just guesswork at this point. Also, you are probably not using the "standard" version of gr-osmosdr, so there may be issues there as well. Unless you can find some resource on the net from people using gr-osmosdr with the same device, or unless you want to dig in the code of gr-osmosdr and troubleshoot, I'm not sure how I could help. The problem is compounded by using SoapyRemote, another layer of complexity below it.
The only devices I can test with are the LimeSDR family, PlutoSDR, RTL-SDR and UHD.

srs4511351, 2020-12-05:
When I installed the version with the fixed gain controls, I renamed
qradiolink.cfg because the SDR didn't start.
Tonight, I restored that file and changed the audio output device and it
runs. Still the AGC isn't working as well as it was after you changed the
threshold. Are there AGC settings in the config file that aren't in the
user interface?

My SDR does have hardware AGC, but if that is off, software AGC would still
work. It does with other SDR software that do have an option for hardware
AGC.

----Steve

kantooon, 2020-12-05:
All AGC settings are exposed in the user interface, there is no hidden setting in the config file.
The most recent change was to enable an even faster setting for attack and decay, with a value of -1 in the GUI. I'll review all the recent commits, perhaps I introduced some regression, but it is unlikely.
I have added a proeminent notification to the README regarding which Pulseaudio sinks and sources should be used for users which have Pulseaudio installed since I found out there is some confusion about the best device setting.

kantooon, 2020-12-08:
Despite being unable to change the behaviour of GNU radio's AGC, I have added Continuous Envelope SSB blocks to the demodulator and I think the behaviour should be improved now (audio will not longer saturate with AGC overshoots, there might be instead some minor distorsion caused by the stretcher).

srs4511351, 2020-12-09:
Yes, the AGC does work much better on SSB now. There is no obvious clipping
while the attack takes hold. With the decay set to 2, the background noise
swishes in and out between syllables as they speak. It sounds like my
Collins 75S-1. Later models came out with AGC that had a slower decay.
When I increase the decay, it doesn't seem to change much. If the decay is
set to 4 or more and I stop/start the SDR, it takes a few minutes for the
AGC to stabilize. Audio is very weak until it does. If I turn it down to 1,
it recovers sooner, then turning it up to a high number has no effect.
noise still swishing in and out between syllables.
It seems strange that the Attack can be set to a negative number.

----Steve

[33 more replies]