qradiolink/qradiolink #84: Installing the SDRplay RSP1A and QRadioLink on a Raspberry Pi

srs4511351, 2020-12-01:
I have attached my procedure to install updated software to support QRadioLink on a Raspberry Pi.

I hope it works for you.

[installing_drivers_and_qradiolink.zip](https://github.com/qradiolink/qradiolink/files/5620425/installing_drivers_and_qradiolink.zip)

kantooon, 2020-12-01:
Many thanks! This is quite a large contribution, can I credit it you by name for it?

srs4511351, 2020-12-01:
Sure.

Steven Sostrom

That way, they know who to complain to when it doesn't work.

----Steve

kantooon, 2020-12-01:
Done. The tutorial was added in the `docs/` directory. Thank you!

closed 2020-12-01 by kantooon

srs4511351, 2020-12-01:
It looks like the way the docs directory handles the document, it is either
quoting the html or just not displaying it as a web page.
All of the tags are visible.

----Steve

kantooon, 2020-12-02:
Github displays the source code of the file unless it uses the markdown (md) extension in which case it parses the markdown language. Don't worry, after cloning the repo the user can open the file manually and it will open in a web browser being displayed normally.

srs4511351, 2021-01-26:
I have updated my installation procedure.
It is attached.
[SDRplay_RSP1A_QRadioLink_on_RaspberryPi.zip](https://github.com/qradiolink/qradiolink/files/5871167/SDRplay_RSP1A_QRadioLink_on_RaspberryPi.zip)

kantooon, 2021-01-28:
Thanks, I'll update it in the repo as soon as I get a chance.