qradiolink/qradiolink #86: Unreadable menus and dropdowns on GTK desktop environments

kantooon, 2020-12-16:
This issue does not affect KDE users.

srs4511351, 2020-12-18:
I have attached a screen shot of QRadioLink on my Raspberry Pi showing the RX Modulation menu with a white background and yellow text.
My desktop environment is LXDE-pi, which is the default.
The theme is called Crux and it has a gray menu background.
I also tried the Adwaita-dark theme that has black menu backgrounds. None of these changed the QRadiolink menu color.
[image]

kantooon, 2020-12-18:
Ok, that looks strange indeed. It seems something in LXDE is overwriting the style for the QComboBox GUI element. I've pushed an update yesterday which fixes a similar but not as sever issue in Gnome, maybe that will help you? Otherwise, I'll need to check at their forums and bugtracker for simlar problems.

srs4511351, 2020-12-18:
The latest version still has the white menu background.

----Steve

kantooon, 2020-12-19:
It's pretty interesting. I just tested LXDE under Debian and the dropdowns are fine. So it must be something specific to the Raspberry Pi OS config.
However I did find out that the memories frame was not displayed correctly under both Gnome and LXDE (fixed now) and that LXDE can't display translucent widgets so the Constellation display is opaque (Gnome handles translucency fine). I guess I should come out of my KDE bubble more frequently.

srs4511351, 2020-12-19:
The Constellation display was translucent until recently. I can’t tell you
exactly when it changed, maybe a week or so. Was it an OS update or a
QRadiolink update?

----Steve

kantooon, 2020-12-19:
It was a change in qradiolink. I have transformed it into a window of its own, previously it was embedded over the spectrum display in a fixed position, now it can be moved around, which can be useful when for instance not running maximized and want to keep the spectrum unobstructed.
Unfortunately LXDE does not handle window translucency well (perhaps a compositor issue).

srs4511351, 2020-12-20:
I tried qradiolink-0.8.3-5 on an old system.
The RX modulation mode menu has a blue background with yellow text.
RX CTSS and the menus in setup have a white background.

P. S. the installation procedure in build_debian.sh has changed and no longer matches the expanded procedure in readme.md.

kantooon, 2020-12-22:
The script and the README have the same commands, it's just a few paths that changed lately.
Regarding the dropdown, it's hard for me to figure out which control does your theme override, because I can't reproduce the problem in Debian's LXDE. Getting the Qt CSS right on various desktops is quite hard because of the way inheritance works, sometimes styles are applied to and abstract item which is not documented in Qt's docs. I won't give up though.

kantooon, 2020-12-22:
There's another issue as well with scrollbars not being displayed in these dropdowns (Gnome at least) despite my explicit request to do so. So you end up with a huge pop-up which is a problem on small screens at least.

srs4511351, 2021-02-15:
I finally decided to try an OS on the Raspberry Pi that is better maintained. It saved hours of build time.
QRadioLink menus look good on the Raspberry Pi4 running Ubuntu 20.04 with the lubuntu desktop.
I was wondering if there was a prerequisite package that may be missing to cause the menu issue?

----Steve

srs4511351, 2021-04-06:
I made a new installation with the new 2021-03-04-raspios-buster-armhf image.
I installed Qt 5.14.1.
The QRadioLink menus now have the blue background and are quite readable. I don't know if it is the new Raspberry Pi OS or if it is the newer version of Qt or even the latest QRadioLink.
I also upgraded to Q 5.15.2 and the menus are still good.

----Steve

kantooon, 2021-04-07:
Thanks for reporting this back. There has been no significant change to qradiolink in this area since last November, so I assume it was just some Qt theme configuration issue on raspios that was resolved meanwhile. I guess we can close this ticket now.

closed 2021-04-07 by kantooon