qradiolink/qradiolink #89: Equipment emission sampling rate setting

longyucheng123, 2020-12-26:
Hello! How do I set the sampling rate of the transmitter (UHD HackRF Limesdr Plutosdr) in Qradiolink?

kantooon, 2020-12-26:
[image]

Hello! Use this dropdown to change sample rates.

longyucheng123, 2020-12-26:
Hello! Can the emission sampling rate be set below 1M?发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月26日周六 15:33收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)

Hello! Use this dropdown to change sample rates.

kantooon, 2020-12-26:
No, sorry, 1 Msps is minimum implemented. Perhaps in the future we'll have lower sample rates.

longyucheng123, 2020-12-26:
Hello! What's the reason why QRadiolink doesn't work properly with hackRF One RX? Stuck in the diagram cannot be closed.发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月26日周六 16:50收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)
No, sorry, 1 Msps is minimum implemented. Perhaps in the future we'll have lower sample rates.

kantooon, 2020-12-26:
Can you expand a bit on what does not work? I don't have such a device but I've heard reports that it works fine.

longyucheng123, 2020-12-27:
Hello! Does QRadiolink support limesdr USB hardware? The description says support for limesdr Mini and LimeNET-Micro。发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月27日周日 02:21收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)
Can you expand a bit on what does not work? I don't have such a device but I've heard reports that it works fine.

longyucheng123, 2020-12-27:
Hello! Thank you very much for designing such an excellent QRadiolink software. I hope to add more transmitting signal modes (such as DRM DRM +, etc.) into the upgraded version, and I also hope to continuously improve the support for SDR hardware driver.发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月27日周日 02:21收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)
Can you expand a bit on what does not work? I don't have such a device but I've heard reports that it works fine.

kantooon, 2020-12-27:
You're welcome!
The LimeSDR USB should be usable too with qradiolink. But I haven't been able to verify this because I don't have one.

closed 2020-12-28 by kantooon

longyucheng123, 2021-01-07:
Hello! How to use it Qradiolink.Appimage When connecting to UHD, the images of UHD cannot be found. What is the default storage path of images?发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月28日周一 19:16收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)
Closed #89.

kantooon, 2021-01-09:
UHD firmware can't be distributed with the appimage, so you need to install UHD manually and it will also bring in the firmware. I think it's best to compile from source when using UHD rather than running the appimage.

longyucheng123, 2021-01-27:
May I ask my digital intercom signal (DMR) in QRadiolink software why not decoding it? See attached figure发自我的华为手机-------- 原始邮件 --------发件人： adrian <notifications@github.com>日期： 2020年12月28日周一 19:16收件人： qradiolink/qradiolink <qradiolink@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [qradiolink/qradiolink] Equipment emission sampling rate setting (#89)
Closed #89.