qradiolink/qradiolink #96: New installation crashes when saving settings

srs4511351, 2021-01-28:
After installing an new system with RSP1A and HackRF One, qradiolinlk crashes any time it tries to save settings, when I click save or just quit QRadiolink with this error message:

terminate called after throwing an instance of 'libconfig::SettingTypeException'
  what():  SettingTypeException
Aborted

I read in the libconfig docs that
A SettingTypeException is thrown when the type of a setting’s value does not match the type requested.

My installation was done much like the previous one except that I added a new HackRF One in addition to my RSP1A.
QRadiolink works fine with just the HackRF connected, but will not save the settings. I'm sure it would work with just the RSP1A connected also.

---Steve

kantooon, 2021-01-28:
I'm a bit busy right now, but can you state which version are you running? Is it by any chance directly from the next branch? Because there are some breaking changes going on there. If you are running from master or 0.8.5-2, then it is some bug somewhere that was previously unseen. Try renaming the config file and restarting it.

srs4511351, 2021-01-28:
If you are busy, do it when you can.
I installed the next branch a couple of days ago.
It was a new installation and it created a new config file with just the
automatically created comment.
I could try an older version.

----Steve

srs4511351, 2021-01-28:
Version 0.8.5-2 works. Take care of it when you have time.

----Steve

kantooon, 2021-01-30:
I haven't been able to reproduce the crash with the tip of the `next` branch, even when starting with a new config file. Can you please attach screenshots to this ticket of both config tabs using the same settings which trigger the crash for you?

srs4511351, 2021-01-31:
I installed next without git pull, which would be the version that I had trouble with.
It was likely done before gnuradio went through its initial...
I renamed the qradiolink.cfg file.

I don't think mattered what I put in the config tabs. I originally tried a lot of things. I don't recall ever having thiswtrouble before.

With hackrf=0 in setup, not having pressed save, and initially pressing RX, it crashed and said 
[FATAL] sdrplay_api_ReleaseDevice() Error: sdrplay_api_ServiceNotResponding
terminate called after throwing an instance of 'std::runtime_error'
  what():  sdrplay_api_ReleaseDevice() Error: sdrplay_api_ServiceNotResponding
Aborted

[image]
[image]

This is what it said when I clicked the save button:

[31/Jan/2021 11:11:17] [Info] Starting qradiolink
[31/Jan/2021 11:11:17] [Debug] Looking up available audio devices
[31/Jan/2021 11:11:18] [Info] Using audio input device default
[31/Jan/2021 11:11:18] [Info] Using audio output device default
[31/Jan/2021 11:11:19] [Info] Setting VOIP bitrate to 24600
[31/Jan/2021 11:11:42] [Info] Using audio output device pulse
terminate called after throwing an instance of 'libconfig::SettingTypeException'
  what():  SettingTypeException
Aborted

----Steve

srs4511351, 2021-01-31:
On another note, I have a new HackRF One and tested transmit with Version 0.8.5-2,

I clicked TX. The TX light on the HackRF came on, TX gains appeared, no transmit output:

[31/Jan/2021 10:42:52] [Info] Starting qradiolink
[31/Jan/2021 10:42:53] [Info] Using audio input device default
[31/Jan/2021 10:42:53] [Debug] Looking up available audio devices
[31/Jan/2021 10:42:53] [Info] Using audio output device default
[31/Jan/2021 10:42:53] [Info] Setting VOIP bitrate to 24600
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio v3.8.2.0-60-g25b63e7e
built-in sink types: uhd hackrf soapy redpitaya file 
Using HackRF One with firmware 2018.01.1
Ncodec2frames: 4 n_speech_samples: 1280 n_codec_bits: 112 nbit: 28  nbyte: 16
Ncodec2frames: 4 n_speech_samples: 1280 n_codec_bits: 112 nbit: 28  nbyte: 16
UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU

--Clicked PTT

UUUUUUU[31/Jan/2021 11:41:12] [Warning] Blocked attempt to set TX frequency outside of configured band limits
UUUU

--Loud series of beeps in the speakers after long delay

ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred

Still no transmit output. HackRF TX light went out when both PTT and TX were cleared.

I am sure that I am in my band limits.
Is there a file that has these limits?

----Steve

kantooon, 2021-01-31:
So many issues I don't know where to start.

  1. [FATAL] sdrplay_api_ReleaseDevice() Error: sdrplay_api_ServiceNotResponding
terminate called after throwing an instance of 'std::runtime_error' is due to an error reported by the SDRplay API. Since I know nothing about it, I can only assume it malfunctioned at some point (perhaps device not connected or USB permissions issue?
  2. terminate called after throwing an instance of 'libconfig::SettingTypeException'
what(): SettingTypeException this is something in qradiolink code, it should not happen, clearly a bug but I couldn't reproduce it so far on the tip of `next` even when starting with a new config. I'll try to look further into it.
  3. Band limits are only implemented for IARU region 1, so if you tried to tune TX above 440 MHz for example it would give this error. There is also the transmit split value, if the receiver is inside the band but the transmit frequency falls outside due to the split, it can also give this error. You can disable the band limits using the TX Limits checkbox in the settings page for now. 
  4. `ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred` this is a message from ALSA, indicating that there is some audio misconfiguration somwhere, most likely on the audio input path since it occured when you tried to transmit. Do you have Pulseaudio installed? If you do, you should be using its dedicated sinks and sources as outlined in OPERATION.md, otherwise if you are using ALSA directly, then it requires some manual configuration (which I'm not very familar with). Nowadays most Linux distributions ship Pulseaudio by default. The fact that you had no TX output probably related to the fact that no audio samples were coming from ALSA so solving this one should give you TX output.

srs4511351, 2021-02-01:
1. I'm not too worried about the fatal sdrplay error, but that it tried to open a device that had never been specified. The default device was rtl=0 and I had entered hackrf=0. The config file was empty because it crashed when I try to save.
2. Thanks. The SettingTypeException error occurred while starting QRadioLink with a good config file. It aborted.
3. I'll disable the band limits and give it a try.
4. I do not have the pci ALSA devices on my Raspberry Pi. Input audio device does have 2 ALSA devices: usbstream:CARD=b1, usbstream:CARD=Headphones. Selecting these results in the following repeated errors:
ALSA lib pcm_usb_stream.c:508:(_snd_pcm_usb_stream_open) Unknown field hint
[1/Feb/2021 00:13:19] [Critical] Raw audio format not supported by backend, cannot capture audio.

Selecting Default or Pulse does not give errors.
It does transmit now with audio set to pulse. I think is is because I unchecked the TX Limits checkbox. The audio is clear.
I was worried because audio is quite choppy with SDRGlut, but it is known to have audio issues.
Like before, the repeated "U"s quit when I press PTT. It displayed the  [Info] TX frequency is outside of configured band limits

----Steve

closed 2021-02-07 by kantooon

srs4511351, 2021-02-15:
I saw that you made some changes for the Raspberry Pi.
I deleted the build directory, did a git pull and re-ran build_debian.sh
On Raspberry Pi OS, I can save settings changes without a crash.

----Steve

commit fe527439 (2021-02-07): Fix integer width on ARM32 platforms (Fixes #96)