qradiolink/qradiolink #99: MMDVM Bridging

g1nsj0h4n, 2021-06-28:
Hi!

Great talk @ SDRA... 

It would be nice to support DMR/DStar/YSF modes in QRadioLink. So how do we go about bridging MMDVM ? 

As you have ruled out mbelib, I guess another option is based on md380 firmware emulation - which I believe MMDVM Bridge also uses. 

Cheers

kantooon, 2021-07-04:
It is possible to achieve, at least to transmit samples from MMDVM to the SDR. I tried a very basic proof of concept and managed to get some DMR decodes this way, but at least DMR is quite tricky because of the strict timing demands which is why it uses a microcontroller. I'll keep this updated with any progress.

g1nsj0h4n, 2021-07-07:
For transmit DMR, [mmdvm-sdr](https://github.com/r4d10n/mmdvm-sdr) had ported the mmdvm firmware to x86/arm, and thereby using FM modulation of generated samples to get decodes on DMR Radio through PlutoSDR / RpiTX / OsmoFL2K...

kantooon, 2021-07-08:
Yeah exactly, so using it, I now have a transmit only duplex hotspot with my plutosdr. Transmission of both timeslots by the pluto is fine, but receiving my bursts is not. At one point I had it decoding only CACH bursts from the MS, but now I lost it somehow. My bet is there are timing issues in mmmdvm-sdr which prevent decoding of what the pluto receives. My own code in qradiolink is on branch mmdvm-integration, and I will push more code but currently busy.

kantooon, 2021-07-08:
Right, so the situation is as follows. The plutosdr works in TX mode as a simplex and duplex DMR hotspot with MMDVM just fine, both timeslots are transmitted correctly to the radio through qradiolink. 
On the other hand, when I try to transmit with my radio towards the Pluto, I can get good decodes for MS access bursts and see Downlink Activate CSBK get sent to MMDVMHost, but timeslot data isn't coming through. If I do a hack to disable the MMDVM logic around timeslots and just send any decoded frame regardless of type, I can get voice transmitted to the DMR network, but it appears on the wrong timeslot, so this leads me to believe there's some timing issue still left when in duplex/repeater mode. Nevertheles I think it might work fine if MMDVM was operating in simplex mode.

The architecture is pretty basic, ZeroMQ is used to push samples towards qradiolink and back, which means there's an added dependency on mmdvm-sdr towards libzmq. 
To operate the whole thing, one needs to use a certain start order: first MMDVM, then MMDVMHost, only then enable TX and RX in qradiolink (mode MMDVM, last one in the list) otherwise there's an issue with ZMQ sinks blocking the flowgraph when the sample buffers are full. Then set Duplex and Split frequency in qradiolink to make sure a duplex hotspot doesn't transmit on top of itself. Frequency does not have to be the one defined in MMDVM.ini. Finally press PTT and it will operate in full duplex mode. 

I also need to push the needed code for mmdvm-sdr, but needs a lot of cleanup, ongoing investigation why the timeslots are not picked up correctly when I transmit, and I should probably contact the original authot for a merge request.
Relevant branch here is `mmdvm_integration`.

kantooon, 2021-07-09:
Duplex hotspot tested, transmits both timeslots since the BS sets the time reference, can receive only timeslot 2 (in DMO mode which is a hack). Radio can transmit on any timeslot but will be picked up as TS 2. Strict MS access time constraints and the gazillions of buffers between MMDVM and the SDR mean there's no way to determine precisely on which timeslot the inbout transmission came form the MS. Ideas, solutions welcome. Outbound FM deviation is not really ETSI standards compliant since the MMDVM sample levels were optimised for different ADCs.

All testing done on PlutoSDR, Debian Bullseye, GNU radio 3.8.2.
See https://github.com/kantooon/mmdvm-sdr for the other side of the integration code. Please report any bugs relevant to the QRadioLink side of the code in separate issues.

g1nsj0h4n, 2021-07-09:
Great... Missed this here... 

 I will try to get the setup back running and follow the instructions to get it working as a duplex TX / DMO RX hotspot... 

Cheers

g1nsj0h4n, 2021-07-10:
So I have a basic setup running on my Arch... I am yet to recieve the transmitted packets from Pluto.. Some issues I noticed:

1) MMDVMHost reports a lot of this:
E: 2021-07-10 14:47:18.382 DMR Slot 2, overflow in the DMR slot RF queue
E: 2021-07-10 14:47:18.382 DMR Slot 2, overflow in the DMR slot RF queue
E: 2021-07-10 14:47:18.382 DMR Slot 2, overflow in the DMR slot RF queue

while the mmdvm-sdr programs gets into the Idle mode and does not change to DMR after initial packets:
D: 2021-07-10 10:47:34.907 IO Int start()
D: 2021-07-10 10:47:35.017 setMode() invoked!
D: 2021-07-10 10:47:47.715 setMode() invoked!
D: 2021-07-10 10:47:47.715 Mode set to DMR
D: 2021-07-10 10:46:49.004 setMode() invoked!
D: 2021-07-10 10:46:49.004 Mode set to Idle

2) In QRadioLink, I can see just a single line in the spectrum on the split transmission, which is a bit off from the expected TX frequency programmed for RX on my DMR radio. How can I change the shift (ppm correction?) ? Also should I not expect a wider waveform if the modulation is happening ?

3) Is there a way to see I'm recieving packets through ZMQ in QRadioLink ? I tried using an external program called zmqcat, which shows some data being sent from mmdvm-sdr.. 

Cheers

kantooon, 2021-07-10:
If you see slot overruns it means no samples are flowing to qradiolink. Check first TX is started in qradiolink, then netstat -tnlp and check both programs are listening on 5990 and 5991.

g1nsj0h4n, 2021-07-10:
I can see CWID messages being transmitted thru QRadioLink... I added a Debug print when ZMQ was sending data from mmdvm-sdr..
So when MMDVMHost times a CWID message: 

E: 2021-07-10 16:40:46.133 DMR Slot 2, overflow in the DMR slot RF queue
E: 2021-07-10 16:40:46.133 DMR Slot 2, overflow in the DMR slot RF queue
M: 2021-07-10 16:40:46.133 DMR Slot 2, received network end of voice transmission to TG 91, 0.8 seconds, 0% packet loss, BER: 0.0%
D: 2021-07-10 16:40:53.119 sending CW ID
E: 2021-07-10 16:40:53.192 DMR Slot 2, overflow in the DMR slot RF queue
E: 2021-07-10 16:40:53.192 DMR Slot 2, overflow in the DMR slot RF queue

At the same time in mmdvm-sdr: 

D: 2021-07-10 16:40:53.119 Message created with length: 87
D: 2021-07-10 16:40:53.119 Message sent by ZMQ.... 
D: 2021-07-10 16:40:53.149 Message sent by ZMQ.... 

And I can see the transmission in QRadioLink... 

I'm quite not able to see the DMR data being transmitted thou.

kantooon, 2021-07-10:
This is strange, because the DMR transmission works perfectly for me. It's the reception which has some quirks. Which version of MMDVMHost are you using? The TX zmq messages should always have 640 short signed samples, so 1280 bytes (buffered before sending).

g1nsj0h4n, 2021-07-10:
Ok.. I think I found the problem... I'd configured the MMDVMHost as Simplex with Slot0 disabled and Slot1 enabled. In my BM Selfcare, I'd routed the TG to TS1, whereas I should have sent it to TS2. Now I tried with Duplex setting and it is transmitting data. Yet to get a DMR decode on my handset. It is blinking and showing signal bars, but no decodes - probably need to tune it around. Do you use some ppm setting or offsets ?

Will update ! I am using MMDVMHost from github and mmdvm-sdr fork from your repo. 

Many thanks for the support..

kantooon, 2021-07-10:
Yes, qradiolink has ppm adjustment in the settings pages. Use a known repeater to figure out the offset.
If you get it working, let me know if your transmissions get through in the first try. For me sometimes my radio says repeater not found and I think it's because of the buffer latencies, and the fact mmdvmhost is not perfectly standards compliant and does not generate idle bursts when CSBK is received. Usually I can get in after a while though.

kantooon, 2021-07-13:
I figured out the reason why my inbound transmissions are sometimes not picked up or are missing a few seconds at the start. It seems the MMDVM SYNC correlator is quite sensitive to being off frequency by as little as 1 kHz, so I need to be less than 300 Hz off-channel in qradiolink. Otherwise works.

g1nsj0h4n, 2021-07-14:
Some progress, but not much... 

When I configure my MMDVMHost with Duplex=0, on TX from my handset - I get this in my mmdvm-sdr log:

D: 2021-07-14 09:59:08.906 DMRDMORX: voice sync found pos/centre/threshold
D: 2021-07-14 09:59:08.906 write DMR data
D: 2021-07-14 09:59:08.966 write DMR data
D: 2021-07-14 09:59:08.966 Frame Start
D: 2021-07-14 09:59:08.966 Frame: 1 231 15
D: 2021-07-14 09:59:08.966 Frame: 149 235 87
D: 2021-07-14 09:59:08.966 Frame: 0 107 253
D: 2021-07-14 09:59:08.966 Frame: 79 251 229
D: 2021-07-14 09:59:08.966 Frame: 166 167 247
D: 2021-07-14 09:59:08.966 Frame: 61 235 178
D: 2021-07-14 09:59:08.966 Frame: 158 10 231
D: 2021-07-14 09:59:08.966 Frame: 119 184 171
D: 2021-07-14 09:59:08.966 Frame: 251 14 171
D: 2021-07-14 09:59:08.966 Frame: 129 255 247
D: 2021-07-14 09:59:08.966 Frame: 213 31 192
D: 2021-07-14 09:59:08.966 Frame: 28 0 0
...
D: 2021-07-14 09:59:09.025 write DMR data
D: 2021-07-14 09:59:09.025 Frame Start
D: 2021-07-14 09:59:09.025 Frame: 2 79 246
D: 2021-07-14 09:59:09.025 Frame: 199 247 184
D: 2021-07-14 09:59:09.025 Frame: 253 183 61
D: 2021-07-14 09:59:09.262 Frame: 251 237 239
D: 2021-07-14 09:59:09.262 Frame: 63 0 0
D: 2021-07-14 09:59:09.263 setMode() invoked!
D: 2021-07-14 09:59:09.263 Mode set to Idle
D: 2021-07-14 09:59:09.267 setMode() invoked!
D: 2021-07-14 09:59:09.267 Mode set to DMR
D: 2021-07-14 09:59:19.296 setMode() invoked!
D: 2021-07-14 09:59:19.296 Mode set to Idle

They are not getting transmitted to the network side through MMDVMHost. When I enable Trace=1 in Modem section, I can see RX DMR packets being printed, without any callsign decoding, etc. 

When I transmit on a Duplex mode frequency (TX handset freq == RX hotspot freq), the hotspot goes into continous transmit mode for a while with constant tone with mmdvm-sdr outputting data like this:

D: 2021-07-14 10:14:04.783 DMRDMORX: voice sync found pos/centre/threshold
D: 2021-07-14 10:14:04.783 write DMR data
D: 2021-07-14 10:14:04.842 write DMR data
D: 2021-07-14 10:14:04.842 Frame Start
D: 2021-07-14 10:14:04.842 Frame: 1 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 15 15 15
D: 2021-07-14 10:14:04.842 Frame: 13 0 0
D: 2021-07-14 10:14:04.901 write DMR data
D: 2021-07-14 10:14:04.901 Frame Start
D: 2021-07-14 10:14:04.901 Frame: 2 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 15 15
D: 2021-07-14 10:14:04.901 Frame: 15 0 0
D: 2021-07-14 10:14:04.931 write DMR data
D: 2021-07-14 10:14:05.551 DMR Lost Here!!!
D: 2021-07-14 10:14:05.551 write DMR lost
D: 2021-07-14 10:14:16.407 setMode() invoked!
D: 2021-07-14 10:14:16.407 Mode set to Idle

In Duplex=0 mode, TX does not otherwise  happen from Pluto (for avoid looping?) for data from network, except for CWID message. Hence I could not check DMR RX in Duplex = 0 mode. 
---
In Duplex=1, I am unable to TX from handset due to handshaking not working. 

When I transmit on Simplex freq from handset, I get this in mmdvm-sdr:
: 2021-07-14 10:30:11.407 DMRIdleRX: data sync found centre/threshold
D: 2021-07-14 10:30:11.768 DMRIdleRX: data sync found centre/threshold
D: 2021-07-14 10:30:14.356 DMRIdleRX: data sync found centre/threshold
D: 2021-07-14 10:30:20.549 DMRIdleRX: data sync found centre/threshold

I am still unable to decode the TX from the hotspot. Tried tweaking around the frequencies.

Does making changes in MMDVM.ini settings (Invert, Levels, etc) make sense ? I did try a bit, but not sure its making any effect.

kantooon, 2021-07-14:
I think I forgot to mention the most important thing: yes, you need to enable both RX invert and TX invert. Here's my Modem section:

[Modem]
Port=/path/to/ttyMMDVM0
Protocol=uart
# Address=0x22
TXInvert=1
RXInvert=1
PTTInvert=0
TXDelay=0
RXOffset=0
TXOffset=0
DMRDelay=0
RXLevel=100
TXLevel=100
RXDCOffset=0
TXDCOffset=0
RFLevel=100
RSSIMappingFile=RSSI.dat
UseCOSAsLockout=0
Trace=0
Debug=3

kantooon, 2021-07-14:
Also, use Duplex=1 in MMDVMHost and Repeater mode on the radio. Simplex should work as well but it's more tricky, because I didn't add any code in qradiolink to handle PTT switching for simplex and it will decode itself with perhaps strange results.

g1nsj0h4n, 2021-07-14:
That's just about it !! Its working very well now.. I'm able to RX and TX to the network thru PlutoSDR as hotspot....Will keep experimenting... 

Many thanks for the support and excellent implementation in such a short while !! 

Cheers...

kantooon, 2021-07-14:
Roger that. Make sure to monitor on the FFT once in while where the center of the radio TX signal is, because these SDRs tend to drift in frequency as they warm or cool. Also make sure the RX gain in qradiolink is set so it doesn't start clipping if you transmit very close to the SDR. I've been bitten by these issues before.

I'll merge this code into the `next` branch soon.

g1nsj0h4n, 2021-07-14:
I'd left the SDR/QRadioLink in hotspot mode and after a long while the radio does not seem to decode the transmissions. So as you said probably there is some freq drift happening. But otherwise its working really well !

alphafox02, 2021-07-17:
I have a PlutoSDR now and running 20.04, so I’ll give this a stab. Unfortunately I don’t own a DMR radio, so I’ll have to borrow. I’d love to include all three pieces in DragonOS once pushed to the next branch.

kantooon, 2021-07-17:
@g1nsj0h4n the problem which occurs after a long time running can also be something related to timing. In mmdvm-sdr there is a kind of problematic sleep() call in the TX loop. That call should not normally be required, but there is some interaction on the TX fifo which I don't yet understand and which can cause slots to be overwritten, especially in simplex only mode. I'll keep looking at that.

kantooon, 2021-07-17:
I've switched ZeroMQ endpoints from TCP to IPC in the latest commits, because two TCP ports were tied up for no good reason.

kantooon, 2021-07-17:
@alphafox02 this feature is now merged in the `next` branch and will be part of the next release.

alphafox02, 2021-07-17:
Awesome I’ll be testing as much of it as I can this weekend. Don’t suppose once I get it all worked in, maybe I could send a link to a test ISO that either of you could boot live and test against real DMR traffic?

g1nsj0h4n, 2021-07-17:
@alphafox02 I don't think it will be any trouble to bake it in... and I'd love to test the Live CD...

alphafox02, 2021-07-17:
Thank you, feedback would be much appreciated. The entire ISO is still a work in progress, but I did confirm just now that it boots after burning using etcher. Qradiolink link and the other two pieces of the puzzle are sitting in /usr/src. I confirmed qradiolink starts (built just awhile ago from next). Mmdvm-SDR is built and I attempted to edit the MMDVMHOST settings to what I thought they would be. SoapySDR and PlutoSDR support is in place. There’s lot of other things preinstalled, but I mainly wanted to see if I got everything spot on in regards to this new development. Test iso is located here  https://drive.google.com/file/d/1Ph185PWrhmr0vw2fyS7HD7o8KBezLePv/view?usp=drivesdk

kantooon, 2021-07-19:
@alphafox02 great work integrating all these SDR applications into your distribution! Unfortunately I don't have the time to run it but I'm sure somebody else will help you test it.

g1nsj0h4n, 2021-07-21:
@alphafox02 I am unable to get the iso to boot... Lubuntu screen comes up and tries to fsck the squashfs file... Then it aborts to dmesg screen, with lots of errors - one of which I remember is stdin Invalid argument. 

Could you check your iso to boot ?

alphafox02, 2021-07-21:
@g1nsj0h4n i didn’t have an issue :/ but I also made a new one for testing. There’s an md5 file to check and make sure there’s no corruption. It’ll boot on legacy or uefi but not secure boot yet. I’ll remove the link after awhile. 

https://drive.google.com/drive/folders/1--rwGIqSs38huxZAOGjKKeU3MTZJjGJc

kantooon, 2021-11-19:
This feature is now released in version 0.8.6, so closing.

kantooon, 2022-04-09:
I've not got a proper TDMA implementation which means that RX works on both timeslots, compliant with ETSI standards, using the LimeSDR (which supports FPGA timestamps). I just need to cleanup the code for a merge.
The branch here is `mmdvm_tdma`.

kantooon, 2025-03-15:
Closing since the feature has been shipped in the last 2 releases. Docs are at https://qradiolink.org/docs/MMDVM/MMDVM.html

closed 2025-03-15 by kantooon