qradiolink/qradiolink #102: Issue connecting to mumble server from Qradiolink only

McPeebs, 2021-09-15:
Hi I have a functioning murmur  I can connect to the same server on the same machine from the standard mumble client. 

When I try to connect in Qradiolink I get 

"**Outgoing connection failed The remote host closed the connection**"

The debug log shows
 "**Mumble message type 18516 not implemented**"

Murmur log 
**<W>2021-09-15 15:50:18.484 1 => <37:(**redacted callsign**)(-1)> Connection closed: The remote host closed the connection [1]
 

 <W>2021-09-15 15:50:18.486 1 => CELT codec switch ffffffff8000000b ffffffff80000010 (prefer ffffffff80000010) (Opus 1)
 
 <W>2021-09-15 15:50:31.507 1 => <38:(-1)> Connection closed: Error during SSL handshake: error:1408A0C1:SSL routines:ssl3_get_client_hello:no shared cipher [13]**

kantooon, 2021-09-16:
qradiolink uses umurmur (https://github.com/qradiolink/umurmur or https://github.com/umurmur/umurmur ) as a server. Murmur implements additional handshake steps which are not documented in the public protocol specs. In addition, umurmur is less complex and requires fewer resources to run.
In future versions it is possible that Murmur compatibility will also be available.

McPeebs, 2021-09-16:
Ah yes I see now that there is a specific maintained version for umurmur. I'll install that, thanks!

kantooon, 2021-09-17:
Glad to be of assistance. Murmur compatibility will be added in a future version as well.

closed 2021-09-17 by kantooon