qradiolink/qradiolink #103: (Request) mobile interface

alphafox02, 2021-11-07:
More of something I thought I’d share, but if there’s a way to make the interface more user friendly on a mobile device that’d be great. 

Here’s qradiolink running on Mobian on my PinePhone. 
[image]

kantooon, 2021-11-07:
Indeed I do intend to work on some UI widgets for Plasma Mobile. Can't say how soon though, but it's on my list. I think it's more likely to be an external application using the remote command API.

alphafox02, 2021-11-07:
That’d be great. I was rotating and doing everything I could to get to some of the buttons. Maybe if I had just plugged a mouse in I could’ve blindly clicked on the rx button. You mention the remote command api - maybe I need to look into that.

kantooon, 2021-11-08:
@alphafox02  I'm not sure what sort of resoluton your device supports, but there are two entries in the config file ~/.config/qradiolink/qradiolink.cfg that you can play with: window_height and window_width.
Here is an example window configuration where UI elements are still shown correctly:
window_width = 1390;
window_height = 880;

The height is not crucial, at most you'll lose some pixels in the spectrum display, but the width needs to accomodate the lower control elements.
Let me know if this helps.

alphafox02, 2021-11-11:
To follow up, I can change the resolution manually in the config file. The problem is that when I open it the settings are somehow reset to  720/288 (I check the config file after closing). I'll keep trying various settings and see if any resolution combination sticks.

kantooon, 2021-11-13:
It looks like the window manager is automatically resizing the window to a resolution that it thinks it supports (which is probably way lower then what the phone actually supports).

I'm currently working on LimeRFE device support but after I'm done with that I'll take a look at the issues with KDE Plasma Mobile and see if I can do anything about it without actually writing new GUI code.

kantooon, 2021-11-17:
@alphafox02 it looks ok on KDE Plasma Mobile if screen is scaled to 100%

https://user-images.githubusercontent.com/968724/142250817-5fd0de37-6836-4e6e-b957-293a7d588f87.mp4

Pinephone, OpenSUSE Tumbleweed, ADALM-Pluto. I'm using Martin Hauke's packages from hardware:sdr
The controls are very small though, almost unusable on the touchscreen at this resolution.

alphafox02, 2021-11-17:
That’s awesome. I’ll try again on my end. The screen scaling was a setting in that OS itself?

kantooon, 2021-11-18:
Yes, it's a control in display settings. In Opensuse's KDE it's set to 200% by default which halves th effective resolution.
But I wouldn't bother with it just yet. It eats through the battery at a scary pace and the controls are barely usable. There's  clearly a need for a new UI, as well as some improvements for CPU eficiency. volk_profile didn't seem to do it's usual magic there.