qradiolink/qradiolink #106: Documentation for radio over IP voice forwarding mode is unclear

kantooon, 2021-12-31:
### Discussed in https://github.com/qradiolink/qradiolink/discussions/98

As discussed in this topic, the documentation is not clear regarding operation in Radio over IP mode, and needs updating with information regarding setup for this mode and how it works, so the user will not try to operate it by toggling PTT manually since qradiolink will operate completely unattended once it is set up correctly.

Also it would be good to include examples for the most common scenarios, like analog/digital, simplex/duplex or repeater mode with TX shift.

Also document most frequent issues encountered (network delays, packet loss, transcoding latency etc.)

Buronn, 2023-07-03:
Im agree with this.

kantooon, 2023-07-05:
We can try to improve it together. Tell me how you want to use the feature and I'll try to guide you through it.