qradiolink/qradiolink #107: corrupted size vs. prev_size in fastbins

bahari, 2022-01-10:
Hi, I'm just set up DragonOS and try QRadiolink with my limeSDR. I manage to received wide FM broadcast with clear audio. But when I disable the RX radio button to stop the receiving, QRadiolink start to abort and it's throw SIGABRT signal from OS stated "corrupted size vs. prev_size in fastbins" and the application exit. Any hint why this happen or is it normal?

kantooon, 2022-01-10:
It's not normal, there is some memory corruption going on there, that's why it receives SIGABRT. But I can't reproduce this locally, so it must be something specific to your environment. Can you provide a gdb backtrace so we can learn more? I think that this could be related to gr-osmosdr.

bahari, 2022-01-10:
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from qradiolink...
(gdb) run
Starting program: /home/bahari/MyWorks/Projects/radio-dev/sdr/source-code/build-qradiolink-Desktop-Debug/qradiolink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffee0ec700 (LWP 29970)]
[New Thread 0x7fffe7fff700 (LWP 29971)]
[10/Jan/2022 16:23:27] [Info] Starting qradiolink
[New Thread 0x7fffe73fd700 (LWP 29972)]
[New Thread 0x7fffe6bfc700 (LWP 29973)]
[New Thread 0x7fffe63fb700 (LWP 29974)]
[New Thread 0x7fffe53f9700 (LWP 29976)]
[New Thread 0x7fffe5bfa700 (LWP 29975)]
[New Thread 0x7fffe4bf8700 (LWP 29977)]
[New Thread 0x7fffc75ef700 (LWP 29978)]
[10/Jan/2022 16:23:28] [Debug] Looking up available audio devices
[New Thread 0x7fffc2aa5700 (LWP 29979)]
[Thread 0x7fffc2aa5700 (LWP 29979) exited]
[10/Jan/2022 16:23:32] [Info] Using audio input device default
[New Thread 0x7fffc2aa5700 (LWP 29980)]
[New Thread 0x7fffc22a4700 (LWP 29981)]
[Thread 0x7fffc22a4700 (LWP 29981) exited]
[10/Jan/2022 16:23:37] [Info] Using audio output device default
[New Thread 0x7fffc22a4700 (LWP 29982)]
[10/Jan/2022 16:23:39] [Info] Setting VOIP bitrate to 24600
[New Thread 0x7fffc0c03700 (LWP 29983)]
[New Thread 0x7fff822ed700 (LWP 29984)]
[New Thread 0x7fff81aec700 (LWP 29985)]
[New Thread 0x7fff812eb700 (LWP 29986)]
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[New Thread 0x7fff7393b700 (LWP 29987)]
[New Thread 0x7fff7313a700 (LWP 29988)]
[New Thread 0x7fff72939700 (LWP 29989)]
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D4C42EEABAC8D'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
[INFO] Selected RX path: LNAW
[INFO] RX LPF configured
[New Thread 0x7fff72138700 (LWP 29991)]
[New Thread 0x7fff71937700 (LWP 29992)]
[New Thread 0x7fff70fe2700 (LWP 30011)]
[New Thread 0x7fff5ffff700 (LWP 30012)]
[New Thread 0x7fff5f7fe700 (LWP 30013)]
[New Thread 0x7fff5effd700 (LWP 30014)]
[New Thread 0x7fff5e7fc700 (LWP 30015)]
[New Thread 0x7fff5dffb700 (LWP 30016)]
[New Thread 0x7fff5d7fa700 (LWP 30017)]
[New Thread 0x7fff5cff9700 (LWP 30018)]
[New Thread 0x7fff3ffff700 (LWP 30019)]
[New Thread 0x7fff3f7fe700 (LWP 30020)]
[New Thread 0x7fff3effd700 (LWP 30021)]
[New Thread 0x7fff3e7fc700 (LWP 30022)]
[... 309 lines cut ...]
#16 0x00005555555c014d in boost::detail::shared_count::~shared_count (this=0x7fffd8007ae8, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:427
#17 0x00005555556e2486 in boost::shared_ptr<gr_demod_freedv>::~shared_ptr (this=0x7fffd8007ae0, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/shared_ptr.hpp:341
#18 0x00005555556d437e in gr_demod_base::~gr_demod_base (this=0x7fffd8007810, __in_chrg=<optimized out>) at ../qradiolink/src/gr/gr_demod_base.cpp:156
#19 0x00005555556d46b2 in gr_demod_base::~gr_demod_base (this=0x7fffd8007810, __in_chrg=<optimized out>) at ../qradiolink/src/gr/gr_demod_base.cpp:160
#20 0x0000555555646a4e in gr_modem::deinitRX (this=0x55555a049890, modem_type=9) at ../qradiolink/src/gr_modem.cpp:88
#21 0x000055555562a463 in RadioController::toggleRX (this=0x555559fe0270, value=false) at ../qradiolink/src/radiocontroller.cpp:1732
#22 0x000055555573bf7f in RadioController::qt_static_metacall (_o=0x555559fe0270, _c=QMetaObject::InvokeMetaMethod, _id=53, _a=0x55555a2ecc40) at moc_radiocontroller.cpp:625
#23 0x00007ffff4db0d5a in QObject::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#24 0x00007ffff595fa66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff59690f0 in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#26 0x00007ffff4d8493a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#27 0x00007ffff4d875b8 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#28 0x00007ffff4ddcf67 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#29 0x00007ffff1df717d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff1df7400 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#31 0x00007ffff1df74a3 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#32 0x00007ffff4ddc5e4 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
[... cut ...]
#34 0x000055555573bb53 in RadioController::qt_static_metacall (_o=0x555559fe0270, _c=QMetaObject::InvokeMetaMethod, _id=32, _a=0x7fffe73fbec0) at moc_radiocontroller.cpp:604
#35 0x00007ffff4db0300 in QMetaObject::activate(QObject*, int, int, void**) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#36 0x00007ffff4bba5ae in QThread::started(QThread::QPrivateSignal) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#37 0x00007ffff4bbc9ba in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x00007ffff5fb0609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#39 0x00007ffff46e0293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb)

bahari, 2022-01-10:
Above are the result when run GDB with backtrace. When debug the code via Qt Creator, the SIGABRT signal occur at function gr_demod_freedv(), and then the debugging stops. Attached are the screen shot.
[image]
[image]

bahari, 2022-01-10:
For your information before this I'm using other linux distribution and compile QRadiolink directly from source using your good instruction. Then I debug it using Qt Creator but I'm getting the same results, SIGABRT signal. So at first I thought it was because my other components are not correctly compile and download. Then I setup another machine using Dragon OS (LUbuntu) where it is comes mainly for SDR usage and development, but the results still the same.

kantooon, 2022-01-10:
You're using an older version of libcodec2 which had this issue: https://github.com/drowe67/codec2/issues/102
This is fixed in Debian since last year with libcodec2 version 0.9.2-4.
Updating your libcodec2 and libcodec2-dev packages should resolve this. 
See changelog at https://debian.pkgs.org/11/debian-main-arm64/libcodec2-0.9_0.9.2-4_arm64.deb.html

bahari, 2022-01-10:
Thanks for the quick response. For your information my machine are not using arm processor, its x86_64. How I can update it because its already updated. Below are my terminal log:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
libcodec2-0.9 is already the newest version (0.9.2-2).
The following packages were automatically installed and are no longer required:
  fonts-mathjax idle-python3.8 libjs-mathjax
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 101 not upgraded.

bahari, 2022-01-10:
If anybody who come to this problem just goes to below link:
[https://answers.launchpad.net/~savoury1/+archive/ubuntu/multimedia/+build/21133628](url)

And download all the below update and install it by dpkg -i command on the terminal:
[https://answers.launchpad.net/~savoury1/+archive/ubuntu/multimedia/+build/21133628/+files/codec2-examples_0.9.2-4~20.04.sav0_all.deb](url) 
[https://answers.launchpad.net/~savoury1/+archive/ubuntu/multimedia/+build/21133628/+files/codec2_0.9.2-4~20.04.sav0_amd64.deb](url)
[https://answers.launchpad.net/~savoury1/+archive/ubuntu/multimedia/+build/21133628/+files/libcodec2-0.9_0.9.2-4~20.04.sav0_amd64.deb](url)
[https://answers.launchpad.net/~savoury1/+archive/ubuntu/multimedia/+build/21133628/+files/libcodec2-dev_0.9.2-4~20.04.sav0_amd64.deb](url)

kantooon, 2022-01-10:
Happy to be of assistance. For any questions, discussions etc. feel free to use the new mail list at https://groups.io/g/qradiolink (preferred) or the Discussions here at github.

closed 2022-01-10 by kantooon