qradiolink/qradiolink #116: qradiolink v0.8.7 debian package is broken

jJit0, 2022-09-29:
Hi,

Try to launch the new version of qradiolink, I got this error
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  rpcmanager: Aggregator not in use, and a rpc booter is already registered

Aborted
```
This happened after updating qradiolink_0.8.6.deb to qradiolink_0.8.7.deb

any ideas ?

kantooon, 2022-10-01:
I don't know if the package is broken. I can't reproduce this issue on two different installs.
The error is a GNU radio runtime error unrelated to qradiolink, see this message thread: https://lists.gnu.org/archive/html/discuss-gnuradio/2014-07/msg00128.html
So maybe give a little more details about your environment and GNU radio installation.

jJit0, 2022-10-01:
Yeah it is gnuradio issue, cuz even after downgrading to v0.8.6 I still have the same error.

I use kali linux btw, and it’s surely because kali updated gnuradio that broke qradiolink.

I have to dig deeper to find the root cause

alphafox02, 2022-10-18:
Kali is probably on 3.10, not 3.8. Qradiolink is still setup for 3.8 right? Keeping an eye out for when it’s working on 3.10 and Qt 5.15.

jJit0, 2022-10-19:
That's right, kali upgraded gnuradio and all libgnuradio-* has to v3.10.5

If I try to downgrade gnuradio to 3.8.0, it'll break other tools.

So we'll have to wait for qradiolink to be compatible with v3.10.5

kantooon, 2022-10-30:
I've started a port to Bookworm and 3.10, but it's going to take a while. Lots of stuff breaks with the new versions.

alphafox02, 2022-10-30:
Can’t wait to test and add. Qradiolink is on my list to add to Jammy (3.10.4 GR).

kantooon, 2022-10-30:
@alphafox02 I've pushed a new branch `gr_3.10` which contains changes to enable build with gnuradio 3.10. This branch builds on Bookworm.
Note I've done nothing to support Qt6 yet.

kantooon, 2023-08-18:
No updates, closing.

closed 2023-08-18 by kantooon