qradiolink/qradiolink #118: Hello, I have a little problem

guet-clg, 2022-12-15:
After I transplanted it to my computer, the header file opus/opus. h in ausioencoder. h was not found. stay https://www.opus-codec.org I downloaded it in but still haven't solved it. What can I do?

kantooon, 2022-12-17:
Which Linux distribution are you using and what release? Opus headers are available in the libopus development package, commonly suffixed with -dev (libopus-dev in Debian). On other flavours of Linux the development package can be named differently.

guet-clg, 2022-12-17:
I used Ubuntu 20.04.03

kantooon, 2022-12-17:
It's libopus-dev as well: https://packages.ubuntu.com/search?suite=all&section=all&arch=any&keywords=libopus&searchon=names
Follow the installation instructions for Debian as Ubuntu is similar in many ways and if a package is not found use that search engine to check if it exists under an alternative name.

guet-clg, 2022-12-17:
Thank you. This problem has been solved！

closed 2022-12-18 by kantooon