qradiolink/qradiolink #120: HackRF One TX works, RX not...

LaneaLucy, 2023-06-08:
Using HackRF One with firmware 2018.01.1
terminate called after throwing an instance of 'zmq::error_t'
  what():  Die Adresse wird bereits verwendet
Abgebrochen (Speicherabzug geschrieben)

LaneaLucy, 2023-06-08:
some other device string, same behavior...

[INFO] Opening HackRF One #0 325866e629417923...
terminate called after throwing an instance of 'zmq::error_t'
  what():  Die Adresse wird bereits verwendet
Abgebrochen (Speicherabzug geschrieben)

kantooon, 2023-06-08:
Thanks for the report. Which version are you using? If using a specific branch instead of release number, please specify the commit hash from which you build the binary.

The error seems to be related to a ZeroMQ socket address already being in use.
Stop qradiolink, remove all "mmdvm*.ipc" files from /tmp and try running again. This error is unlikey to occur if you have only one instance of qradiolink in use though.

LaneaLucy, 2023-06-08:
I downloaded the latest appimage. 0.8.9-1

This worked, there was 5 files

kantooon, 2023-06-08:
You might have been using a very old git hash, this problem should not exist anymore as TCP sockets have been changed to IPC for zmq since at least two releases ago.

LaneaLucy, 2023-06-08:
Like I said, I used the newes release appimage

kantooon, 2023-06-08:
Ok, I understand now. I could not reproduce the issue with the normal Debian package, so for now I'll assume it's specific to the appimage. Can you please provide the full console log from the startup until the issue occurs?