qradiolink/qradiolink #121: .desktop file application categorization

drws, 2023-08-17:
The [`.desktop` file](https://github.com/qradiolink/qradiolink/blob/next/qradiolink.desktop#L11) currently categorizes QRadioLink as an _educational_ application. While that's true, it's still not a proper use of the category. Current `.desktop` makes QRadioLink not displayed alongside other SDR applications.

The [XDG application category registry](https://specifications.freedesktop.org/menu-spec/menu-spec-1.0.html#category-registry) is not the most versatile categorization ever. Not many available categories are suitable for QRadioLink and after investigating `.desktop` files of different SDR apps, the proposed solution is simply:

    Categories=HamRadio;Network;

_HamRadio_ is de facto main category of SDR-type apps. One could say that SDR _hackers_ are a new generation of _hams_, so there's also some logic behind. Other partly-suitable candidates would be _Telephony_ or _DataVisualization_, as a second category alongside _HamRadio_.

kantooon, 2023-08-17:
I agree with you. I'll change the category as you have proposed.

kantooon, 2023-08-18:
Fixed in version 0.8.10

closed 2023-08-18 by kantooon