qradiolink/qradiolink #123: [Feature request] HamTetra Tetra DMO repeater integration

Radiomix2000, 2023-08-20:
There is project which implements TETRA DMO repeater using Lime SDR and Lime mini.
https://github.com/rats-ry/HamTetra
This one was made for HamTetra community. It works but there is lack of communication with external world.
If you integrate it to Qradiolink, it will be possible to connect and implement "all with all" concept using Mumble or SVXLink.
These guys who using Motorola MTM5400 as DMO repeater, using SVXLink with it. But HamTetra DMO repeater implementation doesn't allow to connect it to external world. It would be really cool, if you could integrate TETRA DMO repeater functionality into QRadiolink.
As I know there are TETRA repeaters in your location, at least in Bucharest.

kantooon, 2023-08-20:
Sounds cool, but what exactly would such an integration involve? Do you mean communicate with it via some protocol?

Radiomix2000, 2023-08-20:
To make something like tetra reflector.
Some kind of room (like in Mumble or svxlink), where can somebody connect over internet, or from some other protocol, like dmr, or just analog. 
 To take Rx audio from and send Tx audio to the DMO repeater. That's seems to be the most important thing, which is not implemented there. When we have both rx and tx audio streams we can forward/get them using mumble client connected to the server.

kantooon, 2023-08-21:
I understand. It seems like there would be some overlap with svxreflector or umurmur no? These servers already provide the type of rooms you mention. Unless you mean no transcoding to Opus involved, in which case there is indeed no such server.
In any case, I'd encourage you to join the mailing list for such discussions. Due to an issue in Github, I don't get any notifications from here.