qradiolink/qradiolink #124: Looking for DMR modem

caipiblack, 2023-11-05:
Hello,

I'm working to create a software to send/receive audio/messages using DMR protocol.

Currently I arrive to do the RX/demod using a uggly flow graph (GQRX UDP > GNU Radio flowgraph (verry uggly and CPU consumming) > my software)

Currently I arrive to decode audio & "messages" (transmitted from Retevis RT3 devices: basically it's like IP packets)

My project was in standby during a long time because I don't arrive to do the TX and my RX (& demod) is not very good

I come here because it seems that you made it, but I don't know where to look at.

Do you have a modem class that can do RX/TX for DMR protocol that I can use for my software ?

Is it possible to use QRadioLink as "modem" ? Like writing/receiving demodulated packets to/from UDP/ZeroMQ ?

Note: I don't need mmdvm stuff.

caipiblack, 2023-11-05:
For information this was the GRC that I was using for the RX & demod on the past:

[image]

The missing blocks are just because currently I am on a new computer. But basically it's:

- LimeSDR sink (RX from limesdr)
- A block to transmit packets > UDP (udp client)

With this I don't have the timestamp mechanism and it takes too much CPU usage just for RX.

kantooon, 2023-11-06:
A DMR user radio has been planned for a while, but nothing was done yet (mostly because of the situation of the audio codec). What's currently in qradiolink is base station /repeater operating mode with one or multiple RF carriers, but only together with MMDVM-SDR. If you're interested in the timestamping code, you can find it in the BurstTimer and gr_mmdvm_source  + gr_mmdvm_sink classes.

 IIRC, OP25 had a working DMR transmitter, but only for DMO. Also OP25 has a fully working receiver including for tier III trunking, and it's 100% GNU Radio.

caipiblack, 2025-03-15:
Hello,

Yes i am still interested, my project was again in pause as i don’t arrive to implement the TX side correctly (transmit at correct timing etc) and RX side is very uggly, 

When i have some time i will look into the branch that you mentioned

Thanks

kantooon, 2025-03-15:
You're welcome. Works with the LimeSDR-mini so far, there are still timing issues with the USRP that I haven't figured out. There's no documentation yet but there's a short demo at https://www.youtube.com/watch?v=h4YrMieKY3Y

kantooon, 2025-04-05:
This modem now released in 0.9.1-1, so I'm closing this issue.

closed 2025-04-05 by kantooon