qradiolink/qradiolink #125: Rename QRadioLink desktop icon

AsciiWolf, 2024-03-09:
The desktop [icon](https://github.com/qradiolink/qradiolink/blob/next/src/res/icon.png) file name (and its Icon field in [qradiolink.desktop](https://github.com/qradiolink/qradiolink/blob/next/qradiolink.desktop))  is currently just "icon". This is not correct and will cause many issues on a standard Unix-like system when installed system-wide. Please, change the icon file name to a standard "qradiolink".

kantooon, 2024-03-10:
Ok, thanks for letting me know. It's not very clear to me if the filename needs to be changed from icon.png -> qradiolink.png and if qradiolink.desktop needs an entry like Icon=qradiolink or needs to be Icon=qradiolink.png
I'd appreciate if you could clarify, thanks!

AsciiWolf, 2024-03-10:
The icon name should always be without extension. The only exception is if a full icon path is specified.

commit 2a993687 (2024-03-10): Modify window and desktop icon, fixes #125
closed 2024-03-10 by kantooon