qradiolink/qradiolink #127: build issue on debian

ksbjr, 2024-04-01:
../../qradiolink/src/gr/gr_mod_base.h:34:10: fatal error: gnuradio/filter/rational_resampler_base.h: No such file or directory
   34 | #include <gnuradio/filter/rational_resampler_base.h>
      |

kantooon, 2024-04-02:
Which version of Debian and which branch are you trying to build?
It looks like you are trying to build from `next` or `master` on Debian Bookworm with GNU Radio 3.10. If that is the case, please use branch gr_3.10 which is compatible with this version of GNU Radio.

ksbjr, 2024-04-02:
gr_3.10 built thanks

kantooon, 2024-04-21:
Sorry, I've been busy with other things, I'll take a look at this, before tagging a new release.

D3vil0p3r, 2024-11-18:
> the one issue is the renamed the filegnuradio/filter/rational_resampler_base.h > gnuradio/filter/rational_resampler.h this should fix next for debian
 
@kantooon I tried to replace it, but building still getting error due to other gnuradio missing headers:
```
In file included from ../../qradiolink/src/gr/gr_demod_base.h:50,
                 from ../../qradiolink/src/gr_modem.h:37,
                 from ../../qradiolink/src/radiocontroller.h:47,
                 from ../src/main.cpp:35:
../../qradiolink/src/gr/gr_demod_bpsk.h:28:10: fatal error: gnuradio/digital/cma_equalizer_cc.h: No such file or directory
   28 | #include <gnuradio/digital/cma_equalizer_cc.h>
```
I'm using GNURadio 3.10.11.0. Is qradiolink using deprecated headers of old GNURadio versions?

kantooon, 2024-11-19:
I don't know who you're askking that question to, but it's not me, since I never said anywhere to replace anything in the code. Read my first message above to clear up your confusion.

D3vil0p3r, 2024-11-19:
>I'm using GNURadio 3.10.11.0. Is qradiolink using deprecated headers of old GNURadio versions?
>
I am asking this question to you because you appear as a Collaborator of the project and last committer. If you don't know the answer, nevermind.

D3vil0p3r, 2024-11-19:
Genuine question: did you sleep bad last night?

kantooon, 2024-11-19:
Yes, am sick with the flu, hence my reply in the middle of the night and in a somewhat harsh tone. Anyways, it seems it was unintentional, no hard feelings, misunderstandings do happen. There are better things to do than get worked up over what someone said on the Internet.