qradiolink/qradiolink #130: LimeSuite-NG legacy API support issue (Question)

alphafox02, 2025-03-04:
This is not necessary a problem with QRadioLink, as it was built for LimeSuite and not LimeSuite-NG.

I just wanted to point out that I was working on building QRadioLink on 24.04 with GNU Radio 3.10.11, however, I've installed LimeSuite-NG and so QRadioLink makes various complaints about the older .h files and other files typically in /lime are missing. 

Can QRadioLink be built without direct/native lime support until such time that possibly LimeSuite-NG is supported?

alphafox02, 2025-03-04:
Just recalled it’s probably critical for native LimeSDR support to be included. I’ll look at getting LimeSuite and LimeSuite-NG both installed to some extent as Myriad said it’s possibly to have both apis/so files present.

alphafox02, 2025-03-04:
I was able to install LimeSuite (disabled Soapy, GUI) alongside LimeSuiteNG without a conflict and in turn build QradioLink on 24.04 for testing.

closed 2025-03-04 by alphafox02

kantooon, 2025-03-06:
Ok, thanks for letting me know. Just FYI, the LimeSuiteNG and new gnuradio blocks API don't support all the features of the old one yet, but it's being worked on. For now only the old LimeSuite library should be used.

And since we're here, I know you were also packaging MMDVM-SDR at some point, and I thought I'd let you know that if you want to have all the functionality that is compatible with the latest qradiolink version, you should use the guide located here: https://qradiolink.org/docs/MMDVM/install.html
Of course, ignore the qradiolink build guide since you've already taken care of that. I've had people come to me trying to use your MMDVM-SDR package and it did not work for them (seemed to be some other older version that was not compatible).

alphafox02, 2025-03-06:
Thank you so much, I'll go follow that guide right now and update the MMDVM. Question, now that you mention the blocks, is it okay that I just have the NG soapy vs NG plus the normal soapy block? I was going to also install back gr-limesdr so both the NG and old blocks would be there. I just got to wondering, how does QradioLink get to using the Lime? Is it via Soapy, GR-LimeSDR, or is it direct to the LimeSuite API native somehow?

alphafox02, 2025-03-06:
On 24.04 (gcc 13) I had to perform the following to 
MMDVMHost-SDR

Added #include <cstdint> to both NullController.cpp and M17Utils.cpp in order for uint64_t and uint8_t to be recognized. Without that, there was a few errors.

alphafox02, 2025-03-06:
One question, on MMDVMHost-SDR, after it's build, do I need to keep around the Nextion folders?

kantooon, 2025-03-06:
> Thank you so much, I'll go follow that guide right now and update the MMDVM. Question, now that you mention the blocks, is it okay that I just have the NG soapy vs NG plus the normal soapy block? I [...]

Support is provided in two ways:
1. Direct via gr-limesdr blocks included in the tree (somewhat different to originals)
2. Indirect via gr-osmosdr -> SoapySDR -> SoapyLMS7

The first way (no soapy=.. in the menu) is used for TDMA modes.

kantooon, 2025-03-06:
> On 24.04 (gcc 13) I had to perform the following to MMDVMHost-SDR
> 
> Added #include to both NullController.cpp and M17Utils.cpp in order for uint64_t and uint8_t to be recognized. Without that, there was a few errors.

That shouldn't be needed, stdint headers are usually included upstream.
And no, you don't need any Nextion stuff, just the binary and the config files.

alphafox02, 2025-03-06:
I wonder what’ll happen since I only have the NG soapy implementation currently installed. Gr-osmsodr should be able to use it just the same, but I don’t know how it’ll perform.