qradiolink/qradiolink #133: SVXLink audio is broken

LaneaLucy, 2025-05-30:
constantly getting
```
[30/Mai/2025 11:31:48] [Critical] Raw audio format not supported by backend, cannot play audio.
[30/Mai/2025 11:31:50] [Critical] Raw audio format not supported by backend, cannot capture audio.
[30/Mai/2025 11:31:51] [Critical] Raw audio format not supported by backend, cannot play audio.
[30/Mai/2025 11:31:52] [Critical] Raw audio format not supported by backend, cannot capture audio.
[30/Mai/2025 11:31:54] [Critical] Raw audio format not supported by backend, cannot capture audio.
[30/Mai/2025 11:31:54] [Critical] Raw audio format not supported by backend, cannot play audio.
```
messages and no audio is transmitted.
svxlink.conf
```
[GLOBAL]
#MODULE_PATH=/usr/lib/x86_64-linux-gnu/svxlink
LOGICS=SimplexLogicMic,QRadioLinkSimplexLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%c"
CARD_SAMPLE_RATE=48000
CARD_CHANNELS=1
#LOCATION_INFO=LocationInfo
LINKS=MICToQRadioLink

[SimplexLogicMic]
TYPE=Simplex
RX=LocalRx
TX=LocalTx
MODULES=ModuleHelp,ModuleParrot
CALLSIGN=MYCALL
#SHORT_IDENT_INTERVAL=60
#LONG_IDENT_INTERVAL=60-DUSE-QT=off
#IDENT_ONLY_AFTER_TX=4
#EXEC_CMD_ON_SQL_CLOSE=500
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
#RGR_SOUND_DELAY=0
#RGR_SOUND_ALWAYS=0
#REPORT_CTCSS=136.5
#TX_CTCSS=ALWAYS
#MACROS=Macros
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
#ACTIVATE_MODULE_ON_LONG_CMD=4:EchoLink
#QSO_RECORDER=8:QsoRecorder
#ONLINE_CMD=998877
MUTE_RX_ON_TX=0
#MUTE_TX_ON_RX=0
#STATE_PTY=/var/run/svxlink/state
#DTMF_CTRL_PTY=/dev/shm/simplex_dtmf_ctrl

[QRadioLinkSimplexLogic]
TYPE=Simplex
RX=QRadioLinkRx
TX=QRadioLinkTx
MODULES=
CALLSIGN=XX0XXX
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=30
#IDENT_ONLY_AFTER_TX=4
#EXEC_CMD_ON_SQL_CLOSE=500
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=en_US
RGR_SOUND_DELAY=10
RGR_SOUND_ALWAYS=0
REPORT_CTCSS=103.5
#TX_CTCSS=ALWAYS
MACROS=Macros
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
#ACTIVATE_MODULE_ON_LONG_CMD=4:EchoLink
#QSO_RECORDER=8:QsoRecorder
#ONLINE_CMD=998877
MUTE_RX_ON_TX=1
MUTE_TX_ON_RX=1

[MICToQRadioLink]
CONNECT_LOGICS=SimplexLogicMic:92:Mic,QRadioLinkSimplexLogic:92:SDR
DEFAULT_ACTIVE=0
TIMEOUT=0
#AUTOACTIVATE_ON_SQL=RepeaterLogic

[LocalRx]
TYPE=Local
#RX_ID=?
AUDIO_DEV=alsa:pulse
AUDIO_CHANNEL=0
AUDIO_DEV_KEEP_OPEN=1
#SQL_DET=VOX
#SQL_DET=SIGLEV
SQL_DET=EVDEV
SQL_START_DELAY=0
SQL_DELAY=0
#SQL_HANGTIME=2000
SQL_HANGTIME=0
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=1050
#CTCSS_MODE=2
#CTCSS_FQ=136.5
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
#SERIAL_PORT=/dev/ttyS0
#SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
EVDEV_DEVNAME=/dev/input/event6
EVDEV_OPEN=1,148,1
EVDEV_CLOSE=1,148,1
#GPIO_PATH=/sys/class/gpio
#GPIO_SQL_PIN=gpio30
#PTY_PATH=/tmp/rx1_sql
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=TONE
#SIGLEV_SLOPE=1
#SIGLEV_OFFSET=0
#SIGLEV_SLOPE=-108.33
#SIGLEV_OFFSET=517.49
SIGLEV_SLOPE=-91.59
SIGLEV_OFFSET=313.84
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40
#DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1
#FQ=433475000
#MODULATION=FM
#WBRX=WbRx1
#OB_AFSK_ENABLE=0
#OB_AFSK_VOICE_GAIN=6
#IB_AFSK_ENABLE=0

[LocalTx]
TYPE=Local
AUDIO_DEV=alsa:pulse
AUDIO_CHANNEL=0
AUDIO_DEV_KEEP_OPEN=1
PTT_TYPE=NONE
PTT_PORT=/dev/ttyUSB1
PTT_PIN=RTS
#HID_DEVICE=/dev/hidraw3
#HID_PTT_PIN=GPIO3
#SERIAL_SET_PINS=DTR!RTS
#GPIO_PATH=/sys/class/gpio
#PTT_HANGTIME=1000
TIMEOUT=0
TX_DELAY=500
#CTCSS_FQ=136.5
#CTCSS_LEVEL=9
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_DIGIT_PWR=-15
MASTER_GAIN=10.0
#OB_AFSK_ENABLE=0
#OB_AFSK_VOICE_GAIN=-6
#OB_AFSK_LEVEL=-12
#OB_AFSK_TX_DELAY=100
#IB_AFSK_ENABLE=0
#IB_AFSK_LEVEL=-6
#IB_AFSK_TX_DELAY=100

[QRadioLinkRx]
TYPE=Local
#RX_ID=?
AUDIO_DEV=udp:127.0.0.1:4937
AUDIO_CHANNEL=0
AUDIO_DEV_KEEP_OPEN=1
SQL_DET=PTY
SQL_START_DELAY=40
SQL_DELAY=40
SQL_HANGTIME=100
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
SQL_TIMEOUT=1000
#VOX_FILTER_DEPTH=10
#VOX_THRESH=1
#CTCSS_MODE=2
#CTCSS_FQ=103.5
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
#SERIAL_PORT=/dev/ttyS0
#SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
#GPIO_PATH=/sys/class/gpio
#GPIO_SQL_PIN=gpio30
PTY_PATH=/tmp/sql_pty
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=NOISE
#SIGLEV_SLOPE=1
#SIGLEV_OFFSET=0
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
#SIGLEV_OPEN_THRESH=30
#SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40
#DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1
#FQ=433475000
#MODULATION=FM
#WBRX=WbRx1
#OB_AFSK_ENABLE=0
#OB_AFSK_VOICE_GAIN=6
#IB_AFSK_ENABLE=0

[QRadioLinkTx]
TYPE=Local
AUDIO_DEV=udp:127.0.0.1:4938
AUDIO_CHANNEL=0
AUDIO_DEV_KEEP_OPEN=1
PTT_TYPE=NONE
PTT_PORT=/dev/ttyS0
PTT_PIN=DTRRTS
#HID_DEVICE=/dev/hidraw3
#HID_PTT_PIN=GPIO3
#SERIAL_SET_PINS=DTR!RTS
#GPIO_PATH=/sys/class/gpio
#PTT_HANGTIME=1000
TIMEOUT=1000
TX_DELAY=0
#CTCSS_FQ=136.5
#CTCSS_LEVEL=9
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_DIGIT_PWR=-15
#MASTER_GAIN=0.0
#OB_AFSK_ENABLE=0
#OB_AFSK_VOICE_GAIN=-6
#OB_AFSK_LEVEL=-12
#OB_AFSK_TX_DELAY=100
#IB_AFSK_ENABLE=0
#IB_AFSK_LEVEL=-6
#IB_AFSK_TX_DELAY=100
```

kantooon, 2025-05-30:
Alright, so just to clarify, these error messages have nothing to do with SVXlink right? They are seen on the output of qradiolink right?

If this is the case, then you should not have any audio from RF to the local audio device used by qradiolink (and the other direction as well), but svxlink is not involved here at all. I believe the error is caused by the inability of the Qt audio backend to be intialized with the device you have selected in the settings menu. Upon a little searching, it seems that you may be missing the required Qt plugins. Check please if you have installed libqt5multimedia5-plugins. Here is a similar issue with another application that uses the Qt audio backend: https://github.com/AlbrechtL/welle.io/issues/64

Regarding any issues with audio routing from qradiolink to svxlink and viceversa: to troubleshoot them, you can use Wireshark and set as packet filters the IP address and UDP ports configured in qradiolink and svxlink. If you do not see any UDP packets flowing in Wireshark, this means no audio samples are being sent. This can happen for several reasons: having some type of squelch configured in qradiolink could be one of them. But first let's solve your qradiolink local audio issue.

Let me know how it worked, since I'd like to update the  documentation if something is not clear or missing.

LaneaLucy, 2025-05-31:
Local audio is actually working, but after re compiling it yesterday, im now getting segmentation fault's when starting....

LaneaLucy, 2025-05-31:
Ok, found out when liblimesuiteng is installed, i get segmentation fault, without, it starts normally

kantooon, 2025-05-31:
qradiolink is not yet compatible with LimeSuiteNG, only the regular LimeSuite. in fact, it should not be linked at all at compile time to LiimeSuiteNG due to some issues related to the new gr-limesdr plugin that are not solved yet.

Can you provide a backtrace?
Start the program with gdb:
```
   $ gdb qradiolink
   $ r
  // after segfault:
   $ bt
```

LaneaLucy, 2025-05-31:
Wireshark shows that svxlink is sending audio, but getting icmp port unreachable. Qradiolink isn't sending udp, nor opening any port it looks like

kantooon, 2025-05-31:
How are you using qradiolink? From GUI or console?
Did you follow the instructions regarding enabling UDP audio mode?
https://qradiolink.org/docs/svxlink.html

LaneaLucy, 2025-05-31:
> https://qradiolink.org/docs/svxlink.html

Btw this link should also be added to the svxlink documentation

LaneaLucy, 2025-05-31:
The audio into svxlink is good, tested with the parrot. But everything into qradiolink is distorted. And changing mic level doesn't help. Lowering the value just gives me no audio

LaneaLucy, 2025-05-31:
And the SQL from qradiolink is broken too. Sometimes it closes when im sending another burst, but sometimes it never closes

LaneaLucy, 2025-05-31:
[image]

LaneaLucy, 2025-05-31:
[image]

LaneaLucy, 2025-05-31:
[image]

This is also a Problem....

kantooon, 2025-06-02:
Couldn't reproduce the issue described. Squelch is RMS power in GNU radio, and it should work correctly once configured.

LaneaLucy, 2025-06-03:
Like you can see on the last picture, it constantly open and close.
And the audio send out over RF via qradiolink is totally distorted. Lowering the mic level has no effect on distortion and only lowers volume until you hear nothing

kantooon, 2025-06-03:
Set the squelch 10 dB or more above the local noise floor. To avoid pulsatory noise use CTCSS. Check that the UDP sample rate in qradiolink matches the value for CARD_SAMPLE_RATE in svxlink.

kantooon, 2025-06-03:
I run both a repeater and a simplex node and the issues don't manifest. So it must be specific to your setup. 
I'll need more info to reprodiuce it. Please specify how you are using it (console, GUI, any arguments), what is the sequence of buttons you press when running (if any) and attach your complete qradiolink.cfg and svxlink.conf (minus any private information).

LaneaLucy, 2025-06-03:
GUI, no arguments.
Enable RX, enable TX, then using ptt with pulseaudio monitor or activating UDP sound.

Where do I find cfg?

LaneaLucy, 2025-06-05:
```
rx_device_args = "soapy=3,device=HackRFOne,driver=hackrf,label=HackRFOne#0325866e629417923,part_id=a000cb3c0066475b,serial=0000000000000000325866e629417923,version=2024.02.1";
tx_device_args = "soapy=3,device=HackRFOne,driver=hackrf,label=HackRFOne#0325866e629417923,part_id=a000cb3c0066475b,serial=0000000000000000325866e629417923,version=2024.02.1";
rx_antenna = "Auto";
tx_antenna = "Auto";
rx_freq_corr = 6;
tx_freq_corr = 6;
callsign = "CALL";
video_device = "/dev/video0";
audio_input_device = "alsa_output.pci-0000_00_1b.0.analog-stereo.monitor";
audio_output_device = "alsa_output.pci-0000_00_1b.0.analog-stereo";
tx_power = 21;
bb_gain = 3;
if_gain = 50;
rx_sensitivity = 69;
squelch = -36;
rx_volume = 91;
tx_volume = 50;
voip_volume = 50;
rx_frequency = 433809664L;
tx_carrier_offset = 0L;
tx_shift = 0L;
voip_server = "127.0.0.1";
voip_port = 64738;
voip_password = "";
rx_mode = 2;
tx_mode = 1;
rx_ctcss = 0.0;
tx_ctcss = 0.0;
ip_address = "10.0.0.1";
demod_offset = 190336L;
rx_sample_rate = 1000000L;
scan_step = 2500;
show_controls = 1;
show_constellation = 0;
show_time_domain = 0;
show_fft = 1;
fft_history = 1;
coloured_fft = 0;
enable_duplex = 0;
fft_size = 32768;
fft_averaging = 1.0;
wf_averaging = 0;
draw_constellation_eye = 0;
waterfall_fps = 15;
audio_compressor = 0;
enable_relays = 0;
rssi_calibration_value = -80;
control_port = 4939;
udp_listen_port = 4938;
udp_send_port = 4937;
agc_attack = 1;
agc_decay = 1;
remote_control = 0;
burst_ip_modem = 0;
night_mode = 0;
scan_resume_time = 5;
audio_record_path = "/home/lanealucy";
vox_level = 20;
voip_bitrate = 19200;
end_beep = 0;
mute_forwarded_audio = 1;
block_buffer_size = 0;
radio_tot = 120;
tot_tx_end = 0;
tx_band_limits = 0;
window_width = 1600;
window_height = 836;
relay_sequence = 15;
lnb_lo_freq = 0L;
panadapter_min_db = -120.0;
panadapter_max_db = -30.0;
gpredict_control = 0;
lime_rfe_device = "/dev/ttyUSB0";
enable_lime_rfe = 0;
lime_rfe_attenuation = 2;
lime_rfe_notch = 0;
mmdvm_channels = 3;
mmdvm_channel_separation = 25000;
burst_delay_msec = 60;
m17_can_tx = 0;
m17_can_rx = 0;
m17_src = "CALL";
m17_dest = "";
m17_decode_all_can = 1;
m17_destination_type = 0;
udp_audio_sample_rate = 16000;
sql_pty_path = "/tmp/sql_pty";
udp_audio_local_address = "127.0.0.1";
udp_audio_remote_address = "127.0.0.1";
zmq_proxy_channel = 5;
time_domain_sample_rate = 100000;
time_domain_sample_scaling = 1000;
time_domain_sample_speed = 50000;
time_domain_filter_width = 37500;
vocoder_plugin_path = "/tmp";
dmr_mode = 0;
dmr_timeslot = 1;
dmr_color_code = 1;
dmr_promiscuous_mode = 0;
dmr_vocoder = 0;
dmr_source_id = 1;
dmr_destination_id = 9;
dmr_call_type = 0;
dmr_timing_correction = 0;
dmr_talker_alias = "";
dmr_codec2_bitrate = 0;
```

and like you can see in my config, im using a hackrf one, with 1000000 samplerate.
reception is not the problem, sending (TX) is completely distorted...

LaneaLucy, 2025-06-05:
i now increased my sample rate, and audio is working (maybe put this info sowhere into documentation) and also sql isnt constantly opening and closing. but now i have a new sql problem.
looks like svxlink is only looking at the sql when there is audio. but it looks like qradiocontrol is closing sql and stopping sending audio over udp at the same time, which creates a race condition for svxlink. i would sugest to continue sending audio for another second (or at least some 100-250 ms delay) after closing sql, or never stopping sending audio

kantooon, 2025-06-06:
Not possible architecture-wise. Nor should it be necessary, unless there's some sort of regression in your version of svxlink. This solution has been tried and tested for a long time now.
If still having issues, try the workaround described above.