qradiolink/qradiolink #135: Compiling the latest version of qradiolink at Ubunu 24.04 gives errors

computer-wizard, 2025-06-29:
sh ./build_debian.sh 
+ mkdir -p build
+ cd src/ext/
+ protoc --cpp_out=. Mumble.proto
+ protoc --cpp_out=. QRadioLink.proto
+ cd ../../build/
+ qmake CONFIG+=debug ..
Info: creating stash file /home/boudewijn/DATV/qradiolink/build/.qmake.stash
Project MESSAGE: /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++
Project MESSAGE: Building for GNU/Linux
Project MESSAGE: Building without OpenGL support
Project MESSAGE: Building with Qt audio support.
+ make -j4
/usr/lib/qt5/bin/uic ../src/mainwindow.ui -o ui_mainwindow.h
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o ../src/main.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o dmrcontrol.o ../src/DMR/dmrcontrol.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o dmrtiming.o ../src/DMR/dmrtiming.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o dmrutils.o ../src/DMR/dmrutils.cpp
../src/DMR/dmrutils.cpp: In static member function ‘static void DMRUtils::parseISO7bitToISO8bit(unsigned char*, unsigned char*, unsigned int, unsigned int)’:
../src/DMR/dmrutils.cpp:106:35: warning: left operand of comma operator has no effect [-Wunused-value]
  106 |     for(unsigned int i=0,j=1,k=0;i<size,k<bit7_size;i++,j=j+1,k++)
      |                                  ~^~~~~
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o bursttimer.o ../src/bursttimer.cpp
../src/DMR/dmrcontrol.cpp: In member function ‘bool DMRControl::processAudio(DMRFrame&)’:
../src/DMR/dmrcontrol.cpp:233:13: warning: unused variable ‘errors’ [-Wunused-variable]
  233 |         int errors = frame.runAudioFEC();
      |             ^~~~~~
../src/DMR/dmrcontrol.cpp: In member function ‘void DMRControl::addFrames(std::vector<DMRFrame>)’:
../src/DMR/dmrcontrol.cpp:647:14: warning: variable ‘process_result’ set but not used [-Wunused-but-set-variable]
  647 |         bool process_result = true;
      |              ^~~~~~~~~~~~~~
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o dmrframe.o ../src/DMR/dmrframe.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o crc32.o ../src/ext/crc32.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o devices.o ../src/ext/devices.cpp
g++ -c -pipe -g -D_REENTRANT -Wall -Wextra -fPIC -DUSE_QT_AUDIO -DQT_MULTIMEDIAWIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I../../qradiolink -I. -I../src/M17 -I../src/MMDVM -I/usr/include/x86_64-linux-gnu/qt5 -I/usr/include/x86_64-linux-gnu/qt5/QtMultimediaWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtMultimedia -I/usr/include/x86_64-linux-gnu/qt5/QtWidgets -I/usr/include/x86_64-linux-gnu/qt5/QtGui -I/usr/include/x86_64-linux-gnu/qt5/QtNetwork -I/usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o gpredictcontrol.o ../src/gpredictcontrol.cpp
In file included from ../src/DMR/dmrframe.h:22,
                 from ../src/DMR/dmrframe.cpp:17:
../../qradiolink/src/DMR/constants.h:7:7: error: ‘uint64_t’ does not name a type
    7 | const uint64_t MS_DATA_SYNC_BITS  = 0xD5D7F77FD757U;
      |       ^~~~~~~~
../../qradiolink/src/DMR/constants.h:6:1: note: ‘uint64_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
    5 | #include "src/MMDVM/DMRDefines.h"
  +++ |+#include <cstdint>
    6 | 
../../qradiolink/src/DMR/constants.h:8:7: error: ‘uint64_t’ does not name a type
    8 | const uint64_t MS_VOICE_SYNC_BITS = 0x7F7D5DD57DFDU;
      |       ^~~~~~~~
../../qradiolink/src/DMR/constants.h:8:7: note: ‘uint64_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
../../qradiolink/src/DMR/constants.h:9:7: error: ‘uint64_t’ does not name a type
    9 | const uint64_t BS_DATA_SYNC_BITS  = 0xDFF57D75DF5DU;
      |       ^~~~~~~~
../../qradiolink/src/DMR/constants.h:9:7: note: ‘uint64_t’ is defined in header ‘<cstdint>’; did you forget to ‘#include <cstdint>’?
../../qradiolink/src/DMR/constants.h:10:7: error: ‘uint64_t’ does not name a type
   10 | const uint64_t BS_VOICE_SYNC_BITS = 0x755FD7DF75F7U;
      |       ^~~~~~~~
[... cut ...]
  486 |     memcpy(_frame_data, repacked_data, DMR_FRAME_LENGTH_BYTES);
      |            ^~~~~~~~~~~
      |            __locale_data
../src/DMR/dmrframe.cpp: In member function ‘void DMRFrame::addVoiceSync()’:
../src/DMR/dmrframe.cpp:500:28: error: ‘_frame_data’ was not declared in this scope; did you mean ‘__locale_data’?
  500 |     CSync::addDMRAudioSync(_frame_data, false);
      |                            ^~~~~~~~~~~
      |                            __locale_data
make: *** [Makefile:1974: dmrframe.o] Error 1
make: *** Waiting for unfinished jobs....

computer-wizard, 2025-06-30:
Did forget,

gcc/g++ version is:  13.3.0

kantooon, 2025-06-30:
I assume you are building it from the master branch right? There is a known issue where on some Linux distros the default include of <cstdint> was removed recently. It has been fixed in pull https://github.com/qradiolink/qradiolink/pull/131
Can you try building from the `next` branch instead? The change delta is minimal besides this fix.

closed 2025-07-01 by computer-wizard